zephyrproject-rtos/zephyr #69394: tests: kernel : cache: NXP- RT10xx :  enable CONFIG_CACHE_MANAGEMENT cause NXP RT series failure

hakehuang, 2024-02-23:
**Describe the bug**

the tests/kernel/cache case used to be run patial as below, it fails on user space due to MPU region limitation,

```
*** Booting Zephyr OS build zephyr-v3.5.0-3034-g9be3b8490b13 ***
Running TESTSUITE cache_api
===================================================================
START - test_data_cache_api
PASS - test_data_cache_api in 0.001 seconds
===================================================================
START - test_data_cache_api_user
E: Bad system call id 480 invoked
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x80002000 (test_data_cache_api_user)
E: Halting system
```

but after enable the CONFIG_CACHE_MANAGEMENT  even none_user_space case fails as well.

```
*** Booting Zephyr OS build v3.6.0-rc3-11-gdc8dff0fce08 ***
Running TESTSUITE cache_api
===================================================================
START - test_data_cache_api
```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 -  all NXP MIMXRT10xx series boards. e.g mimxrt1060_evk, mimxrt1020_evk
 - What have you tried to diagnose or workaround this issue?
 -  bisect found it is introduced with below commit
 ```
commit a2be813d99a4c2a9ba51d14810d2ee9bb635db9e
Author: Tomasz Bursztyka <tomasz.bursztyka@linux.intel.com>
Date:   Mon Nov 27 20:17:28 2023 +0100

    tests/kernel: Add qemu_x86/x86_64 to cache test and enable cache

    How cache test could be ran without CONFIG_CACHE_MANAGEMENT?

    Adding qemu_x86/x86_64 as cache related API in x86 has proven to be
    very brittle against changes recently: no test was at least verifying
    it builds properly, not to mention running it.

    Signed-off-by: Tomasz Bursztyka <tomasz.bursztyka@linux.intel.com>

 tests/kernel/cache/prj.conf      | 1 +
 tests/kernel/cache/testcase.yaml | 4 ++++
 2 files changed, 5 insertions(+)
```
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b mimxrt1020_evk tests/kernel/cache
2. west flash

**Expected behavior**
no user space parts case can pass

**Impact**
CONFIG_CACHE_MANAGEMENT  with RT series board

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.6.0-rc3-11-gdc8dff0fce08

erwango, 2024-02-23:
See https://github.com/zephyrproject-rtos/zephyr/issues/66710#issuecomment-1961382304

nashif, 2024-02-27:
@carlocaione  can you please take a look?

carlocaione, 2024-02-28:
The problem here is that we should not really use `sys_cache_data_invd_all` in the test, that's dangerous and not really related to MPU.

carlocaione, 2024-02-28:
Opened https://github.com/zephyrproject-rtos/zephyr/pull/69556

PR #69556 (2024-03-01): tests: kernel/cache: Do not invd the whole cache
That is dangerous and a big no-no. Remove it.

Fixes: #69394
closed 2024-03-01 by aescolar