zephyrproject-rtos/zephyr #69413: The Modbus RTU server is not working as expected.

seyednasermoravej, 2024-02-24:
**Describe the bug**
I done step by step according to [Modbus RTU](https://docs.zephyrproject.org/latest/samples/subsys/modbus/rtu_server/README.html)
But the result is different with what is shown in this page.

**To Reproduce**
1. run these commands: 
`west build -b nrf52840dk_nrf52840 samples/subsys/modbus/rtu_server -- -DDTC_OVERLAY_FILE=cdc-acm.overlay -DEXTRA_CONF_FILE=overlay-cdc-acm.conf`
`west flash`
2. install pymodbus using PIP.
3. run this comamnd:
`pymodbus.console serial --port COM8 --baudrate 19200  --bytesize 8 --parity N --stopbits 2`
5. run this command
> client.write_coils address=0 values=0,1,1 unit=1
6. run this command
> client.read_coils address=0 count=3 unit=1

**Expected behavior**
I'm expecting this output:
[image]

**Logs and console output**
But I received this output:
[image]

**Environment (please complete the following information):**
OS: windows 11
Zephyr version: 3.6
Zephyr sdk: 0.16.5
Target: nrf52840DK

**Additional context**
The only difference between the instructions implied in the document with what I have done, is that in the document is says `--port /dev/ttyUSB0` in `pymodbus.console serial --port /dev/ttyUSB0 --baudrate 19200 --bytesize 8 --parity N --stopbits 2` but i have changed the port because I'm using windows.

seyednasermoravej, 2024-02-24:
Update: I tested using a USB to TTL adapter instead of the NRF52840 USB, but I obtained the same result.

seyednasermoravej, 2024-04-23:
Any update?
@jfischer-no

seyednasermoravej, 2024-06-25:
Hi,
@nashif @henrikbrixandersen @jfischer-no 
Any update?

jfischer-no, 2024-07-01:
> 3\. `rity N --stopbits 2`
> 
>     4\. run this command
> 
> 
> > client.write_coils address=0 values=0,1,1 unit=1
> 
>     6. run this command
> 
> 
> > client.read_coils address=0 count=3 unit=1

Sorry, finally found some time to look at this and reproduce it, pymodbus changed the user interface in newer releases, our sample documentation is fixed in 8e026ba19e9a4706dfb269ccc8bfee1e50b2622. It looks like if device address (slave argument) is not specified, it defaults to 0, what is broadcast address, from the spec, "No response is returned to broadcast requests sent by the master. The broadcast requests are necessarily writing commands.".

https://docs.zephyrproject.org/latest/samples/subsys/modbus/rtu_server/README.html
```
> client.write_coils address=0 values=0,1,1 slave=1
{
    "address": 0,
    "count": 3
}

> client.read_coils address=0 count=3 slave=1
{
    "bits": [
        false,
        true,
        true,
        false,
        false,
        false,
        false,
        false
    ]
}
```

closed 2024-07-01 by jfischer-no