zephyrproject-rtos/zephyr #69433: openamp_rsc_table App in imx7d-pico (Side-m4 pico_pi_m4) I cant see /dev/ttyRPMSG

neuberfran, 2024-02-25:
Hi,

https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/ipc/openamp_rsc_table#id6

I stay trying this since last month in my imx7d-pico (m4 side) Zephyr upstream board is pico_pi_m4 using mailbox=mub

I can run Remoteproc and zephyr on this device but without resource table, vrings and mailbox(mu) by NXP: https://www.youtube.com/watch?v=NFyEJXIpq4E

Now I stay trying openamp_rsc_table
based on this link below:
https://github.com/zephyrproject-rtos/zephyr/pull/60455

I can go up and run everything but the channel and /dev/ttyRPMSG are not created

cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories
```
Carveout memory entry:
	Name: vdev0vring0
	Virtual address: 7938acbc
	DMA address: 0x90000000
	Device address: 0x90000000
	Length: 0x8000 Bytes

Carveout memory entry:
	Name: vdev0vring1
	Virtual address: 3a612924
	DMA address: 0x90008000
	Device address: 0x90008000
	Length: 0x8000 Bytes
```
 cat /sys/kernel/debug/remoteproc/remoteproc0/resource_table 
```
Entry 0 is of type vdev
  ID 7
  Notify ID 0
  Device features 0x1
  Guest features 0x1
  Config length 0x0
  Status 0x7
  Number of vrings 2
  Reserved (should be zero) [0][0]

  Vring 0
    Device Address 0x90000000
    Alignment 16
    Number of buffers 8
    Notify ID 0
    Physical Address 0x0

  Vring 1
    Device Address 0x90008000
    Alignment 16
    Number of buffers 8
    Notify ID 1
    Physical Address 0x0
```
pico_pi_m4.overlay
```

/ {
	chosen {
		/*
		 * shared memory reserved for the inter-processor communication
		 */
		zephyr,ipc_shm = &ocramfake;
		zephyr,ipc = &mub;
	};

	ocramfake: memory@90000000  {
		compatible = "mmio-sram";
		reg = <0x90000000  0x110000>;
	};
};

&mub {
	status = "okay";
};

**In linux Yocto A7 Side imx7d-pico-pi-qca-m4.dts (file):**
```
[image]

 #include "imx7d-pico-pi-qca.dts"

 / {
     memory {
         linux,usable-memory = <0x80000000 0x1ff00000>;
     };
 
     reserved-memory {
         #address-cells = <1>;
         #size-cells = <1>;
         ranges;
 
         //  mailbox_reserved: vrings0@30ab0000  {
        //    reg = <0x30ab0000 0x10000>;
        //    no-map;
        //  };

        m4_reserved: m4@90000000 {
           reg = <0x90000000 0x100000>;
           no-map;
         };        

        vdev0vring0: vdev0vring0@90000000 {
            compatible = "shared-dma-pool";
//            compatible = "shared-dma-pool";
            reg = <0x90000000  0x8000>;
            no-map;
        };

        vdev0vring1: vdev0vring1@90008000 {
            compatible = "shared-dma-pool";
//            compatible = "shared-dma-pool";
            reg = <0x90008000  0x8000>;
            no-map;
        };

        // rsc_table: rsc_table@900ff000 {
        //  reg = <0x900ff000  0x1000>;
        //  no-map;
        // };

        vdev0buffer: vdev0buffer@90010000 {
            compatible = "shared-dma-pool";
            reg = <0x90010000 0x80000>;
            no-map;
        };
 
        //  cm4tcmcode: cm4tcmcode@0x7f8000 {
        //      compatible = "shared-dma-pool";
        //      reg = <0x007f8000 0x8000>;
        //      no-map;
        //  };
 
        //  cm4sramcode: cm4sramcode@0x900000 {
        //      compatible = "shared-dma-pool";
        //      reg = <0x00900000 0x40000>;
        //      no-map;
        //  };
     };
 
     imx7d-cm4 {
         compatible = "fsl,imx7d-cm4";
         clocks = <&clks IMX7D_ARM_M4_ROOT_CLK>;
//         fsl,auto-boot;
         mbox-names = "tx0", "rx0", "rxdb0";
		 mboxes = <&mub 0 1
			  &mub 1 1
			  &mub 3 1>;
  //	     memory-region = <&rpmsg_vrings>, <&cm4tcmcode>, <&cm4sramcode>;
        memory-region = <&vdev0buffer>, <&vdev0vring0>, <&vdev0vring1>;
        //, <&cm4tcmcode>;
        fsl,rproc-auto-boot = <1>;
        fsl,rproc-fw-name="zephyr_openamp_rsc_table.elf";
        syscon = <&src>;
        status = "okay"; 
     };
 
     m4_tcm: tcml@007f8000 {
         compatible = "fsl, m4_tcml";
         reg = <0x007f8000 0x8000>;
     };
 
     gpio-leds {
         status = "disabled";
     };
 };
 
 &adc1 {
     status = "disabled";
 };
 
 &adc2 {
     status = "disabled";
 };
 
 &gpt3 {
     status = "disabled";
 };
 
 &gpt4 {
     status = "disabled";
 };
 
 &ocram {
     reg = <0x00901000 0xf000>;
 };
 
 &rpmsg{
     vdev-nums = <1>;
     memory-region = <&vdev0buffer>;
//     memory-region = <&vdev0vring0>, <&vdev0vring1>, <&vdevbuffer>, <&cm4tcmcode>, <&cm4sramcode>;
     reg = <0x9fff0000 0x10000>;
     status = "okay";   // Tava disabled
 };
 
 &uart6 {
     status = "disabled";
 };
 
 &mub {
     status = "okay";
 };
 
 &wdog3{
     status = "disabled";
 };

```

```
Why I can`t see vdev0buffer in result of the cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories And cat /sys/kernel/debug/remoteproc/remoteproc0/resource_table  commands

Why when I change MAX_RPMSG_NUM_BUFS and MAX_RPMSG_BUF_SIZE from 512 to 256 in virtio_rpmsg_bus.c (file in rpmsg/drivers in my linux kernel), nothing change in **Number of buffers 8** in my resource_table?
[image]

neuberfran, 2024-03-22:
https://linuxlists.cc/l/1/linux-kernel/t/3486837/(patch_1_2)_remoteproc:_imx_rproc:_dummy_kick_method#google_vignette

jlh-makeen, 2024-03-23:
Hello,

I'm trying to acheive the same RPSMG communication on an CL-SOM-IMX7D.

Have you got any progress or success ?

I've seen a first thing is that we are missing the ressource table definition in the linker script of the imx7d soc.

Similar to the one on the imx8m m7 core:
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/nxp/imx/imx8m/m7/linker.ld

When you compile the openamp_rsc_table example, there should not be saying "orphaned section .ressource_table...".

add the linker section to a linker.ld file like on the imx8m/m7 example and change the linker script path in the imx7d soc 
 Cmakelist:

```
...
if(CONFIG_OPENAMP_RSC_TABLE)
  zephyr_linker_section(NAME .resource_table GROUP ROM_REGION NOINPUT)
  zephyr_linker_section_configure(SECTION .resource_table KEEP INPUT ".resource_table*")
endif()

set(SOC_LINKER_SCRIPT ${CMAKE_CURRENT_SOURCE_DIR}/linker.ld CACHE INTERNAL "")
```

jlh-makeen, 2024-03-23:
BTW have you also compiled and inserted the drivers/rpmsg/imx_rpmsg_tty.c kernel module as mentionned by the sample ?

neuberfran, 2024-03-23:
@jlh-makeen I'm glad you showed up about imx7 and remoteproc using resource table. I'm kinda alone in this (imx7d) for now

[image]
```

echo start > /sys/class/remoteproc/remoteproc0/state
[  667.827358] remoteproc remoteproc0: powering up imx-rproc
[  667.833825] remoteproc remoteproc0: Booting fw image zephyr_openamp_rsc_table.elf, size 660180
[  667.842789]  remoteproc0#vdev0buffer: assigned reserved memory node vdev0buffer@00924000
[  667.851255] virtio_rpmsg_bus: probe of virtio1 failed with error -12
[  667.857725]  remoteproc0#vdev0buffer: registered virtio1 (type 7)
[  667.863868] remoteproc remoteproc0: remote processor imx-rproc is now up
root@scorpion:/lib/firmware# 
```
https://community.toradex.com/t/rpmsg-does-not-work-on-imx8m-plus/13740/92?page=4

-12 =  missing memory

Every codes to github analyzer:
https://gist.github.com/neuberfran/ce6861c962e8c5fa361ff9453ce85a10
==============================================================
#define RPMSG_MAX_SIZE		64 (imx_rpmsg_tty.c)
==============================================================
below = imx_rpmsg.c==============================================
#define REMOTE_READY_WAIT_MAX_RETRIES	500

#define RPMSG_NUM_BUFS		(128)
#define RPMSG_BUF_SIZE		(128)
#define RPMSG_BUFS_SPACE	(RPMSG_NUM_BUFS * RPMSG_BUF_SIZE)
#define RPMSG_VRING_ALIGN	(1024)
#define RPMSG_RING_SIZE	((DIV_ROUND_UP(vring_size(RPMSG_NUM_BUFS / 2, \
				RPMSG_VRING_ALIGN), PAGE_SIZE)) * PAGE_SIZE)

#define to_imx_virdev(vd) container_of(vd, struct imx_virdev, vdev)

rpdev->ivdev[i]->vring[0] = start;
			rpdev->ivdev[i]->vring[1] = start + 0x2000;
			start += 0x4000;
=============================================================
below = virtio_rpmsg_bus.c
=====================================================
#define MAX_RPMSG_NUM_BUFS	(128)
#define MAX_RPMSG_BUF_SIZE	(128)

/*
 * Local addresses are dynamically allocated on-demand.
 * We do not dynamically assign addresses from the low 1024 range,
 * in order to reserve that address range for predefined services.
 */
#define RPMSG_RESERVED_ADDRESSES	(1024)

neuberfran, 2024-03-23:
> BTW have you also compiled and inserted the [drivers/rpmsg/imx_rpmsg_tty.c](https://elixir.bootlin.com/linux/latest/source/drivers/tty/rpmsg_tty.c) kernel module as mentionned by the sample ?
> 
> [The Sample Docs](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/ipc/openamp_rsc_table#id6:~:text=Running%20the%20rpmsg,0x0401%3A%20Hello%20Zephyr)

Yes

And

You said: _I've seen a first thing is that we are missing the ressource table definition in the linker script of the imx7d soc._

I suggest you read and participate in this meeting
https://github.com/OpenAMP/open-amp/discussions/275#discussioncomment-8283357

neuberfran, 2024-03-23:
my **Kconfig.defconfig.mcimx7_m4**  don`t have config IPM_IMX_REV2 Compared to mimx8mm6_m4

# iMX7 M4 core series

# Copyright (c) 2017, NXP
# SPDX-License-Identifier: Apache-2.0

if SOC_MCIMX7_M4

config SOC
	default "mcimx7d"

config SYS_CLOCK_HW_CYCLES_PER_SEC
	default 200000000

config GPIO
	default y

endif # SOC_MCIMX7_M4

jlh-makeen, 2024-03-24:
You are using Yocto kirkstone with Linux kernel 5.15 right ?

A the moment I have a Kirkstone image with Linux 5.10.9 and a backport of the imx-rproc drivers from 5.15 and a hack to boot my image from OCRAM (our app program uses more than 32kB). Might be too outdated as mainline kernel imx-rproc driver for 5.10.9 has quite some differences from 5.15 driver.
```
![image](https://github.com/zephyrproject-rtos/zephyr/assets/67622881/492b0298-bba0-4574-b9b1-8568f1b4508e)
```

As soon as I define the mailboxes in the A7 DT and start the M4, I get a seg fault. That might be our fault from missing something in the backport, but we are struggling with Compulab not having released a kernel and patches for newer than linux 5.10.9.

```
root@cl-som-imx7:/lib/firmware# ./boot_m4.sh zephyr_pnet_ipm.elf
./boot_m4.sh: line 1: echo: writeremoteproc remoteproc0: powering up imx-rproc
 error: Invalid argument
remoteproc remoteproc0: Booting fw image zephyr_pnet_ipm.elf, size 1629760
rproc_elf_load_segments: So... booting from OCRAM on IMX7 ??? Grabbing PC and SP for you...
rproc_elf_load_segments: SP: 0x20003348 PC: 0x00908205
rproc_elf_load_segments: APPLYING TEMPORARY HACK TO GET SP AND PC LOADED INTO M4 BOOT VECTOR
8<--- cut here ---
Unhandled fault: imprecise external abort (0x1c06) at 0xc0800000
pgd = da7dd62d
[c0800000] *pgd=885d2811, *pte=8854365f, *ppte=8854345f
Internal error: : 1c06 [#1] PREEMPT SMP ARM
Modules linked in: wl18xx evbug wlcore mousedev caam_jr caamkeyblob_desc caamhash_desc caamalg_desc crypto_engine wlcore_sdio imx_sdma caam secvio error pnet(O) fsl_ftm_ic(O)
CPU: 0 PID: 371 Comm: sh Tainted: G           O      5.10.9 #1
Hardware name: Freescale i.MX7 Dual (Device Tree)
PC is at imx_mu_xcr_rmw+0x0/0x70
LR is at imx_mu_generic_tx+0x8c/0xf8
pc : [<809266e0>]    lr : [<80926b10>]    psr: 600f0093
sp : 851e9c48  ip : 80dc7cb4  fp : b8144e00
r10: 00002000  r9 : b8144000  r8 : 00000000
r7 : f5ab0004  r6 : 00000000  r5 : 8409b040  r4 : 00000000
r3 : 00000016  r2 : 00000000  r1 : 00400000  r0 : 8409b040
Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 10c5387d  Table: 8849006a  DAC: 00000051
Process sh (pid: 371, stack limit = 0xab2517e1)
Stack: (0x851e9c48 to 0x851ea000)
9c40:                   8409b128 600f0013 8409b1a0 851e9ca8 00000cc0 80925aac
9c60: 00000036 000007ff 86749260 6759df8c 851e9c78 8409b128 00000000 851e9ca8
9c80: 8409b1a0 80925c60 00000000 84290ae8 84ec0400 00000000 00000cc0 8092c494
9ca0: b8144e00 8092db84 00000000 6759df8c 00000001 885d3180 00000001 8092afd4
9cc0: 885d3180 8062e4ac 885d3100 8092de8c 00000035 00000000 00000000 80308de0
9ce0: 884119f8 81053348 00000000 bfdabd92 00000e00 00000200 00000000 8092e104
9d00: 8092d840 885d3180 885d3200 6759df8c 84ec0600 84ec0410 84ec0400 00000000
9d20: 8141ec14 00000001 00000000 00000001 00000000 8062e38c 84ec0410 00000000
9d40: 814971e4 84ec0410 00000000 814971e4 00000000 814971dc 8141ec14 00000021
9d60: 00000000 806ca270 00000000 84ec0410 8141ec14 806caa88 84ec0454 00000000
9d80: 00000000 814097c0 00000000 806ca930 00000000 851e9dc4 806caa88 806c833c
9da0: 00000000 8415d06c 8437f338 6759df8c 84ec0410 813fb0a0 00000001 806ca700
9dc0: 8106acd8 84ec0410 00000001 6759df8c 84ec0410 84ec0410 813fb0a0 84ec0410
9de0: 84ec1c20 806c93d4 84ec0410 00000000 814971b4 806c5240 813fb0a0 84ec0410
9e00: 84290800 80589940 84ec0400 6759df8c 84ec0400 8092aff8 813fb0a0 84ec0410
9e20: 84290800 84ec0400 88329a40 8062e018 00000000 84ec1c00 00000007 84ec1c20
9e40: 84290800 8092b60c 84ec1c04 84290a48 84290820 88329c80 84290820 88329c80
9e60: 88329a40 80c8876c 84290800 00000000 84290820 80c8897c 842909f8 84290800
9e80: 00000000 84290a04 84290820 80929658 88329c80 6759df8c 84290820 88329cc0
9ea0: 00000006 84290800 8504be90 00000051 00000000 8092ada4 00000006 8504be80
9ec0: 88329cc0 851e9f80 8504be90 8030b700 00000000 00000000 846f6900 846f6780
9ee0: 00000006 8030b608 02082100 851e9f80 851e8000 00000004 00000000 802718d8
9f00: 843f2000 6759df8c 00000001 00000800 84cc7240 00000001 00000000 00000001
9f20: 84cc7240 0000000a 84cc7240 80285660 ffffe000 84cc7240 0000000a 851e8000
9f40: 84cc7240 6759df8c 00000000 84cc7240 883c0a28 883c0800 00000000 6759df8c
9f60: 851e9fb0 846f6780 846f6780 00000000 00000000 80100264 851e8000 80271d80
9f80: 00000000 00000000 80100264 6759df8c 00000006 00000006 02082100 76fa65a0
9fa0: 00000004 80100240 00000006 02082100 00000001 02082100 00000006 00000000
9fc0: 00000006 02082100 76fa65a0 00000004 00000006 0058cc00 00000000 00000000
9fe0: 00000004 7e8729c8 76ec06e7 76e41ae6 40010030 00000001 00000000 00000000
[<809266e0>] (imx_mu_xcr_rmw) from [<000007ff>] (0x7ff)
Code: ebdfb925 e5845000 e3a00000 e8bd8070 (e92d47f0)
---[ end trace 15ec0de3866fc2d9 ]---
note: sh[371] exited with preempt_count 1
Segmentation fault
```

neuberfran, 2024-03-24:
@jlh-makeen Yes I have Yocto Kirkstone 5.15.52.

What about your device vendor? Have qspi (I don`t have in imx7d-pico by TechNexion)? Have debug port? (I don`t have).

I stay now with issue with virtio0:
https://gist.github.com/neuberfran/2ba9978af735a7c19f90508bae0ec01b
[image]

Why I can`t see vdev0buffer in result of the cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories And cat /sys/kernel/debug/remoteproc/remoteproc0/resource_table commands?
[image]

jlh-makeen, 2024-03-25:
We are trying to achieve the same goal which is have /dev/ttyRPMSG up in Linux to talk to the M4 chip.

We are using the Compulab vendor with their CL-SOM-IMX7. The module is a 1GB Ram variant.
We have placed the sh_men area at the end of the 1GB.

But their BSP is outdated and is using the linux kernel lf-5.10.y with vendor patches:
https://github.com/compulab-yokneam/meta-bsp-imx7/tree/kirkstone

We also have a custom carrier board with 4x RS485 ports used by the Cortex M and 1x UART for the Cortex A.

My kernel at the moment:
```
Linux cl-som-imx7 5.10.9 #1
```

As per my colleague who is responsible for the Linux yocto image, the generic ttyRPMSG driver, mentionned by the openamp_rsc_table sample was introduced in Linux 5.16 (CONFIG_TTY_RPMSG).

I'm using this Zephyr version, before the switch to v2 boards/soc directory structure.
```
*** Booting Zephyr OS build v3.6.0-294-gec089ac5c251 ***
````

```
-> My DT on zephyr side, we boot from OCRAM at 0x0901000
/ {
	model = "Compulab CL-SOM-IMX7d on Makeen Datagateway carrier";
	compatible = "nxp,mcimx7d_m4";

	chosen {
		zephyr,flash = &ocram_code_linux; /*&tcml_code;*/
		zephyr,sram = &tcmu_sys;
		zephyr,console = &uart2;
		zephyr,shell-uart = &uart2;
		zephyr,ipc_shm = &ddr_shm;
		zephyr,ipc = &mub;
	};

	ddr_shm: memory1@bff00000 {
		compatible = "mmio-sram";
		/* use DDR_SYS area as mmio, 0xbff00000 0x100000 */
		reg = <0xbff00000 0x100000>;
		status = "okay";
	};
	
	...
        soc {
              ocram_code_linux: code@901000 {
	              compatible = "nxp,imx-code-bus";
	              reg = <0x00901000 DT_SIZE_K(124)>;
              };
        };
```

Regarding the A7 device tree:

```
       	mu@30aa0000 {
		compatible = "fsl,imx7d-mu\0fsl,imx6sx-mu";
		reg = <0x30aa0000 0x10000>;
		interrupts = <0x00 0x58 0x04>;
		clocks = <0x01 0x1b1>;
		clock-names = "mu";
		#mbox-cells = <0x02>;
		phandle = <0x4a>;
	};

	mu@30ab0000 {
		compatible = "fsl,imx7d-mu\0fsl,imx6sx-mu";
		reg = <0x30ab0000 0x10000>;
		interrupts = <0x00 0x61 0x04>;
		clocks = <0x01 0x1b1>;
		clock-names = "mu";
		#mbox-cells = <0x02>;
		phandle = <0x60>;
	};
       ...
       rpmsg {
              compatible = "fsl,imx7d-rpmsg";
              mbox-names = "tx\0rx\0rxdb";
              mboxes = <0x4a 0x00 0x01 0x4a 0x01 0x01 0x4a 0x03 0x01>;
              status = "okay";
      };
      ...
	reserved-memory {
		#address-cells = <0x01>;
		#size-cells = <0x01>;
		ranges;

		cm4tcmcode@7f8000 {
			compatible = "shared-dma-pool";
			reg = <0x7f8000 0x8000>;
			no-map;
			phandle = <0x57>;
		};

		cm4sramcode@901000 {
			compatible = "shared-dma-pool";
			reg = <0x901000 0x40000>;
			no-map;
			phandle = <0x58>;
		};

		vdev0vring0@bff00000 {
			compatible = "shared-dma-pool";
			reg = <0xbff00000 0x1000>;
			no-map;
			phandle = <0x61>;
		};

		vdev0vring1@bff01000 {
			compatible = "shared-dma-pool";
			reg = <0xbff01000 0x1000>;
			no-map;
			phandle = <0x62>;
		};

		vdev0buffer@bff02000 {
			compatible = "shared-dma-pool";
			reg = <0xbff02000 0xfe000>;
			no-map;
			phandle = <0x56>;
		};
	};

	imx7d-cm4 {
		compatible = "fsl,imx7d-cm4";
	    mbox-names = "tx", "rx", "rxdb";
		mboxes = <0x60 0x00 0x01 0x60 0x01 0x01 0x60 0x03 0x01>;
		memory-region = <0x56 0x61 0x62 0x57 0x58>;
		syscon = <0x23>;
		clocks = <0x01 0x42>;
	};
```

And the carveout/ressource tables outputs:

```
root@cl-som-imx7:/# cat /sys/kernel/debug/remoteproc/remoteproc0/resource_table
Entry 0 is of type vdev
  ID 7
  Notify ID 0
  Device features 0x1
  Guest features 0x1
  Config length 0x0
  Status 0x7
  Number of vrings 2
  Reserved (should be zero) [0][0]

  Vring 0
    Device Address 0xbff00000
    Alignment 16
    Number of buffers 8
    Notify ID 0
    Physical Address 0x0

  Vring 1
    Device Address 0xbff01000
    Alignment 16
    Number of buffers 8
    Notify ID 1
    Physical Address 0x0

root@cl-som-imx7:/# cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories
Carveout memory entry:
        Name: vdev0buffer
        Virtual address: dd5cdfbe
        DMA address: 0xbff02000
        Device address: 0xbff02000
        Length: 0xfe000 Bytes

Carveout memory entry:
        Name: vdev0vring0
        Virtual address: 473e2d00
        DMA address: 0xbff00000
        Device address: 0xbff00000
        Length: 0x1000 Bytes

Carveout memory entry:
        Name: vdev0vring1
        Virtual address: 8d625bae
        DMA address: 0xbff01000
        Device address: 0xbff01000
        Length: 0x1000 Bytes

Carveout memory entry:
        Name: cm4tcmcode
        Virtual address: 02e0dcc6
        DMA address: 0x007f8000
        Device address: 0x7f8000
        Length: 0x8000 Bytes

Carveout memory entry:
        Name: cm4sramcode
        Virtual address: 3ecff66e
        DMA address: 0x00901000
        Device address: 0x901000
        Length: 0x40000 Bytes

```

neuberfran, 2024-03-25:
@jlh-makeen 

You said: _"As per my colleague who is responsible for the Linux yocto image, the generic ttyRPMSG driver, mentionned by the openamp_rsc_table sample was introduced in Linux 5.16 (CONFIG_TTY_RPMSG)."_

I never knew anything about it. If you could provide more details, I would appreciate it.

jlh-makeen, 2024-03-25:
@neuberfran 

I managed to fix my driver crashing and can now start my M4 processor correctly and have rproc probe the core and kick the MU.

```
root@cl-som-imx7:/lib/firmware# ./boot_m4.sh zephyr_pnet_ipm.elf
./boot_m4.sh: line 1: echo: writeremoteproc remoteproc0: powering up imx-rproc
 error: Invalid argument
remoteproc remoteproc0: Booting fw image zephyr_pnet_ipm.elf, size 1638784
rproc_elf_load_segments: So... booting from OCRAM on IMX7 ??? Grabbing PC and SP for you...
rproc_elf_load_segments: SP: 0x20003348 PC: 0x009086a1
rproc_elf_load_segments: APPLYING TEMPORARY HACK TO GET SP AND PC LOADED INTO M4 BOOT VECTOR
 remoteproc0#vdev0buffer: vdev 0 buffer already mapped
virtio_rpmsg_bus virtio0: rpmsg host is online
 remoteproc0#vdev0buffer: registered virtio0 (type 7)
remoteproc remoteproc0: remote processor imx-rproc is now up

root@cl-som-imx7:/lib/firmware# cat /sys/kernel/debug/remoteproc/remoteproc0/resource_table
Entry 0 is of type vdev
  ID 7
  Notify ID 0
  Device features 0x1
  Guest features 0x1
  Config length 0x0
  Status 0x7
  Number of vrings 2
  Reserved (should be zero) [0][0]

  Vring 0
    Device Address 0xbff00000
    Alignment 16
    Number of buffers 64
    Notify ID 0
    Physical Address 0x0

  Vring 1
    Device Address 0xbff01000
    Alignment 16
    Number of buffers 64
    Notify ID 1
    Physical Address 0x0

root@cl-som-imx7:/lib/firmware# cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories
Carveout memory entry:
        Name: vdev0buffer
        Virtual address: 34229b00
        DMA address: 0xbff02000
        Device address: 0xbff02000
        Length: 0xfe000 Bytes

Carveout memory entry:
        Name: vdev0vring0
        Virtual address: 9abcaa63
        DMA address: 0xbff00000
        Device address: 0xbff00000
        Length: 0x1000 Bytes

Carveout memory entry:
        Name: vdev0vring1
        Virtual address: bc7cb9f6
        DMA address: 0xbff01000
        Device address: 0xbff01000
        Length: 0x1000 Bytes

Carveout memory entry:
        Name: cm4tcmcode
        Virtual address: 5f71bfd0
        DMA address: 0x007f8000
        Device address: 0x7f8000
        Length: 0x8000 Bytes

Carveout memory entry:
        Name: cm4sramcode
        Virtual address: 9f061b54
        DMA address: 0x00901000
        Device address: 0x901000
        Length: 0x40000 Bytes

```

Yes, up until 5.15 we had the imx_rpmsg_tty driver but there was introduced a new more generic one as rpmsg_tty from 5.16. Not sure it has more feature though.

Check out the nxp/linux-imx repo, latest revision:
https://github.com/nxp-imx/linux-imx/blob/lf-6.6.y/drivers/tty/rpmsg_tty.c

Which is different from:
https://github.com/nxp-imx/linux-imx/blob/lf-6.6.y/drivers/rpmsg/imx_rpmsg_tty.c

It is now part of the imx_v8_defconfig file:
https://github.com/nxp-imx/linux-imx/blob/lf-6.6.y/arch/arm64/configs/imx_v8_defconfig#L959

I'm also suspecting that something is missing in about the NS annoucement in the RPMSG on Linux. 
because the Zephyr side sends some NS annoucements for RPMSG_TTY and RPMSG_CLIENT_SAMPLE endpoints.
Therefore my guess is that these should wakeup the right driver that should get probed and mounted by the system.

```
/* from zephyrproject/lib/open-amp/lib/rpmsg/rpmsg.c */

int rpmsg_create_ept(struct rpmsg_endpoint *ept, struct rpmsg_device *rdev,
		     const char *name, uint32_t src, uint32_t dest,
		     rpmsg_ept_cb cb, rpmsg_ns_unbind_cb unbind_cb)
{
	int status = RPMSG_SUCCESS;
	uint32_t addr = src;
	...
	/* Send NS announcement to remote processor */
	if (ept->name[0] && rdev->support_ns &&
	    ept->dest_addr == RPMSG_ADDR_ANY)
		status = rpmsg_send_ns_message(ept, RPMSG_NS_CREATE);
       ...
}
```

Here is an associated driver source that listens to annoucements:
https://github.com/nxp-imx/linux-imx/blob/lf-6.6.y/drivers/rpmsg/rpmsg_ns.c

jlh-makeen, 2024-03-25:
Here is a hint that may indicate that I'm right about the NS annoucement thing:

Linux lf 6.6.y:
[image]
[image]

The tty_rpmsg and imx_rpmsg_tty don't use the same annoucement id's.

And guess what, the Zephyr implementation uses the "rpmsg-tty" channel id:
[image]

neuberfran, 2024-03-25:
@jlh-makeen 

You have a lot of useful knowledge.

Please help me more on my imx7d-pico (Remoteproc and echo test > /dev/ttyRPMSG regarding mounting the DT and the zephyr overlay file:
https://docs.zephyrproject.org/latest/boards/technexion/pico_pi/doc/index.html
[image]
There is this undefined question about less to m4:
https://community.toradex.com/t/is-it-possible-to-increase-the-m4-memory-in-the-vf61/5591/6
And about why it starts at 0x1ff00000 and not 0x20000000:
https://community.nxp.com/t5/Other-NXP-Products/RPMsg-in-imx7d-Why-0x1FF00000-and-Not-0x10000000-in-linux-usable/m-p/1805773#M21111

As you will see below, I am now focusing on using OCRAM (0X00901000 using F000) and not 9FF00000 in Remoteproc. Is it right?
My last DT on A7-M4-Linux Kirkstone Side is this (I have rpmsg_ctrl0 and rpmsg_ctrl1, but I never have /dev/ttyRPMSG):
```

/*
 * Copyright 2018 TechNexion Ltd.
 *
 * Author: Tapani Utriainen <tapani@technexion.com>
 *         Richard Hu <richard.hu@technexion.com>
 *         Ray Chang <ray.chang@technexion.com>
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License version 2 as
 * published by the Free Software Foundation.
 *
 */

 #include "imx7d-pico-pi-qca.dts"

 / {
     memory {
         linux,usable-memory = <0x80000000 0x1ff00000>;
     };
 
     reserved-memory {
         #address-cells = <1>;
         #size-cells = <1>;
         ranges;
 
         //  mailbox_reserved: vrings0@30ab0000  {
        //    reg = <0x30ab0000 0x10000>;
        //    no-map;
        //  };

        // m4_reserved: m4@8ff00000 {
        //    reg = <0x8ff00000 0x100000>;
        //    no-map;
        //  };        

         vdev0vring0: vdev0vring0@00902000 {
            compatible = "shared-dma-pool";
//            compatible = "shared-dma-pool";
//            reg = <0x9ff00000  0x8000>;
            reg = <0x00902000 0x2000>;
            no-map;
        };

        vdev0vring1: vdev0vring1@00904000 {
            compatible = "shared-dma-pool";
//            compatible = "shared-dma-pool";
//            reg = <0x9ff08000  0x8000>;
            reg = <0x00904000 0x2000>;
            no-map;
        };

        // rsc_table: rsc_table@900ff000 {
        //  reg = <0x900ff000  0x1000>;
        //  no-map;
[... 237 lines cut ...]

root@scorpion:/sys/bus/virtio/devices# cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories
Carveout memory entry:
	Name: vdev0vring0
	Virtual address: 26d194e1
	DMA address: 0x00902000
	Device address: 0x902000
	Length: 0x2000 Bytes

Carveout memory entry:
	Name: vdev0vring1
	Virtual address: d8145475
	DMA address: 0x00904000
	Device address: 0x904000
	Length: 0x2000 Bytes
root@scorpion:/lib/firmware# modprobe rpmsg_client_sample && modprobe imx_rpmsg_tty 
[   52.776939] NLB imx_rpmmsg_tty
root@scorpion:/lib/firmware# echo start > /sys/class/remoteproc/remoteproc0/state    
[   58.498550] remoteproc remoteproc0: powering up imx-rproc
[   58.522741] remoteproc remoteproc0: Booting fw image zephyr_openamp_rsc_table.elf, size 660180
[   58.531721]  remoteproc0#vdev0buffer: assigned reserved memory node vdev0buffer@00906000
[   58.543938] virtio_rpmsg_bus virtio1: rpmsg host is online
[   58.551535]  remoteproc0#vdev0buffer: registered virtio1 (type 7)
[   58.557738] remoteproc remoteproc0: remote processor imx-rproc is now u
```

jlh-makeen, 2024-03-25:
@neuberfran  

Well it's hard to tell because I've not gotten /dev/ttyRPMSG to appear on my system yet.

I don't see errors in when you start the M4 core though, so that's a first good thing.

I'm not sure if the shared memories and vrings can reside in the OCRAM region 0x009xxxxx.

On our system we have put them in DDR area at address 0xBFF00000 since we only have 1GB of RAM (from 0x80000000 to 0xBFFFFFFF).
The nice thing is that when using the DDR, the addresses are the same on A7 and  M4 "System bus". (aka from 0x80000000)

I believe you are accessing the sh_mem on the "System bus" aka as RAM and not CODE. Therefore pay attention to the addresses.
[image]
I think that if you use the OCRAM, you need to change your Zephyr shared memory address to the matching in 0x20200000's.

Maybe I'm wrong, but I'd suggest you use DDR instead of OCRAM.

If you have 1GB ram, use 0xBFF00000, if you have 512MB, use 0x9FF00000
In your case it would probably look something like this:
```
 reserved-memory {
         #address-cells = <1>;
         #size-cells = <1>;
         ranges;
    

         vdev0vring0: vdev0vring0@9ff00000 {
            compatible = "shared-dma-pool";
            reg = <0x9ff00000 0x1000>;
            no-map;
        };

        vdev0vring1: vdev0vring1@9ff01000 {
            compatible = "shared-dma-pool";
            reg = <0x9ff01000 0x1000>;
            no-map;
        };

        vdev0buffer: vdev0buffer@9ff02000 {
            compatible = "shared-dma-pool";
              reg = <0x9ff02000 0xfe000>;
            no-map;
        };
 
        cm4tcmcode: cm4tcmcode@0x7f8000 {
             compatible = "shared-dma-pool";
             reg = <0x007f8000 0x8000>;
             no-map;
         };
 
         cm4sramcode: cm4sramcode@0x900000 {
             compatible = "shared-dma-pool";
             reg = <0x00900000 0x40000>;
            no-map;
         };
     };
```

neuberfran, 2024-03-25:
@jlh-makeen I have 1gb here (imx7d-pico with 1gb Model and not 512mb model)

Are you using ipm_imx_rev2??? Or not?

For a better basis, we have this updated an5317:
https://www.nxp.com/docs/en/application-note/AN5317.pdf

jlh-makeen, 2024-03-25:
> @jlh-makeen I have 1gb here (imx7d-pico with 1gb Model and not 512mb model)
> 
> Are you using ipm_imx_rev2??? Or not?
> 
> For a better basis, we have this updated an5317: https://www.nxp.com/docs/en/application-note/AN5317.pdf

I don't believe we are using IPM_IMX_REV2 no.

jlh-makeen, 2024-03-26:
@neuberfran 

I'm sorry I'm no sure if this will be useful.

One thing my colleague pointed out is that the function virtqueue_get_available_buffer() fails on my zephyr program:
[image]

This could explain why the Name Service (NS) notification don't wakeup the drivers in Linux (as for my case at least).

Here is what I can log from my zephyr appplication:
```
OpenAMP[remote] Linux responder demo started
metal_init
:metal_register_generic_device
:metal_device_open
:metal_io_init
:metal_device_io_region
:rsc_table_get
:metal_io_init
:metal_device_io_region
:device_is_ready
[00:00:00.541,000] \x1B[0m<dbg> openamp_rsc_table: platform_ipm_callback: platform_ipm_callback: msg received from mb 1
\x1B[0m
:ipm_set_enabled
:platform_init
rproc_virtio_create_vdev
rproc_virtio_create_vdev
rproc_virtio_init_vring0 : 0xbff00000 / 64
rproc_virtio_init_vring1: 0xbff01000 / 64
rpmsg_init_vdev
:platform_create_rpmsg_vdev
[00:00:00.542,000] \x1B[0m<dbg> openamp_rsc_table: receive_message: recv message
\x1B[0m
ERR: virtqueue_get_available_buffer
OpenAMP[remote] Linux sample client responder started
[00:00:00.602,000] \x1B[0m<dbg> openamp_rsc_table: mailbox_notify: mailbox_notify: msg received -> priv:(nil) id:0
\x1B[0m
Endpoint rpmsg-client-sample created: res:0 support_ns: 1

OpenAMP[remote] Linux TTY responder started
[00:00:00.602,000] \x1B[0m<dbg> openamp_rsc_table: mailbox_notify: mailbox_notify: msg received -> priv:(nil) id:0
\x1B[0m
Endpoint rpmsg-tty created: res:0 support_ns: 1
```

neuberfran, 2024-03-30:
@jlh-makeen 

Unfortunately, I still don't have carveout for vdev0buffer or /dev/ttyRPMSG.
```

root@scorpion:/lib/firmware# modprobe rpmsg_client_sample && modprobe imx_rpmsg_tty 
[   86.807476] NLB imx_rpmmsg_tty
root@scorpion:/lib/firmware# echo start > /sys/class/remoteproc/remoteproc0/state    
[   92.828659] remoteproc remoteproc0: powering up imx-rproc
[   92.852848] remoteproc remoteproc0: Booting fw image zephyr_openamp_rsc_table.elf, size 660180
[   92.861895]  remoteproc0#vdev0buffer: assigned reserved memory node vdev0buffer@9ff02000
[   92.875105] virtio_rpmsg_bus virtio1: rpmsg host is online
[   92.882776]  remoteproc0#vdev0buffer: registered virtio1 (type 7)
[   92.889343] remoteproc remoteproc0: remote processor imx-rproc is now up
root@scorpion:/lib/firmware# cat /sys/kernel/debug/remoteproc/remoteproc0/resource_table 
Entry 0 is of type vdev
  ID 7
  Notify ID 0
  Device features 0x1
  Guest features 0x1
  Config length 0x0
  Status 0x7
  Number of vrings 2
  Reserved (should be zero) [0][0]

  Vring 0
    Device Address 0x9ff00000
    Alignment 16
    Number of buffers 16
    Notify ID 0
    Physical Address 0x0

  Vring 1
    Device Address 0x9ff01000
    Alignment 16
    Number of buffers 16
    Notify ID 1
    Physical Address 0x0

root@scorpion:/lib/firmware# cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories
Carveout memory entry:
	Name: vdev0vring0
	Virtual address: 7b0edb64
	DMA address: 0x9ff00000
	Device address: 0x9ff00000
	Length: 0x1000 Bytes

Carveout memory entry:
	Name: vdev0vring1
	Virtual address: 7c87c534
	DMA address: 0x9ff01000
	Device address: 0x9ff01000
	Length: 0x1000 Bytes

Carveout memory entry:
	Name: cm4tcmcode
	Virtual address: 75013cd3
	DMA address: 0x007f8000
	Device address: 0x7f8000
	Length: 0x8000 Bytes

Carveout memory entry:
	Name: cm4sramcode
	Virtual address: 243f5356
	DMA address: 0x00900000
	Device address: 0x900000
	Length: 0x40000 Bytes

root@scorpion:/lib/firmware# ls /dev/r*
/dev/ram0   /dev/ram13	/dev/ram4  /dev/ram9	     /dev/rpmsg_ctrl1
/dev/ram1   /dev/ram14	/dev/ram5  /dev/random	     /dev/rtc
/dev/ram10  /dev/ram15	/dev/ram6  /dev/remoteproc0  /dev/rtc0
/dev/ram11  /dev/ram2	/dev/ram7  /dev/rfkill
/dev/ram12  /dev/ram3	/dev/ram8  /dev/rpmsg_ctrl0
root@scorpion:/lib/firmware# ls /dev/tty*
/dev/tty    /dev/tty19	/dev/tty3   /dev/tty40	/dev/tty51  /dev/tty62
/dev/tty0   /dev/tty2	/dev/tty30  /dev/tty41	/dev/tty52  /dev/tty63
/dev/tty1   /dev/tty20	/dev/tty31  /dev/tty42	/dev/tty53  /dev/tty7
/dev/tty10  /dev/tty21	/dev/tty32  /dev/tty43	/dev/tty54  /dev/tty8
/dev/tty11  /dev/tty22	/dev/tty33  /dev/tty44	/dev/tty55  /dev/tty9
/dev/tty12  /dev/tty23	/dev/tty34  /dev/tty45	/dev/tty56  /dev/ttymxc4
/dev/tty13  /dev/tty24	/dev/tty35  /dev/tty46	/dev/tty57  /dev/ttymxc6
/dev/tty14  /dev/tty25	/dev/tty36  /dev/tty47	/dev/tty58
/dev/tty15  /dev/tty26	/dev/tty37  /dev/tty48	/dev/tty59
/dev/tty16  /dev/tty27	/dev/tty38  /dev/tty49	/dev/tty6
/dev/tty17  /dev/tty28	/dev/tty39  /dev/tty5	/dev/tty60
/dev/tty18  /dev/tty29	/dev/tty4   /dev/tty50	/dev/tty61
root@scorpion:/lib/firmware# 
root@scorpion:/lib/firmware# dmesg | grep -i "remoteproc"
[    2.139313] remoteproc remoteproc0: imx-rproc is available
[    2.159148] remoteproc remoteproc0: attaching to imx-rproc
[    2.164721] remoteproc remoteproc0: remote processor imx-rproc is now attached
[   27.148570] remoteproc remoteproc0: stopped remote processor imx-rproc
[   92.828659] remoteproc remoteproc0: powering up imx-rproc
[   92.852848] remoteproc remoteproc0: Booting fw image zephyr_openamp_rsc_table.elf, size 660180
[   92.861895]  remoteproc0#vdev0buffer: assigned reserved memory node vdev0buffer@9ff02000
[   92.882776]  remoteproc0#vdev0buffer: registered virtio1 (type 7)
[   92.889343] remoteproc remoteproc0: remote processor imx-rproc is now up
root@scorpion:/lib/firmware# dmesg | grep -i "virtio0"
[    0.134940] virtio_rpmsg_bus virtio0: rpmsg host is online
root@scorpion:/lib/firmware# dmesg | grep -i "virtio1"
[   92.875105] virtio_rpmsg_bus virtio1: rpmsg host is online
[   92.882776]  remoteproc0#vdev0buffer: registered virtio1 (type 7)
root@scorpion:/lib/firmware# dmesg | grep -i "virtio0"
[    0.134940] virtio_rpmsg_bus virtio0: rpmsg host is online
```

neuberfran, 2024-03-30:
@jlh-makeen I really need vdev0buffer to appear in Cat Carveout result and you (maybe) need to check the correct mailbox passage (mub or mu?)
[
[image]
[image]
](url)

jlh-makeen, 2024-03-31:
@neuberfran 

So regarding the DT configs, Zephyr needs to use MU_B 0x30ab0000 and Linux needs to use MU_A 0x30aa0000.

Otherwise you get a ressource conflict.

neuberfran, 2024-03-31:
@jlh-makeen Pls. Tell me Why I don't have vdev0buffer in Cat Carveout Cmd Result

neuberfran, 2024-03-31:
> @neuberfran
> 
> Portanto, com relação às configurações do DT, o Zephyr precisa usar MU_B 0x30ab0000 e o Linux precisa usar MU_A 0x30aa0000.
> 
> Caso contrário, você terá um conflito de recursos.

damn!  Am I always doing it wrong using mub on Linux and Zephyr?

And what do you think about my mbox-namex (tx0, rx0, rxdb0 and not tx, rx, rxdb) and my mboxes (mub 0 1, mub 1 1 and mub 3 1)
[image]

jlh-makeen, 2024-03-31:
@neuberfran 

It looks mostly right, your zephyr side was missing a zero in the  reg= <0x9ff00000...

But you know I'm not expert here and I still haven't got my ttyrpmsg to show up.

I would disable the rpsmg node as the rproc is also allocating the mailboxes and virtio rpmsg bus.
Can you try disabling the "rpmsg" node  and double check your addresses in DT ?

neuberfran, 2024-03-31:
@jlh-makeen 
Tks. There was really a 0 missing. It still hasn't been resolved. I already tried disabling rpmsg but the imx_rproc driver gave an error without rpmsg. Unlike you, I'm using mu (for rpmsg) and mub in imx7d-cm4 (Remoteproc). Maybe that's why I couldn't disable rpmsg:
[image]

jlh-makeen, 2024-03-31:
> @jlh-makeen Tks. There was really a 0 missing. It still hasn't been resolved. I already tried disabling rpmsg but the imx_rproc driver gave an error without rpmsg. Unlike you, I'm using mu (for rpmsg) and mub in imx7d-cm4 (Remoteproc). Maybe that's why I couldn't disable rpmsg: And what do you think about my mbox-namex (tx0, rx0, rxdb0 and not tx, rx, rxdb) and my mboxes (mub 0 1, mub 1 1 and mub 3 1)? ![Captura de tela de 2024-03-31 13-43-37](https://private-user-images.githubusercontent.com/38701397/318286232-8c889e98-f96b-4d80-abea-1dbe0cdb78f9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE5MDY3MjcsIm5iZiI6MTcxMTkwNjQyNywicGF0aCI6Ii8zODcwMTM5Ny8zMTgyODYyMzItOGM4ODllOTgtZjk2Yi00ZDgwLWFiZWEtMWRiZTBjZGI3OGY5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzMxVDE3MzM0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU0MjE1MzkzYmU4N2I4ZTY3M2I3NzJjOThmYmZmMzk2OGQ0YTUwZGExOWE1MjliNjU4NTBmNjBhMWQ2NjZhMzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TAegWtQ7yk4NJT6QUKGz1kMRk4bWzwcL4UWPV0hBei4)

I believe this is correct for the mbox definitions

You can try with tx/rx/rxdb instead of rx0 etc.. but I don't think it will make a difference as rhe driver is looking up the prefixes tx/rx/rxdb therefore what's after like rx0 doesn't matter.

neuberfran, 2024-04-01:
@jlh-makeen 

I have some steps that I follow that might reveal the issue of /dev/ttyRPMSG not showing up:

1)  west build --sysbuild -b pico_pi_m4 samples/subsys/ipc/openamp_rsc_table
1.1) sudo cp zephyr_openamp_rsc_table.bin /media/neuberfran/boot16
1.2) sudo cp zephyr_openamp_rsc_table.elf /media/neuberfran/root/lib/firmware/

I don't know if you are also doing step 1.1 (I would like to know) above, that is, copying the *.bin to the FAT/Uboot partition and starting at 0x7f800000.

So, if I give the command: cat /sys/class/remoteproc/remoteproc0/state I receive the attached message

neuberfran, 2024-04-01:
@jlh-makeen 

Since I don't have /dev/ttyRPMSG(just rpmsg_ctrl_0), can I transpose the code below to Python3 and use it to send messages to m4?

```
#include <stdio.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <linux/rpmsg.h>
#include <sys/ioctl.h>
#include <unistd.h>

int main(void)
{
        char data_buf [] = {'a', 'b', 'c', 'd', 'e', '\0'};
        struct rpmsg_endpoint_info ept_info = {"rpmsg-openamp-demo-channel", 0x2, 0x1};
        int fd = open("/dev/rpmsg_ctrl0", O_RDWR);
        /* create endpoint interface */
        ioctl(fd, RPMSG_CREATE_EPT_IOCTL, &ept_info);  // /dev/rpmsg0 is created
        /* create endpoint */
        int fd_ept = open("/dev/rpmsg0", O_RDWR); // backend creates endpoint
        /* receive data from remote device */
        read(fd_ept, &data_buf, sizeof(data_buf));
        /* send data to remote device */
        write(fd_ept, &data_buf, sizeof(data_buf));
        /* destroy endpoint */
        ioctl(fd_ept, RPMSG_DESTROY_EPT_IOCTL);
        close(fd_ept);
        close(fd);
}
```
Code source:  https://developer.toradex.com/software/real-time/freertos/freertos-on-the-cortex-m4-of-a-colibri-imx7/

jlh-makeen, 2024-04-03:
> @jlh-makeen
>
> Since I don't have /dev/ttyRPMSG(just rpmsg_ctrl_0), can I transpose the code below to Python3 and use it to send messages to m4?
>
> ```
> #include <stdio.h>
> #include <sys/types.h>
> #include [...]

I believe you can transpose it to Python :)

Try to ask GPT to give you the code and post the resultd here I'm interested to see what you can do :)

I have an issue starting the M4 from uboot as it prevents my Linux from starting at this point. So more investigation is necessary on my side.

neuberfran, 2024-04-03:
@jlh-makeen 

In reality there are three things: First: In the code above, what is the line below for? (since I don't have it: rpmsg-openamp-demo-channe And /dev/ttyRPMSG)

    `    struct rpmsg_endpoint_info ept_info = {"rpmsg-openamp-demo-channel", 0x2, 0x1};`

Second: I still don't know why vdev0buffer doesn't appear in my case here, when I command cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories

Third: Technexion has Yocto mickledore in the very early testing phase, I've already tried it on two computers here, but the Yocto build always gives an error:
https://github.com/TechNexion/tn-imx-yocto-manifest/issues/40

```
ERROR: linux-tn-imx-6.1.55+gitAUTOINC+4ce21f3007-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/home/neuberfran/edm_yocto/build-wayland-pico-imx7/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/neuberfran/edm_yocto/sources/poky/scripts:/home/neuberfran/edm_yocto/build-wayland-pico-imx7/tmp/work/pico_imx7-poky-linux-gnueabi/linux-tn-imx/6.1.55+gitAUTOINC+4ce21f3007-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi:/home/neuberfran/edm_yocto/build-wayland-pico-imx7/tmp/work/pico_imx7-poky-linux-gnueabi/linux-tn-imx/6.1.55+gitAUTOINC+4ce21f3007-r0/recipe-sysroot/usr/bin/crossscripts:/home/neuberfran/edm_yocto/build-wayland-pico-imx7/tmp/work/pico_imx7-poky-linux-gnueabi/linux-tn-imx/6.1.55+gitAUTOINC+4ce21f3007-r0/recipe-sysroot-native/usr/sbin:/home/neuberfran/edm_yocto/build-wayland-pico-imx7/tmp/work/pico_imx7-poky-linux-gnueabi/linux-tn-imx/6.1.55+gitAUTOINC+4ce21f3007-r0/recipe-sysroot-native/usr/bin:/home/neuberfran/edm_yocto/build-wayland-pico-imx7/tmp/work/pico_imx7-poky-linux-gnueabi/linux-tn-imx/6.1.55+gitAUTOINC+4ce21f3007-r0/recipe-sysroot-native/sbin:/home/neuberfran/edm_yocto/build-wayland-pico-imx7/tmp/work/pico_imx7-poky-linux-gnueabi/linux-tn-imx/6.1.55+gitAUTOINC+4ce21f3007-r0/recipe-sysroot-native/bin:/home/neuberfran/edm_yocto/sources/poky/bitbake/bin:/home/neuberfran/edm_yocto/build-wayland-pico-imx7/tmp/hosttools"; export HOME="/home/neuberfran"; LANG=C git -c gc.autoDetach=false -c core.pager=cat clone --bare --mirror https://github.com/TechNexion/linux-tn-imx.git /home/neuberfran/edm_yocto/downloads//git2/github.com.TechNexion.linux-tn-imx.git --progress failed with exit code 128, no output
ERROR: linux-tn-imx-6.1.55+gitAUTOINC+4ce21f3007-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/TechNexion/linux-tn-imx.git;protocol=https;branch=tn-imx_6.1.55_2.2.0-next')
ERROR: Logfile of failure stored in: /home/neuberfran/edm_yocto/build-wayland-pico-imx7/tmp/work/pico_imx7-poky-linux-gnueabi/linux-tn-imx/6.1.55+gitAUTOINC+4ce21f3007-r0/temp/log.do_fetch.22758
ERROR: Task (/home/neuberfran/edm_yocto/sources/meta-tn-imx-bsp/recipes-kernel/linux/linux-tn-imx_6.1.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1579 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/neuberfran/edm_yocto/sources/meta-tn-imx-bsp/recipes-kernel/linux/linux-tn-imx_6.1.bb:do_fetch
```
[image]

jlh-makeen, 2024-04-04:
@neuberfran 

1- So this line defines a rpmsg driver with the topic/endpoint : "rpmsg-openamp-demo-channel".

This should probe the imx-tty-rpmsg driver or the rpmsg sample client driver as it's part of the endpoint names.

The 0x2/0x1 parameters I'm not sure what they are used for, I'll look up the documentation at some point.

Normally, the RPMSG Name Service should receive the announcement message from the M4 zephyr program and probe the appropriate driver (tty-rpmsg and rpmsg-sample-client).

Here when you use the /dev/rpmsg_ctrl_0 device, you are explicitly asking the kernel to probe the imx-tty-rpmsg or sample-client-rpmsg drivers and therefore have /dev/ttyrpmsg0 to show up.

2- I do not have an explaination for this fact to be honest... It looked like it was defined correctly in your DT

3- I'm not sure I can help with this one unfortunately

neuberfran, 2024-04-05:
@jlh-makeen 
https://gist.github.com/neuberfran/2cddda9e87582484a7090bfcbc12d93e

Still investigating here why vdev0buffer does not appear in cat .../caverout_memories, I have questions:

First:
Do you have the device tree node called ocram in m4 part?

```
&ocram {
     reg = <0x00901000 0xf000>;
 };
```
Why do you use 0x901000 and not 0x900000 like me?

```
 cm4sramcode: cm4sramcode@0x900000 {
             compatible = "shared-dma-pool";
             reg = <0x00900000 0x40000>;
            no-map;
         };
```
https://www.youtube.com/watch?v=_18E5te_sIw&t=13s
[image]

neuberfran, 2024-04-06:
@jlh-makeen I put 64 (egual u) and Not 16 in zephyr_remoteproc.elf And not I have vdevbuffer in carveout_memories result.

neuberfran, 2024-04-07:
@jlh-makeen Pls. Help-me to interpreter my issue in main.c code (by openamp_rsc_table Project:

Note: If I don't comment on the line **// k_sem_take(&data_tty_sem, K_FOREVER);** it doesn't even pass printk("\r\nOpenAMP[remote] Linux tty **ccc1.2** started\r\n");

```
/*
 * Copyright (c) 2020, STMICROELECTRONICS
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <string.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include <zephyr/drivers/ipm.h>

#include <openamp/open_amp.h>
#include <metal/device.h>
#include <resource_table.h>

#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(openamp_rsc_table, LOG_LEVEL_DBG);

#define SHM_DEVICE_NAME	"shm"

#if !DT_HAS_CHOSEN(zephyr_ipc_shm)
#error "Sample requires definition of shared memory for rpmsg"
#endif

/* Constants derived from device tree */
#define SHM_NODE		DT_CHOSEN(zephyr_ipc_shm)
#define SHM_START_ADDR	DT_REG_ADDR(SHM_NODE)
#define SHM_SIZE		DT_REG_SIZE(SHM_NODE)

#define APP_TASK_STACK_SIZE (1024)

/* Add 1024 extra bytes for the TTY task stack for the "tx_buff" buffer. */
#define APP_TTY_TASK_STACK_SIZE (1536)

K_THREAD_STACK_DEFINE(thread_mng_stack, APP_TASK_STACK_SIZE);
K_THREAD_STACK_DEFINE(thread_rp__client_stack, APP_TASK_STACK_SIZE);
K_THREAD_STACK_DEFINE(thread_tty_stack, APP_TTY_TASK_STACK_SIZE);

static struct k_thread thread_mng_data;
static struct k_thread thread_rp__client_data;
static struct k_thread thread_tty_data;

static const struct device *const ipm_handle =
	DEVICE_DT_GET(DT_CHOSEN(zephyr_ipc));

static metal_phys_addr_t shm_physmap = SHM_START_ADDR;

struct metal_device shm_device = {
	.name = SHM_DEVICE_NAME,
	.num_regions = 2,
	.regions = {
		{.virt = NULL}, /* shared memory */
		{.virt = NULL}, /* rsc_table memory */
	},
	.node = { NULL },
	.irq_num = 0,
	.irq_info = NULL
};

struct rpmsg_rcv_msg {
	void *data;
[... 361 lines cut ...]
                                                                                
OpenAMP[remote] Linux tty ccc1.1 started                                        
                                                                                
OpenAMP[remote] Linux tty ccc1.2 started                                        
                                                                                
OpenAMP[remote] Linux tty ccc3 started                                          
                                                                                
OpenAMP[remote] Linux tty ccc1.1 started                                        
                                                                                
OpenAMP[remote] Linux tty ccc1.2 started                                        
                                                                                
OpenAMP[remote] Linux tty ccc3 started                                          
                                                                                
OpenAMP[remote] Linux tty ccc1.1 started                                        
                                                                                
OpenAMP[remote] Linux tty ccc1.2 started                                        
                                                                                
OpenAMP[remote] Linux tty ccc3 started                                          
                                                                                
OpenAMP[remote] Linux tty ccc1.1 started                                        
                                                                                
OpenAMP[remote] Linux tty ccc1.2 started                                        
                                                                                
OpenAMP[remote] Linux tty ccc3 started   
```

jlh-makeen, 2024-04-07:
> @jlh-makeen Pls. Help-me to interpreter my issue in main.c code (by openamp_rsc_table Project:
>
> Note: If I don't comment on the line **// k_sem_take(&data_tty_sem, K_FOREVER);** it doesn't even pass [...]

Seems normal to me, the semaphore tty_sem will only pass when data is received from Linux dev/ttyRPMSG0 to Zephyr.

We used 0x901000 because for some reason 0x1000 bytes were reserved for our Linux (probably necessary for some device or drivers). That was given by my colleague who's responsible for the Linux side.

He said he had an interesting breakthrough last week where he could get dev/ttyrpmsg0 to show up and communication was working, but apparently there were memory addresses issues and mbox ids mismatching between Zephyr and Linux. 

I'll know more about it next week.

neuberfran, 2024-04-07:
@jlh-makeen 
Makes a lot of sense. Thanks. But I have another mystery
I stay using vdevbuffer, vdevvring0, vdevvring1(Pic). But **cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories** command result is: vdev0buffer, vdev0vring0, vdev0vring1. 
[image]
```
imx7d-cm4 {
         compatible = "fsl,imx7d-cm4";
         clocks = <&clks IMX7D_ARM_M4_ROOT_CLK>;
//         fsl,auto-boot;
         mbox-names = "tx0", "rx0", "rxdb0";
		 mboxes = <&mu0a 0 1
			  &mu0a 1 1
			  &mu0a 3 1>;
  //	     memory-region = <&rpmsg_vrings>, <&cm4tcmcode>, <&cm4sramcode>;
        memory-region = <&vdevbuffer>, <&vdevvring0>, <&vdevvring1>, <&cm4tcmcode>;
        // <&cm4sramcode>; 
        fsl,rproc-auto-boot = <1>;
        fsl,rproc-fw-name="zephyr_openamp_rsc_table.elf";
        syscon = <&src>;
        status = "okay"; 
     };
```

```
root@scorpion:/lib/firmware# cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories
Carveout memory entry:
	Name: vdevbuffer
	Virtual address: 9e80f0da
	DMA address: 0x00902000
	Device address: 0x902000
	Length: 0xfe000 Bytes

Carveout memory entry:
	Name: vdev0vring0
	Virtual address: 8ce027f2
	DMA address: 0x00900000
	Device address: 0x900000
	Length: 0x1000 Bytes

Carveout memory entry:
	Name: vdev0vring1
	Virtual address: 053d417f
	DMA address: 0x00901000
	Device address: 0x901000
	Length: 0x1000 Bytes

Carveout memory entry:
	Name: cm4tcmcode
	Virtual address: 26abf5d5
	DMA address: 0x007f8000
	Device address: 0x7f8000
	Length: 0x8000 Bytes

root@scorpion:/lib/firmware# ls /sys/bus/virtio/devices/
virtio0
root@scorpion:/lib/firmware# ls  /sys/bus/rpmsg/devices
virtio0.rpmsg_ctrl.0.0	virtio0.rpmsg_ns.53.53
root@scorpion:/lib/firmware# dmesg | grep -i "virtio"
[   33.352854] virtio_rpmsg_bus virtio0: rpmsg host is online
[   33.359978]  remoteproc0#vdev0buffer: registered virtio0 (type 7)
root@scorpion:/lib/firmware# dmesg | grep -i -E "(rproc|rpmsg)"
[    0.084748] imx rpmsg driver is registered.
[    2.178748] remoteproc remoteproc0: imx-rproc is available
[    2.184307] remoteproc remoteproc0: attaching to imx-rproc
[    2.189942] remoteproc remoteproc0: remote processor imx-rproc is now attached
[   33.157165] remoteproc remoteproc0: stopped remote processor imx-rproc
[   33.203752] remoteproc remoteproc0: powering up imx-rproc
[   33.346460] imx-rproc imx7d-cm4: imx_rproc_kick: failed (0, err:-62)
[   33.352854] virtio_rpmsg_bus virtio0: rpmsg host is online
[   33.366383] remoteproc remoteproc0: remote processor imx-rproc is now up
```

neuberfran, 2024-04-07:
@jlh-makeen Mistery

When I change vdevbuffer, vdevvring0, vdevvring1, cm4tcmcode, cm4sramcode to 
vdev0buffer, vdev0vring0, vdev0vring1, cm4tcmcode, cm4sramcode I can`t see vdev0buffer in  **cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories** result. 

[image]

```
imx7d-cm4 {
         compatible = "fsl,imx7d-cm4";
         clocks = <&clks IMX7D_ARM_M4_ROOT_CLK>;
//         fsl,auto-boot;
         mbox-names = "tx0", "rx0", "rxdb0";
		 mboxes = <&mu0a 0 1
			  &mu0a 1 1
			  &mu0a 3 1>;
  //	     memory-region = <&rpmsg_vrings>, <&cm4tcmcode>, <&cm4sramcode>;
        memory-region = <&vdev0buffer>, <&vdev0vring0>, <&vdev0vring1>, <&cm4tcmcode>, <&cm4sramcode>;
        // <&cm4sramcode>; 
        fsl,rproc-auto-boot = <1>;
        fsl,rproc-fw-name="zephyr_openamp_rsc_table.elf";
        syscon = <&src>;
        status = "okay"; 
     };
vdev0vring0: vdev0vring0@900000 {
            compatible = "shared-dma-pool";
            reg = <0x900000 0x1000>;
            no-map;
        };

        vdev0vring1: vdev0vring1@901000 {
            compatible = "shared-dma-pool";
            reg = <0x901000 0x1000>;
            no-map;
        };

        vdev0buffer: vdev0buffer@902000 {
            compatible = "shared-dma-pool";
			reg = <0x902000 0xfe000>;
            no-map;
        };
 
        cm4tcmcode: cm4tcmcode@7f8000 {
             compatible = "shared-dma-pool";
             reg = <0x007f8000 0x8000>;
             no-map;
         };
 
         cm4sramcode: cm4sramcode@0x900000 {
             compatible = "shared-dma-pool";
             reg = <0x00900000 0x40000>;
            no-map;
         };
root@scorpion:/lib/firmware# 
root@scorpion:/lib/firmware# cat /sys/kernel/debug/remoteproc/remoteproc0/carveout_memories
Carveout memory entry:
	Name: vdev0vring0
	Virtual address: 6899ed74
	DMA address: 0x00900000
	Device address: 0x900000
	Length: 0x1000 Bytes

Carveout memory entry:
	Name: vdev0vring1
	Virtual address: 7421fba4
	DMA address: 0x00901000
	Device address: 0x901000
	Length: 0x1000 Bytes

Carveout memory entry:
	Name: cm4tcmcode
	Virtual address: 2beae070
	DMA address: 0x007f8000
	Device address: 0x7f8000
	Length: 0x8000 Bytes

Carveout memory entry:
	Name: cm4sramcode
	Virtual address: dd673f21
	DMA address: 0x00900000
	Device address: 0x900000
	Length: 0x40000 Bytes
```

jlh-makeen, 2024-04-08:
@neuberfran 

We have found out that you need to use mbox 0 :

 mboxes = <&mu0a 0 0
	            &mu0a 1 0
		    &mu0a 3 0>;

neuberfran, 2024-04-08:
@jlh-makeen 
https://lwn.net/Articles/815905/

I can't put 5.16 here. Please ask your colleague to make a patch so I can use rpmsg-tty here

jlh-makeen, 2024-04-09:
> @jlh-makeen https://lwn.net/Articles/815905/
> 
> I can't put 5.16 here. Please ask your colleague to make a patch so I can use rpmsg-tty here

I'm not sure it will work if you cannot get a 5.16 or newer kernel.

I'll know more on Thursday.

jlh-makeen, 2024-04-12:
@neuberfran 

I have spent the day with my colleague working on Zephyr/Linux with RPMSG.
We got some good news where we finally managed to get Zephyr to get loaded by rproc and rpmsg to show up WITHOUT any kernel modifications ! 

We have therefore reverted all our patches in the Linux Kernel and only properly defined the correct nodes in Linux and Zephyr to get the system working.

One important note though, Zephyr is not correctly configured to place the boot_vector (aka SP and PC) into the OCRAM_S region.

Therefore if your Zephyr DT is not using the &ocram_s_code as the zephyr,flash node, your M4 core will not boot your application.

See here in the FREERTOS linker script provided by Toradex on the M4 examples for IMX7D:
https://github.com/toradex/FreeRTOS-Colibri-iMX7/blob/master/platform/devices/MCIMX7D/linker/gcc/MCIMX7D_M4_tcm.ld#L102

the complete interrupt vector table is placed in the OCRAM_S which represents approx 0x240 bytes.

The rest of the code can be located anywhere in the TCM/OCRAM/OCRAM_S/DDR as per your application needs.
The only limitation is that you MUST have at least the first 8 bytes of the rom_start region of the zephyr.elf file placed into the OCRAM_S (address 0x00180000).

The problem is that Zephyr places the rom_start region (containing the SP/PC and vector table) at the start of the ROMABLE_REGION (aka FLASH, aka CODE read only region).

This is incorrect as it prevents the M4 core to boot as per the AN5317 from NXP which clearly states you must load the SP/PC into the OCRAM_S to boot the M4 correctly.

At first we did that by modifying the elf loader of Linux RPROC driver. But that was a hack and not very nice.

But it turns out we can modify the include/zephyr/arch/arm/cortex_m/scripts/linker.ld to place the rom_start region in the OCRAM_S:

```

...

MEMORY
    {
	BOOTVECTOR (rx) : ORIGIN = 0x00180000, LENGTH = 32K
    FLASH (rx) : ORIGIN = ROM_ADDR, LENGTH = ROM_SIZE
    RAM   (wx) : ORIGIN = RAM_ADDR, LENGTH = RAM_SIZE
#if defined(CONFIG_LINKER_DEVNULL_MEMORY)
    DEVNULL_ROM (rx) : ORIGIN = DEVNULL_ADDR, LENGTH = DEVNULL_SIZE
#endif
    LINKER_DT_REGIONS()
    /* Used by and documented in include/linker/intlist.ld */
    IDT_LIST (wx) : ORIGIN = 0xFFFF7FFF, LENGTH = 32K
    }

ENTRY(CONFIG_KERNEL_ENTRY)

SECTIONS
    {

#include <zephyr/linker/rel-sections.ld>

    /*
     * .plt and .iplt are here according to 'arm-zephyr-elf-ld --verbose',
     * before text section.
     */
    /DISCARD/ :
	{
	*(.plt)
	}

    /DISCARD/ :
	{
	*(.iplt)
	}

	SECTION_PROLOGUE(rom_start,,)
	{

/* Located in generated directory. This file is populated by calling
 * zephyr_linker_sources(ROM_START ...). This typically contains the vector
 * table and debug information.
 */
#include <snippets-rom-start.ld>

	} GROUP_LINK_IN(BOOTVECTOR )

	GROUP_START(ROMABLE_REGION)

	__rom_region_start = ROM_ADDR;

....
```

I plan to do this nicely with config flags in order to not disturb other cortex_m devices and keep this configuration selectable.

This will generate a valid elf file that will correctly load into Linux using RPROC and show up the RPMSG if you have correctly used the mu0a and channel 0 as I mentioned before.

Either you use the OCRAM_S as your CODE partitions for zephyr,flash or you need to adjust the linker script to always place the irq vectors in the OCRAM_S. I hope this is clear.

More information and details to come later. I hope this helps you understand the issues :)

neuberfran, 2024-04-12:
@jlh-makeen 
You said:
_"Either you use the OCRAM_S as your CODE partitions for zephyr,flash or you need to adjust the linker script to always place the irq vectors in the OCRAM_S. I hope this is clear."_

Unfortunately, it is not yet clear. Which region should I map to vdevbuffer, vring0 and vring1?
[image]

```
reserved-memory {
         #address-cells = <1>;
         #size-cells = <1>;
         ranges;

        vdev0vring0: vdev0vring0@900000 {
            compatible = "shared-dma-pool";
            reg = <0x900000 0x1000>;
            no-map;
        };

        vdev0vring1: vdev0vring1@901000 {
            compatible = "shared-dma-pool";
            reg = <0x901000 0x1000>;
            no-map;
        };

        vdevbuffer: vdevbuffer@902000 {
            compatible = "shared-dma-pool";
			reg = <0x902000 0xfe000>;
            no-map;
        };
 
        cm4tcmcode: cm4tcmcode@7f8000 {
             compatible = "shared-dma-pool";
             reg = <0x007f8000 0x8000>;
             no-map;
         };
 
         cm4sramcode: cm4sramcode@0x900000 {
             compatible = "shared-dma-pool";
             reg = <0x00900000 0x40000>;
            no-map;
         };
    };       

     imx7d-cm4 {
         compatible = "fsl,imx7d-cm4";
         clocks = <&clks IMX7D_ARM_M4_ROOT_CLK>;
         mbox-names = "tx", "rx", "rxdb";
		 mboxes = <&mu0a 0 0
			  &mu0a 1 0
			  &mu0a 3 0>;
        memory-region = <&vdevbuffer>, <&vdev0vring0>, <&vdev0vring1>, <&cm4tcmcode>, <&cm4sramcode>;
        fsl,rproc-auto-boot = <1>;
        fsl,rproc-fw-name="zephyr_openamp_rsc_table.elf";
        syscon = <&src>;
        status = "okay"; 
     };
```

jlh-makeen, 2024-04-12:
Try to modify the Zephyr Linker script instead like I mentionned previously, it is important that the rproc elf loader is copying the boot vector into OCRAM_S.

jlh-makeen, 2024-04-12:
> @jlh-makeen Pls, give-me your **main.c** from openamp_rsc_table. I think there are differences between your main.c and mine. I want to keep them both the same.

You can run the default openamp_rsc_table sample without modifications.

It is only necessary to provide the Zephyr ipc and shm nodes with the same setting as your Linux side, I advise that you use DDR for the shared memory vrings/vdevbuffer and zephyr side shm, in order to have symetric setup in zephyr/linux (the addresses will be the same). 

When the openamp_rsc_table sample starts up, you should see rpmsg virtio bus ns-announcements, even if you don't have tty-rpmsg driver from Linux 5.16. 

If you have that, you'll be ready for rpmsg :)

neuberfran, 2024-04-12:
> -rom-start.ld

I need change my /home/neuberfran/zephyrproject/zephyr/include/zephyr/arch/arm/cortex_m/scripts/linker.ld from this:
```

/*
 * Copyright (c) 2013-2014 Wind River Systems, Inc.
 *
 * SPDX-License-Identifier: Apache-2.0
 */

/**
 * @file
 * @brief Linker command/script file
 *
 * Linker script for the Cortex-M platforms.
 */

#include <zephyr/linker/sections.h>
#include <zephyr/devicetree.h>

#include <zephyr/linker/devicetree_regions.h>
#include <zephyr/linker/linker-defs.h>
#include <zephyr/linker/linker-tool.h>

/* physical address of RAM */
#ifdef CONFIG_XIP
#define ROMABLE_REGION FLASH
#else
#define ROMABLE_REGION RAM
#endif
#define RAMABLE_REGION RAM

#if !defined(CONFIG_XIP) && (CONFIG_FLASH_SIZE == 0)
#define ROM_ADDR RAM_ADDR
#else
#define ROM_ADDR (CONFIG_FLASH_BASE_ADDRESS + CONFIG_FLASH_LOAD_OFFSET)
#endif

#if CONFIG_FLASH_LOAD_SIZE > 0
#define ROM_SIZE CONFIG_FLASH_LOAD_SIZE
#else
#define ROM_SIZE (CONFIG_FLASH_SIZE*1K - CONFIG_FLASH_LOAD_OFFSET)
#endif

#if defined(CONFIG_XIP)
#if defined(CONFIG_IS_BOOTLOADER)
#define RAM_SIZE (CONFIG_BOOTLOADER_SRAM_SIZE * 1K)
#define RAM_ADDR (CONFIG_SRAM_BASE_ADDRESS + \
	(CONFIG_SRAM_SIZE * 1K - RAM_SIZE))
#else
#define RAM_SIZE (CONFIG_SRAM_SIZE * 1K)
#define RAM_ADDR CONFIG_SRAM_BASE_ADDRESS
#endif
#else
#define RAM_SIZE (CONFIG_SRAM_SIZE * 1K - CONFIG_BOOTLOADER_SRAM_SIZE * 1K)
#define RAM_ADDR CONFIG_SRAM_BASE_ADDRESS
#endif

#if defined(CONFIG_CUSTOM_SECTION_ALIGN)
_region_min_align = CONFIG_CUSTOM_SECTION_MIN_ALIGN_SIZE;
#else
/* Set alignment to CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE
 * to make linker section alignment comply with MPU granularity.
 */
#if defined(CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE)
_region_min_align = CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE;
#else
/* If building without MPU support, use default 4-byte alignment. */
_region_min_align = 4;
[... 380 lines cut ...]
	{
	KEEP(*(.ARM.attributes))
	KEEP(*(.gnu.attributes))
	}

    /* Sections generated from 'zephyr,memory-region' nodes */
    LINKER_DT_SECTIONS()

/* Must be last in romable region */
SECTION_PROLOGUE(.last_section,,)
{
#ifdef CONFIG_LINKER_LAST_SECTION_ID
  /* Fill last section with a word to ensure location counter and actual rom
   * region data usage match. */
  LONG(CONFIG_LINKER_LAST_SECTION_ID_PATTERN)
#endif
} GROUP_LINK_IN(ROMABLE_REGION)

/* To provide the image size as a const expression,
 * calculate this value here. */
_flash_used = LOADADDR(.last_section) + SIZEOF(.last_section) - __rom_region_start;

    }
```
**to??????????**

jlh-makeen, 2024-04-12:
As I explained before to this:
```

/*
 * Copyright (c) 2013-2014 Wind River Systems, Inc.
 *
 * SPDX-License-Identifier: Apache-2.0
 */

/**
 * @file
 * @brief Linker command/script file
 *
 * Linker script for the Cortex-M platforms.
 */

#include <zephyr/linker/sections.h>
#include <zephyr/devicetree.h>

#include <zephyr/linker/devicetree_regions.h>
#include <zephyr/linker/linker-defs.h>
#include <zephyr/linker/linker-tool.h>

/* physical address of RAM */
#ifdef CONFIG_XIP
#define ROMABLE_REGION FLASH
#else
#define ROMABLE_REGION RAM
#endif
#define RAMABLE_REGION RAM

#if !defined(CONFIG_XIP) && (CONFIG_FLASH_SIZE == 0)
#define ROM_ADDR RAM_ADDR
#else
#define ROM_ADDR (CONFIG_FLASH_BASE_ADDRESS + CONFIG_FLASH_LOAD_OFFSET)
#endif

#if CONFIG_FLASH_LOAD_SIZE > 0
#define ROM_SIZE CONFIG_FLASH_LOAD_SIZE
#else
#define ROM_SIZE (CONFIG_FLASH_SIZE*1K - CONFIG_FLASH_LOAD_OFFSET)
#endif

#if defined(CONFIG_XIP)
#if defined(CONFIG_IS_BOOTLOADER)
#define RAM_SIZE (CONFIG_BOOTLOADER_SRAM_SIZE * 1K)
#define RAM_ADDR (CONFIG_SRAM_BASE_ADDRESS + \
	(CONFIG_SRAM_SIZE * 1K - RAM_SIZE))
#else
#define RAM_SIZE (CONFIG_SRAM_SIZE * 1K)
#define RAM_ADDR CONFIG_SRAM_BASE_ADDRESS
#endif
#else
#define RAM_SIZE (CONFIG_SRAM_SIZE * 1K - CONFIG_BOOTLOADER_SRAM_SIZE * 1K)
#define RAM_ADDR CONFIG_SRAM_BASE_ADDRESS
#endif

#if defined(CONFIG_CUSTOM_SECTION_ALIGN)
_region_min_align = CONFIG_CUSTOM_SECTION_MIN_ALIGN_SIZE;
#else
/* Set alignment to CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE
 * to make linker section alignment comply with MPU granularity.
 */
#if defined(CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE)
_region_min_align = CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE;
#else
/* If building without MPU support, use default 4-byte alignment. */
_region_min_align = 4;
#endif
#endif
[... 378 lines cut ...]
    SECTION_PROLOGUE(.ARM.attributes, 0,)
	{
	KEEP(*(.ARM.attributes))
	KEEP(*(.gnu.attributes))
	}

    /* Sections generated from 'zephyr,memory-region' nodes */
    LINKER_DT_SECTIONS()

/* Must be last in romable region */
SECTION_PROLOGUE(.last_section,,)
{
#ifdef CONFIG_LINKER_LAST_SECTION_ID
  /* Fill last section with a word to ensure location counter and actual rom
   * region data usage match. */
  LONG(CONFIG_LINKER_LAST_SECTION_ID_PATTERN)
#endif
} GROUP_LINK_IN(ROMABLE_REGION)

/* To provide the image size as a const expression,
 * calculate this value here. */
_flash_used = LOADADDR(.last_section) + SIZEOF(.last_section) - __rom_region_start;

    }
```

neuberfran, 2024-04-12:
@jlh-makeen Unfortunately, it hasn't been resolved yet. The *.bin file generated after the changes did not fit in the Uboot partition (I had to remove it to load Linux). According to the commands in the photo, I still have virtio0 in the directories, but I still don't have /dev/ttyRPMSG
[image]
```

pwd
/home/neuberfran/zephyrproject/zephyr/build/openamp_rsc_table/zephyr
neuberfran@ubuntu2204:~/zephyrproject/zephyr/build/openamp_rsc_table/zephyr$ ls -lha -S
total 515M
-rwxrwxr-x  1 neuberfran neuberfran 511M abr 12 12:42 zephyr_openamp_rsc_table.bin
-rw-rw-r--  1 neuberfran neuberfran 655K abr 12 12:42 edt.pickle
-rwxrwxr-x  1 neuberfran neuberfran 647K abr 12 12:42 zephyr_pre0.elf
-rwxrwxr-x  1 neuberfran neuberfran 647K abr 12 12:42 zephyr_openamp_rsc_table.elf
-rw-rw-r--  1 neuberfran neuberfran 412K abr 12 12:42 libzephyr.a
-rw-rw-r--  1 neuberfran neuberfran 319K abr 12 12:42 zephyr_pre0.map
-rw-rw-r--  1 neuberfran neuberfran 318K abr 12 12:42 zephyr_final.map
-rw-rw-r--  1 neuberfran neuberfran 318K abr 12 12:42 zephyr_openamp_rsc_table.map
-rw-rw-r--  1 neuberfran neuberfran 282K abr 12 12:42 dts.cmake
-rw-rw-r--  1 neuberfran neuberfran 188K abr 12 12:42 zephyr.dts.pre
-rw-rw-r--  1 neuberfran neuberfran  42K abr 12 12:42 zephyr.dts
-rw-rw-r--  1 neuberfran neuberfran  39K abr 12 12:42 .config
-rw-rw-r--  1 neuberfran neuberfran  14K abr 12 12:42 cmake_install.cmake
-rw-rw-r--  1 neuberfran neuberfran  13K abr 12 12:42 linker_zephyr_pre0.cmd
-rw-rw-r--  1 neuberfran neuberfran  13K abr 12 12:42 linker.cmd
-rw-rw-r--  1 neuberfran neuberfran  11K abr 12 12:42 isr_tables.c
-rw-rw-r--  1 neuberfran neuberfran 6,0K abr 12 12:42 linker_zephyr_pre0.cmd.dep
-rw-rw-r--  1 neuberfran neuberfran 6,0K abr 12 12:42 linker.cmd.dep
-rw-rw-r--  1 neuberfran neuberfran 4,7K abr 12 12:42 zephyr_openamp_rsc_table.stat
drwxrwxr-x 14 neuberfran neuberfran 4,0K abr 12 12:42 .
drwxrwxr-x  7 neuberfran neuberfran 4,0K abr 12 12:42 ..
drwxrwxr-x  5 neuberfran neuberfran 4,0K abr 12 12:42 arch
drwxrwxr-x  3 neuberfran neuberfran 4,0K abr 12 12:42 boards
drwxrwxr-x  5 neuberfran neuberfran 4,0K abr 12 12:42 cmake
drwxrwxr-x  6 neuberfran neuberfran 4,0K abr 12 12:42 CMakeFiles
drwxrwxr-x 15 neuberfran neuberfran 4,0K abr 12 12:42 drivers
drwxrwxr-x  3 neuberfran neuberfran 4,0K abr 12 12:42 include
drwxrwxr-x  3 neuberfran neuberfran 4,0K abr 12 12:42 kconfig
drwxrwxr-x  3 neuberfran neuberfran 4,0K abr 12 12:42 kernel
drwxrwxr-x  9 neuberfran neuberfran 4,0K abr 12 12:42 lib
drwxrwxr-x  3 neuberfran neuberfran 4,0K abr 12 12:42 misc
drwxrwxr-x  4 neuberfran neuberfran 4,0K abr 12 12:42 soc
drwxrwxr-x 22 neuberfran neuberfran 4,0K abr 12 12:42 subsys
-rw-rw-r--  1 neuberfran neuberfran  922 abr 12 12:42 zephyr.dts.d
-rw-rw-r--  1 neuberfran neuberfran  801 abr 12 12:42 snippets_generated.cmake
-rw-rw-r--  1 neuberfran neuberfran  110 abr 12 12:42 .config.sysbuild
-rw-rw-r--  1 neuberfran neuberfran   96 abr 12 12:42 .cmake.dotconfig.checksum
-rwxrwxr-x  1 neuberfran neuberfran   40 abr 12 12:42 isrList.bin
```

neuberfran, 2024-04-18:
@jlh-makeen I'm already suspecting that my vrings may not be 0x1000 but rather 0x8000. So I don't know if I have enough (remaining) memory for the vdevbuffer and what the vdevbuffer would look like. Could you tell me?
https://community.toradex.com/t/verdin-mini-rpmsg-lite-increase-buffer-size/18356/5
https://community.nxp.com/t5/i-MX-Processors/Changing-RPMsg-Buffer-Size/m-p/1221466/highlight/true#M168448

jlh-makeen, 2024-04-18:
> @jlh-makeen I'm already suspecting that my vrings may not be 0x1000 but rather 0x8000. So I don't know if I have enough (remaining) memory for the vdevbuffer and what the vdevbuffer would look like. [...]

@neuberfran 

In order to get rpmsg to work, we had to update to kernel 5.16 to have the official tty-rpmsg and rpmsgctrl drivers.
Then in the DT, we have defined 0x2000 size for the VRINGs and 0x8000 for the vdevbuffer all in DDR memory.

When rproc cannot allocate the memory you request (512 bytes * CONFIG_OPENAMP_RSC_TABLE_NUM_RPMSG_BUFF)

For 0x8000 you can go up to CONFIG_OPENAMP_RSC_TABLE_NUM_RPMSG_BUFF=32 buffers
You will get error -62 when kicking the virtio when remoteproc.
If you get -62 you need to reduce the number of buffers or increase the memory size for the vdevbuffer.

Our particular setup:
```
Entry 0 is of type vdev
  ID 7
  Notify ID 0
  Device features 0x1
  Guest features 0x1
  Config length 0x0
  Status 0x7
  Number of vrings 2
  Reserved (should be zero) [0][0]

  Vring 0
    Device Address 0xbfff0000
    Alignment 16
    Number of buffers 32
    Notify ID 0
    Physical Address 0x0

  Vring 1
    Device Address 0xbfff1000
    Alignment 16
    Number of buffers 32
    Notify ID 1
    Physical Address 0x0
```

When our app starts we created 3 tty endpoints:
```
[ 8883.780211] remoteproc remoteproc0: powering up imx-rproc
[ 8883.807068] remoteproc remoteproc0: Booting fw image zephyr_pnet_ipm.elf, size 2223228
[ 8883.814115] remoteproc remoteproc0: no SP or PC loaded into M4 bootrom vector. Core will not start!
[ 8883.823367]  remoteproc0#vdev0buffer: assigned reserved memory node vdev0buffer0@bfff2000
[ 8883.835930]  remoteproc0#vdev0buffer: registered virtio0 (type 7)
[ 8883.849667] remoteproc remoteproc0: remote processor imx-rproc is now up
[ 8883.862012] virtio_rpmsg_bus virtio0: rpmsg host is online
[ 8883.862321] virtio_rpmsg_bus virtio0: creating channel rpmsg-tty addr 0x400
[ 8883.880476] virtio_rpmsg_bus virtio0: creating channel rpmsg-tty addr 0x401
[ 8883.892294] virtio_rpmsg_bus virtio0: creating channel rpmsg-tty addr 0x402
```

In our case we needed the linker script to start the M4 boot-vector as I mention in this discussion:

https://github.com/zephyrproject-rtos/zephyr/discussions/71617

I'll provide a PR for the bootvector when I have time.

Regarding the if your kernel doesn't have the rpmsg-tty driver but instead the:
when creating the endpoint for imx-rpmsg-tty driver you need the endpoint name: "rpmsg-virtual-tty-channel"

You also need to use mbox id 0 as mentionned before in the Linux DT.

neuberfran, 2024-04-21:
@jlh-makeen 

After I chage zephyr,flash  (to)= &ocram_code; (from  &tcml_code) 
messages from zephyr_openamp_rsc_table.bin no longer appear on the zephyr side
Question: 
Do I need to start zephyr_openamp_rsc_table.bin in Uboot at 0x7f8000 or 0x900000?

```
## No elf image at address 0x007f8000
## Starting auxiliary core stack = 0x200020B0, pc = 0x00900CE9...
baseboard is pi
lmb_dump_all:
 memory.cnt  = 0x1
 memory[0]	[0x80000000-0xbfffffff], 0x40000000 bytes flags: 0
 reserved.cnt  = 0x1
 reserved[0]	[0xbcd5d858-0xbfffffff], 0x032a27a8 bytes flags: 0
```

[image]

neuberfran, 2024-04-21:
@jlh-makeen 
```

readelf -S zephyr_openamp_rsc_table.elf 
Há 31 cabeçalhos de secção, começando no desvio 0xa14b4:

Cabeçalho de Seções:
  [Nr] Nome Fora do Tipo de Endereço ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] rom_start         PROGBITS        00180000 000114 00023c 00 WAX  0   0  4
  [ 2] text              PROGBITS        00901000 000350 00455c 00  AX  0   0  4
  [ 3] .ARM.exidx        ARM_EXIDX       0090555c 0048ac 000008 00  AL  2   0  4
  [ 4] initlevel         PROGBITS        00905564 0048b4 000040 00   A  0   0  4
  [ 5] device_area       PROGBITS        009055a4 0048f4 000028 00   A  0   0  4
  [ 6] sw_isr_table      PROGBITS        009055cc 00491c 0003f8 00  WA  0   0  4
  [ 7] tbss              NOBITS          009059c4 004d14 000004 00 WAT  0   0  4
  [ 8] rodata            PROGBITS        009059c4 004d14 0009c0 00   A  0   0  4
  [ 9] .ramfunc          PROGBITS        20000000 005858 000000 00   W  0   0  1
  [10] datas             PROGBITS        20000000 0056d4 0000e0 00  WA  0   0  4
  [11] device_states     PROGBITS        200000e0 0057b4 000004 00  WA  0   0  1
  [12] k_heap_area       PROGBITS        200000e4 0057b8 000014 00  WA  0   0  4
  [13] k_sem_area        PROGBITS        200000f8 0057cc 000030 00  WA  0   0  4
  [14] bss               NOBITS          20000128 005858 000442 00  WA  0   0  8
  [15] noinit            NOBITS          20000570 005858 001f4c 00  WA  0   0  8
  [16] .comment          PROGBITS        00000000 005858 000020 01  MS  0   0  1
  [17] .debug_aranges    PROGBITS        00000000 005878 001298 00      0   0  8
  [18] .debug_info       PROGBITS        00000000 006b10 048893 00      0   0  1
  [19] .debug_abbrev     PROGBITS        00000000 04f3a3 00a53b 00      0   0  1
  [20] .debug_line       PROGBITS        00000000 0598de 017ab0 00      0   0  1
  [21] .debug_frame      PROGBITS        00000000 071390 002c24 00      0   0  4
  [22] .debug_str        PROGBITS        00000000 073fb4 0099a6 01  MS  0   0  1
  [23] .debug_loc        PROGBITS        00000000 07d95a 015d1f 00      0   0  1
  [24] .debug_ranges     PROGBITS        00000000 093680 0035f8 00      0   0  8
  [25] .ARM.attributes   ARM_ATTRIBUTES  00000000 096c78 000032 00      0   0  1
  [26] .last_section     PROGBITS        009064ac 0057fc 000004 00   A  0   0  1
  [27] .resource_table   PROGBITS        009064b0 005800 000058 00  WA  0   0  4
  [28] .symtab           SYMTAB          00000000 096cac 005ca0 10     29 787  4
  [29] .strtab           STRTAB          00000000 09c94c 004a2a 00      0   0  1
  [30] .shstrtab         STRTAB          00000000 0a1376 00013c 00      0   0  1
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required)
```, G (group), T (TLS),
  C (compressed), x (unknown), o (OS specific), E (exclude),
  D (mbind), y (purecode), p (processor specific)

neuberfran, 2024-04-21:
@jlh-makeen 

I carried out several tests today, but unfortunately still without success.

[image]

jlh-makeen, 2024-04-24:
> @jlh-makeen Hello. Firstly, I want to congratulate you for being able to upload /dev/ttyRPMSG using Remoteproc/Linux-UserSpace/Zephyr Upstream on your imx7d device (even though it only has 1gb)
> 
> For my part, I haven't managed it yet and I would like you (if possible) to clarify my doubts using the photos below:
> 
> 1. Do you change something in NS announcement in the RPMSG on Linux?
>    ![Captura de tela de 2024-04-23 12-04-03](https://private-user-images.githubusercontent.com/38701397/324887407-3b1c29ce-81a2-4fa8-b208-16c29931a45b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5MzUwNjEsIm5iZiI6MTcxMzkzNDc2MSwicGF0aCI6Ii8zODcwMTM5Ny8zMjQ4ODc0MDctM2IxYzI5Y2UtODFhMi00ZmE4LWIyMDgtMTZjMjk5MzFhNDViLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDA0NTkyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYxZmMyMzQyOTVkZjE4MmE1ZjdkM2JmYTNhNGVkMzI4MWIxY2FmNTc5MDhlMzUyNjY4ZWMyZDc1ZjZlN2JhYTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OT7l904iozlHXNG6aspv5cuEGe3JHEOCSHUHBRoRaiU)
> 2. Is there a possibility that I can be successful using Linux 5.15? If so, what changes should I make?
>    ![Captura de tela de 2024-04-23 11-59-59](https://private-user-images.githubusercontent.com/38701397/324888081-fa5bb35f-ffcf-4244-ab4b-ac3cc995b2be.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5MzUwNjEsIm5iZiI6MTcxMzkzNDc2MSwicGF0aCI6Ii8zODcwMTM5Ny8zMjQ4ODgwODEtZmE1YmIzNWYtZmZjZi00MjQ0LWFiNGItYWMzY2M5OTViMmJlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDA0NTkyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJhNTU3N2IyY2ZjZmNhZjkxNDg3NWM5NjhjMTQwZGFhYTY3NmQwMDkzNzQwMTcxYmY0NWZmY2FjZDU1OGZhOTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.kudZjQmfaXVe5UUBkCny3okJWEQnJQkC8ZcN7rRPnys)
> 3. Can I use &tcml_code in zephyr,flash ?

@neuberfran 

There is no need to modify the Linux driver, at least in Linux 5.16.

We have only been able to successfully have ttyRPMSG on Linux 5.16 (and newer should work too)

You need PR #71684 in order to have the zephyr.elf file place the rom_start region (aka boot vector) into the OCRAM_S in order to be able to boot the M4 core using tcml_code memory.

There is a trick where you can get the M4 to boot by using ocram_s by changing the definition in the DT for ocram_s_code from address to the alias 0x00000000, this way the rproc driver is going to properly load the zephyr.elf file and boot-vectors at the right place. 
This will also be a valid addresssing for the M4 core as address space 0x00180000 for OCRAM_S code is aliased at 0x00000000.

I do not have the ressources and time to try it with linux 5.15, we need people from NXP or your vendor to support us here. As I explained linux 5.15 doesn't include the generic "tty-rpmsg" driver but instead the "imx-tty-rpmsg" which uses different rpmsg-endpoint names.

As of now, people have been very silent on the subject.

neuberfran, 2024-05-02:
@jlh-makeen I can do echo "test2" > /dev/ttyRPMSG0  and led light red blink in cp2102 UART-USB adapter in USB1 (by m4 side) But nothing appear in USB1 m4 side
[image]

jlh-makeen, 2024-05-02:
@neuberfran  
Looks like you are very close to the goal in fact your Linux kernel is now receiving the RPMSG name service annoucements message and it is creating the proper channels rpmsg-tty and rpmsg-client-sample.

Does /dev/ttyRPMSG0 show up by itself in the /dev directory ?

Because I can see the dmesg output it not showing that the rpmsg-tty driver was probed by the annoucement on channel rpmsg-tty / 0x401.

Did you manage to update to Linux 5.16 or are you still on 5.15 ?

If on 5.15, you probably don't have the rpmsg-tty driver but rather the imx_rpmsg_tty driver instead.
If that's the case, need to replace "rpmsg-tty" on the Zephyr side for the rpmsg endpoint name to the same as imx_rpmsg_tty driver instead "rpmsg-virtual-tty-channel".

See the endpoint names here:
https://github.com/nxp-imx/linux-imx/blob/lf-5.15.y/drivers/rpmsg/imx_rpmsg_tty.c#L211-L216

Change is done here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/ipc/openamp_rsc_table/src/main_remote.c#L290

I haven't tested on kernel 5.15 and cannot test it for you. 

Make sure you also enable the LOGGING framework in Zephyr to see all the messages coming on the M4 side as it doesn't use PRINTK the print statements in openamp_rsc_table example.

paoloturim, 2024-05-05:
@jlh-makeen

Yes, I use kernel 5.16 now,
Yes, I have /dev/ttyRPMSG now

So I'm trying two approaches:

1) First with your PR-71684 in order to have the zephyr.elf file place the rom_start region

In this case the resulting zephyr.bin became very large (511m) (i don't know why).

1,1) I even tried to approach your approach (that is, using PR) and uploading zephyr.elf only in sysfs (echo start) in Userspace, but I got error -62 and I can't upload rproc that way, unfortunately.

2) In the second approach, I got a smaller zephyr.bin and had better success (as shown in the print screen in the last post)  

I could be wrong, but this concludes that I have to use zephyr.bin and zephyr.elf in this project on a mandatory basis. Otherwise I may have error -62

Now comes the issue that Linux (TX part of Linux) is not sending messages (**RPMsg/OpenAMP messages) to the Zephyr

Now comes the issue that Linux (TX part of Linux) is not sending messages (**RPMsg/OpenAMP messages) to the Zephyr

** You can see in the window on the right side of the last photo (print screen) I posted that (despite CONFIG_PRINTK=n
and CONFIG_LOG_PRINTK=n) some messages already arrive in m4, but not the main ones about RPMsg and OpenAMP

What I've already tried to resolve:

neuberfran, 2024-05-06:
@jlh-makeen 

Yes, I use kernel 5.16 now,
Yes, I have /dev/ttyRPMSG now

So I'm trying two approaches:

1) First with your PR-71684 in order to have the zephyr.elf file place the rom_start region

In this case the resulting zephyr.bin became very large (511m) (i don't know why).

1,1) I even tried to approach your approach (that is, using PR) and uploading zephyr.elf only in sysfs (echo start) in Userspace, but I got error -62 and I can't upload rproc that way, unfortunately.

2) In the second approach, I got a smaller zephyr.bin and had better success (as shown in the print screen in the last post)  

I could be wrong, but this concludes that I have to use zephyr.bin(Uboot starts) and zephyr.elf(rproc starts) in this project on a mandatory basis. Otherwise I may have error -62

Now comes the issue that Linux (TX part of Linux) is not sending messages (**RPMsg/OpenAMP messages) to the Zephyr

** You can see in the window on the right side of the last photo (print screen) I posted that (despite CONFIG_PRINTK=n and CONFIG_LOG_PRINTK=n) some messages already arrive in m4, but not the main ones about RPMsg and OpenAMP, type test in  echo "test" > /dev/ttyRPMSG0	

What I've already tried to resolve:
1) Change CONFIG_PRINTK=n and CONFIG_LOG_PRINTK=n to CONFIG_PRINTK=y and CONFIG_LOG_PRINTK=y (Not solved)
2) Change vdev0vring0 and vdev0vring1 to vdevvring0 and vdevvring1, in Linux DT (Not solved)
3) Number of buffers 32

```
printk("\r\nOpenAMP[remote] Linux tty ccc1.2 started\r\n");            

        if (tty_msg.len) {
            LOG_INF("ipc_service_open_instance() 14 failure");
            snprintk(tx_buff, 13, "TTY 0x%04x: ", tty_ept.addr);
            memcpy(&tx_buff[12], tty_msg.data, tty_msg.len);
            LOG_INF("ipc_service_open_instance() 15 failure");
            rpmsg_send(&tty_ept, tx_buff, tty_msg.len + 13);
            rpmsg_release_rx_buffer(&tty_ept, tty_msg.data);
            LOG_INF("ipc_service_open_instance() 16 failure");
        }
        printk("\r\nOpenAMP[remote] Linux tty ccc3 started\r\n"); 
```  
```
OpenAMP[remote] Linux sample client ccc10 started                               
OpenAMP Linux sample client responder ended                                     
                                                                                
OpenAMP[remote] Linux task ccc21 started                                        
                                                                                
OpenAMP[remote] Linux task ccc20 started                                        
                                                                                
OpenAMP[remote] Linux tty ccc1.2 started                                        
I: ipc_service_open_instance() 14 failure                                       
I: ipc_service_open_instance() 15 failure                                       
I: ipc_service_open_instance() 16 failure                                       
                                                                                
OpenAMP[remote] Linux tty ccc3 started                                          
                                                                                
OpenAMP[remote] Linux tty ccc1.1 started  
```

Please, give-me your main_remote.c and your prj.conf generic and your_board.conf

jlh-makeen, 2024-05-07:
@neuberfran 

Let me try to be as clear as possible.

We have a custom carrier board using the cl-som-imx7d from Compulab so some changes might not be necessary on your board:

```
~/zephyrproject/zephyr/boards/makeen$ tree
.
├── index.rst
└── makeen_gw
    ├── board.yml
    ├── doc
    │   ├── dw_makeen_imx7d_m4.png
    │   └── index.rst
    ├── Kconfig.defconfig
    ├── Kconfig.makeen_gw
    ├── makeen_gw_mcimx7d_m4_defconfig
    ├── makeen_gw_mcimx7d_m4.dts
    ├── makeen_gw-pinctrl.dtsi
    └── makeen_gw.yaml

2 directories, 10 files

~/zephyrproject/zephyr/boards/makeen$ cat makeen_gw/makeen_gw_mcimx7d_m4_defconfig 
#
# Copyright (c) 2024, MAKEEN Energy
#
# SPDX-License-Identifier: Apache-2.0
#

CONFIG_UART_CONSOLE=y
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_CONSOLE=y
CONFIG_XIP=y
CONFIG_PINCTRL=y
CONFIG_GPIO_HOGS=y
```
In the Zephyr DT I have &mub set to status = "okay"

Regarding the project I can build the native example from **samples/subsys/ipc/openamp_rsc_table** using this board config:

Create **boards/makeen_gw_mcimx7d_m4.conf** in **samples/subsys/ipc/openamp_rsc_table**:
```
CONFIG_IPM=y
CONFIG_IPM_IMX_MAX_DATA_SIZE_4=y
CONFIG_OPENAMP_WITH_DCACHE=y
CONFIG_PLATFORM_SPECIFIC_INIT=n
CONFIG_OPENAMP=y
CONFIG_OPENAMP_RSC_TABLE_NUM_RPMSG_BUFF=32
CONFIG_OPENAMP_RSC_TABLE=y
CONFIG_OPENAMP_MASTER=n
```

Before Building you must use my ROMSTART_REGION relocation PR in order to have the correct boot-vector placement for Linux RPROC. You don't need the BIN files anymore, only ELF file.

Then on your Linux board create the following **boot_m4.sh** script:
```
#!/bin/bash

fw=$1

cp $fw /lib/firmware

echo stop > /sys/class/remoteproc/remoteproc0/state
echo $fw > /sys/class/remoteproc/remoteproc0/firmware
echo start > /sys/class/remoteproc/remoteproc0/state
```
Don't forget to `chmod +x boot_m4.sh`

Build :
```
cd ~/zephyrproject/zephyr/samples/subsys/ipc/openamp_rsc_table
west build -p always -b makeen_gw_mcimx7d_m4
```

Then SCP the `zephyr_openamp_rsc_table.elf` file and `boot_m4.sh` to your home directory of the linux board:

Then call the script:
```
su root
./boot_m4.sh zephyr_openamp_rsc_table.elf && dmesg
```

It should output the following:

```
[ 1529.626593] remoteproc remoteproc0: stopped remote processor imx-rproc
[ 1529.637142] remoteproc remoteproc0: powering up imx-rproc
[ 1529.645231] remoteproc remoteproc0: Booting fw image zephyr_openamp_rsc_table.elf, size 1282424
[ 1529.661279]  remoteproc0#vdev0buffer: assigned reserved memory node vdev0buffer0@bfff2000
[ 1529.718327] virtio_rpmsg_bus virtio0: rpmsg host is online
[ 1529.718981] virtio_rpmsg_bus virtio0: creating channel rpmsg-tty addr 0x400
[ 1529.722666]  remoteproc0#vdev0buffer: registered virtio0 (type 7)
[ 1529.730217] virtio_rpmsg_bus virtio0: creating channel rpmsg-client-sample addr 0x401
[ 1529.741235] virtio_rpmsg_bus virtio0: creating channel rpmsg-tty addr 0x402
[ 1529.741946] remoteproc remoteproc0: remote processor imx-rproc is now up
```

Now I have 2 `/dev/ttyRPMSG0` and `/dev/ttyRPMSG1` showing up if you are using the latest openamp_rsc_table sample that enables the SHELL and SHELL_RPMSG_BACKEND.

To use the shell you can use your TTY terminal of choice like **minicom**:
```
su root
minicom -D /dev/ttyRPMSG0
```
Then press CTRL+C a few times to trigger the shell prompt:
You should see `ipc:~$`, then you can type kernel version to see that you are currently running Zephyr on the M4 !

```
Welcome to minicom 2.8

OPTIONS: I18n
Compiled on Jan  1 2021, 17:45:55.
Port /dev/ttyRPMSG0, 06:09:54

Press CTRL-A Z for help on special keys

ipc:~$ kernel version
Zephyr version 3.6.99
ipc:~$
```

You can also use /dev/ttyRPMSG1 to see the regular TTY echo of the sample when you send data to the tty
```
Welcome to minicom 2.8

OPTIONS: I18n
Compiled on Jan  1 2021, 17:45:55.
Port /dev/ttyRPMSG1, 06:10:44

Press CTRL-A Z for help on special keys

TTY 0x0402: eTTY 0x0402: eTTY 0x0402: eTTY 0x0402: e
```

Be aware that the sample uses approximately 48kB of CODE memory so I think you must run the code from **OCRAM**, but RAM usage is approx 20kB and can fit on any of TCM/OCRAM_S/DDR

```
[171/171] Linking C executable zephyr/zephyr_openamp_rsc_table.elf
Memory region         Used Size  Region Size  %age Used
 ROMSTART_REGION:         572 B         1 KB     55.86%
           FLASH:       48304 B       128 KB     36.85%
```

I hope this helps.

neuberfran, 2024-05-07:
@jlh-makeen Tks. But I don't want open /dev/ttyRPMSG0 in minicom or picocom. I want:
1) open Linux a7 side in picocom usb0
2) open Zephyr m4 side in minicom usb1
3) Execute rproc sysfs in linux and in linux -> echo "test" > /dev/ttyRPMSG0.
4) At that moment I would see the word test next to the Zephyr (in usb1 minicom window)

jlh-makeen, 2024-05-07:
> @jlh-makeen Tks. But I don't want open /dev/ttyRPMSG0 in minicom or picocom. I want:
> 
> 1. open Linux a7 side in picocom usb0
> 2. open Zephyr m4 side in minicom usb1
> 3. Execute rproc sysfs in linux and in linux -> echo "test" > /dev/ttyRPMSG0.
> 4. At that moment I would see the word test next to the Zephyr (in usb1 minicom window)
>    Note: 1) All that remains to be done is resolve this issue at the moment 4)
> 5. I try LOG_INF("TTY 0x%04x: ", tty_ept.addr); Result is: TTY 0x401
> 6. Nothing to  tty_ept.addr, Why
> 
> ```
> if (tty_msg.len) {
>             LOG_INF("ipc_service_open_instance() 14 failure");
>             snprintf(tx_buff, 13, "TTY 0x%04x: ", tty_ept.addr);
>             memcpy(&tx_buff[12], tty_msg.data, tty_msg.len);
>             LOG_INF("ipc_service_open_instance() 15 failure");
>             rpmsg_send(&tty_ept, tx_buff, tty_msg.len + 13);
>             rpmsg_release_rx_buffer(&tty_ept, tty_msg.data);
>             LOG_INF("ipc_service_open_instance() 16 failure");
>         }
> ```
> 
> https://developer.nordicsemi.com/nRF_Connect_SDK/doc/2.1.3/zephyr/samples/subsys/ipc/openamp_rsc_table/README.html
> 
> ![Screenshot from 2024-05-07 14-20-52](https://private-user-images.githubusercontent.com/38701397/328617142-07d7e914-3aeb-4691-bbf9-beeceee74932.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUxMDM3ODAsIm5iZiI6MTcxNTEwMzQ4MCwicGF0aCI6Ii8zODcwMTM5Ny8zMjg2MTcxNDItMDdkN2U5MTQtM2FlYi00NjkxLWJiZjktYmVlY2VlZTc0OTMyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA3VDE3MzgwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRhYzE4OGZhNGQ2MzhiNDI1YmY4OTZmNGYyYmUzNzhiMzBjMDA3ZTNhNjRkYjgxOGUzOTQ4YTk2NDBkYWQ5Y2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Pw4hlVFNBplyqPIkBPVlwLC1U4bp8a5s4ilNCI1mBmA)

Congrats ! 
I can see in your screenshot here that the Zephyr TTY side has successfully received the message from the Linux and sent the Echo back to Linux ! See the **TTY 0x0401: Hello Zephyr** so it is working perfectly now !

You don't see anything from your LOG_INF because you have not enabled CONFIG_LOG=y.

Try to add a printk statement instead, like this:
```
if (tty_msg.len) {
  printk("Linux rx message %*.s\n", tty_msg.len, tty_msg.data);
  snprintf(tx_buff, 13, "TTY 0x%04x: ", tty_ept.addr);
  memcpy(&tx_buff[12], tty_msg.data, tty_msg.len);
  rpmsg_send(&tty_ept, tx_buff, tty_msg.len + 13);
  rpmsg_release_rx_buffer(&tty_ept, tty_msg.data);
}
```

When you receive a message from Linux, you'll the message content appear in the Zephyr console.

Now it is implementation details you will have to care about by yourself, now that /dev/ttyRPMSG shows in Linux and the RPMSG sample works on your platform I think we should close the issue as it's resolved.

If you encounter application specific requests, feel free to report to the documentation or open a new issue for the new subject.

neuberfran, 2024-05-07:
@jlh-makeen Unfortunately, I still couldn't get echo test > /dev/ttyRPMSG0 to appear in the other window

I am one of the first interested parties to close this topic.

I thought that adding the printk you suggested would solve it (it was the only thing I hadn't tried yet). But unfortunately not yet

[image]

```
if (tty_msg.len) {
            printk("Linux rx message %*.s\n", tty_msg.len, tty_msg.data);
            LOG_INF("ipc_service_open_instance() 14 failure");
            snprintf(tx_buff, 13, "TTY 0x%04x: ", tty_ept.addr);
            memcpy(&tx_buff[12], tty_msg.data, tty_msg.len);
            LOG_INF("ipc_service_open_instance() 15 failure");
            rpmsg_send(&tty_ept, tx_buff, tty_msg.len + 13);
            rpmsg_release_rx_buffer(&tty_ept, tty_msg.data);
            LOG_INF("ipc_service_open_instance() 16 failure");
        }
```

**prj.conf**
```

CONFIG_KERNEL_BIN_NAME="zephyr_openamp_rsc_table"
CONFIG_PRINTK=y
CONFIG_LOG=y 
CONFIG_LOG_DEFAULT_LEVEL=0
CONFIG_LOG_MODE_MINIMAL=y
CONFIG_IPM=y
CONFIG_MAIN_STACK_SIZE=1024
CONFIG_HEAP_MEM_POOL_SIZE=1024
CONFIG_OPENAMP=y
CONFIG_OPENAMP_RSC_TABLE_NUM_RPMSG_BUFF=32
CONFIG_OPENAMP_RSC_TABLE=y
CONFIG_OPENAMP_MASTER=n
CONFIG_PLATFORM_SPECIFIC_INIT=n
#CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y

# CONFIG_LOG=y
# CONFIG_LOG_BACKEND_UART=y
# CONFIG_SERIAL=y

# CONFIG_IPC_SERVICE=ys
# CONFIG_PRINTK=y 
# CONFIG_LOG=y # required to allocate virtqueues 
# #CONFIG_HEAP_MEM_POOL_SIZE=2048
# CONFIG_IPC_SERVICE_LOG_LEVEL_INF=y 
```
**pico_pi_m4.conf**

```
CONFIG_UART_CONSOLE=y
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_CONSOLE=y
CONFIG_LOG_PRINTK=y
CONFIG_IPM_IMX_MAX_DATA_SIZE_16=n
CONFIG_IPM_IMX_MAX_DATA_SIZE_4=y
CONFIG_OPENAMP_WITH_DCACHE=y

# CONFIG_IPM=y--> Interrupt-based inter-processor mailboxes
# CONFIG_IPM_IMX=y--> Your IPM driver here for Renesas R-Car
# CONFIG_OPENAMP=y--> Enable OpenAMP IPC library
# CONFIG_OPENAMP_SLAVE=y --> Virtqueue are initialized by the host (Linux)
# CONFIG_OPENAMP_RSC_TABLE=y
```

[11 more replies]

closed 2024-05-11 by neuberfran