zephyrproject-rtos/zephyr #69443: Changing VERSION requires prestine build / no warning if not done

GenericGuy, 2024-02-26:
**Describe the bug**
When using MCUboot in conjunction with an app VERSION file, updates to the VERSION file are not reflected in the signed binary unless a pristine build is performed.
I would expect a warning if a version mismatch is detected in case this is expected behavior.

**To Reproduce**
1. Follow the Zephyr Getting Started Guide
2. Add a `VERSION` file to the "Sample with MCUboot" sample (create `samples/application_development/sysbuild/with_mcuboot/VERSION`)
3. Build the application as outlined in the "Sample with MCUboot" page (`west build -b reel_board --sysbuild samples/application_development/sysbuild/with_mcuboot`)
4. Observe that the version is correctly reflected in the image (`imgtool dumpinfo build/with_mcuboot/zephyr/zephyr.signed.bin | grep version`)
5. Change the version file in any way and perform a rebuild
6. Observe that the version from step 4 is still the same

**Expected behavior**
Either the version should be updated as one would assume or a warning should be displayed, prompting the user to perform a pristine build.

**Possible cause**
As far as I understand, the problem is the Kconfig generation step. Even though the version changes, the Kconfig variables are not updated to reflect this change (i.e. the field `CONFIG_MCUBOOT_IMGTOOL_SIGN_VERSION` inside the file `build/with_mcuboot/zephyr/.config` is not updated).

**Workaround**
By adding the following section to line 1315 of the file `scripts/kconfig/kconfiglib.py`, building doesn't succeed and informs the user to perform a pristine build.

```python
elif sym.orig_type is STRING:
    if name == 'MCUBOOT_IMGTOOL_SIGN_VERSION' and val != '"' + os.getenv("APPVERSION").replace("-dev", "") + '"':
        self._warn("App version changed. Please perform a pristine build.")
    match = _conf_string_match(val)
```

**Environment**
 - OS: Ubuntu 22.04
 - Toolchain: Zephyr SDK 0.16.5
 - Commit SHA: ee9c44fee63171ad6af94b6f2659c57a23fef892

nordicjm, 2024-02-26:
This is expected, when the application version idea was submitted there were 2 ways of doing it, one did not use a VERSION file and thus did not have this flaw, the other (which was chosen) has this flaw because the VERSION file just sets the *default* value of the signing key version, and at the time another user raised the issue that they wanted to use a completely different application version to a signing version. So this is not a bug and is expected behaviour.

GenericGuy, 2024-02-26:
I see, thanks.
Any thoughts on the workaround (or something similar to the one I proposed)? For someone not familiar with this restriction it might be quite surprising that the build process does not take care of this, or at least it was for me.

GenericGuy, 2024-02-26:
Got it, thanks!

closed 2024-02-26 by GenericGuy