zephyrproject-rtos/zephyr #69447: drivers: audio: mpxxdtyy: Driver is only able to handle PCM block size of 1 ms

petarsusac, 2024-02-26:
**Describe the bug**
The DMIC driver API allows the user to choose an arbitrary PCM block size, as demonstrated by the sample at `samples/drivers/audio/dmic`. However, the `mpxxdtyy` microphone drivers only work properly when the PCM block size is equal to 1 millisecond of audio (for a sampling frequency of 16 kHz, this is 16 samples or 32 bytes).  Otherwise, only the first millisecond of each PCM block contains the audio signal, and the rest are garbage values.

The issue is caused by the PDM to PCM filtering layer. The functions `Open_PDM_Filter_64()` and `Open_PDM_Filter_128()`, part of the OpenPDMFilter library, do not take a signal length parameter and process a fixed length of 1 millisecond of audio.

Since I assume that this is the intended behavior of the OpenPDMFilter library, a potential workaround would be to call the filtering function multiple times for each PDM/PCM block in the `mpxxdtyy` driver, once for each millisecond of audio.

**To Reproduce**
Build and run the sample at `samples/drivers/audio/dmic` with a MPxxDTyy microphone connected to the I2S bus and inspect the PCM blocks returned by `dmic_read()`.

**Expected behavior**
The PCM blocks should contain the audio signal acquired from the microphone, regardless of the block size.

henrikbrixandersen, 2024-02-26:
Please use our bug template when reporting bugs and avoid deleting a number of the required sections. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

petarsusac, 2024-02-26:
The issue has been edited to comply with the bug template.

avisconti, 2024-03-21:
Hi @petarsusac ,

Yes, I see your point. Unfortunately I do not have the correct setup to reproduce the issue and correct it, but seems that your suggestion (i.e. calling the Open_PDM_Filter multiple times) is good. I guess that we could do it in sw_filter_lib_run() using the pcm_size argument.

Would it be possible for you to prepare/test a PR with a fix?

petarsusac, 2024-03-25:
Sure, I already have a quick fix, I just have to clean it up a bit and prepare a PR. I just wanted to confirm that this was the right place to do it first.