zephyrproject-rtos/zephyr #69456: doc: Documentation issue in 'boards/arm/mimxrt1015_evk/doc/index'

nuncio-bitis, 2024-02-26:
Cannot build samples

Running  "west build -p always -b mimxrt1015_evk samples/basic/blinky" causes the following output:
-- west build: making build dir /home/abbott/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/abbott/zephyrproject/zephyr/samples/basic/blinky
-- CMake version: 3.28.3
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/abbott/.cache/zephyr
-- Zephyr version: 3.6.0-rc2 (/home/abbott/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: mimxrt1015_evk
-- Found host-tools: zephyr 0.16.5 (/home/abbott/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/home/abbott/zephyr-sdk-0.16.5)
-- Found Dtc: /home/abbott/zephyr-sdk-0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/abbott/zephyrproject/zephyr/boards/arm/mimxrt1015_evk/mimxrt1015_evk.dts
In file included from /home/abbott/zephyrproject/zephyr/boards/arm/mimxrt1015_evk/mimxrt1015_evk.dts:10,
                 from <command-line>:
/home/abbott/zephyrproject/zephyr/boards/arm/mimxrt1015_evk/mimxrt1015_evk-pinctrl.dtsi:9:10: fatal error: nxp/nxp_imx/rt/mimxrt1015daf5a-pinctrl.dtsi: No such file or directory
    9 | #include <nxp/nxp_imx/rt/mimxrt1015daf5a-pinctrl.dtsi>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMake Error at /home/abbott/zephyrproject/zephyr/cmake/modules/extensions.cmake:4139 (message):
  failed to preprocess devicetree files (error code 1):
  /home/abbott/zephyrproject/zephyr/boards/arm/mimxrt1015_evk/mimxrt1015_evk.dts
Call Stack (most recent call first):
  /home/abbott/zephyrproject/zephyr/cmake/modules/dts.cmake:215 (zephyr_dt_preprocess)
  /home/abbott/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /home/abbott/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/abbott/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/abbott/zephyrproject/zephyr/build -GNinja -DBOARD=mimxrt1015_evk -S/home/abbott/zephyrproject/zephyr/samples/basic/blinky

* Page: `boards/arm/mimxrt1015_evk/doc/index`
* Version: main
* SHA-1: e5fe1c162570fcd2be4de583ce5e21273e282c41

nordicjm, 2024-02-26:
did you do `west update`?

closed 2024-02-26 by nordicjm

nuncio-bitis, 2024-02-26:
Oh thank you! That did it!   The instructions didn’t say to do that.
Thanks so much!

--------------------------------------------------------
Jim Parziale

Reading, MA 01867

--------------------------------------------------------

Some people, when confronted with a problem, think "I know, I'll use regular expressions."

Now they have two problems.