zephyrproject-rtos/zephyr #69478: Particle Boron - Add a sample that uses the builtin LTE modem

zpm1066, 2024-02-26:
**Is your enhancement proposal related to a problem? Please describe.**
I'm trying to use the LTE modem on the Particle Boron. The Boron seems to be pretty well supported board for quite sometime and was wondering if someone can share an example to get things off the ground for anyone new to using the LTE modem.

**Describe the solution you'd like**
I'd like to see a modem sample added for Particle Boron board.

**Describe alternatives you've considered**
I've been using the Particle DeviceOS that fully supports the Boron LTE modem and connects to their cloud. However, I'd like put Zephyr on the Boron and use it other cloud services.

**Additional context**

WilliamGFish, 2024-02-26:
The modem on the Boron is a UBlox Sara (U2, R4 & R5) modem, have a look at the driver for supporting protocols.
Billy..

zpm1066, 2024-02-26:
@WilliamGFish Thanks. Will do.
I have a mixture of the two Sara modem versions.
Will I be able to connect to the Particle cloud once I put Zephyr on the Boron and still use their free tier for LTE access?

WilliamGFish, 2024-02-26:
> @WilliamGFish Thanks. Will do.
> I have a mixture of the two Sara modem versions.
> Will I be able to connect to the Particle cloud once I put Zephyr on the Boron and still use their free tier for LTE access?
> 
> 

Two different questions:
- Can zephyr connect and control the Sara modems? Absolutely.
- Can you connect to the particle platform using zephyr? Now that's a question for particle.
Billy..

zpm1066, 2024-02-26:
Are you using the Boron modem? If so, Are you connecting to a cloud service like Golioth, AWS, Nordic or other? 

I’ll check with Particle but doubt very much they’ll support a Boron running Zephyr connecting to their cloud.