zephyrproject-rtos/zephyr #69483: STM32H5 Missing from MCO2

cradzik, 2024-02-26:
Hello,

I'm trying to get a clock out on MCO2 on a STM32H5, and ran into some issues.  

I couldn't get it to work until I added "SOC_SERIES_STM32H5X || \" after this line:
https://github.com/zephyrproject-rtos/zephyr/blob/4a72cb16cfd5a8df86b7f1653632dd49b3ea5ff5/drivers/clock_control/Kconfig.stm32#L256C1-L256C36

I believe the STM32H5 is missing from the list of dependencies.

Thanks
Chris

FRASTM, 2024-02-27:
thank @cradzik to raise the pb , 
Yes, the MCO2 div is actually missing the stm32H5 serie.
Do you intend to push a PR for correcting this ?

cradzik, 2024-02-27:
@FRASTM.  Here is my pull request:
https://github.com/zephyrproject-rtos/zephyr/pull/69535

Please let me know if I did anything incorrectly.

Thanks!
Chris

PR #69535 (2024-03-04): drivers: clock_control: Fix STM32H5 MCO2 support
Add STM32H5 to the list of dependencies for CLOCK_STM32_MCO2_DIV.  Without the change, attempting to build using MCO2 would cause build errors.

Fixes #69483.
closed 2024-03-04 by aescolar