zephyrproject-rtos/zephyr #69488: Able to leave off stop bit for I2C transactions on nRF52/nRF53

RobMeades, 2024-02-26:
we raised issue https://github.com/zephyrproject-rtos/zephyr/issues/69290 on the STM32 I2C driver as we had found that the driver forced `I2C_MSG_STOP` at the end of every transfer; we didn't want this as we wanted to perform a write without a stop bit, followed by a read, as separate transfers.

@teburd has pointed out that the STM32 behaviour is the correct one, `I2C_MSG_STOP` _should_ be forced by the I2C driver at the end of every transaction and a note has been added to the I2C API to this effect (see https://github.com/zephyrproject-rtos/zephyr/pull/69484/files).

We have changed our implementation to work with this but @teburd asked if we could raise on issue on the nRF52/nRF53 drivers for NOT forcing `I2C_MSG_STOP` at the end of a transfer.

Issue hereby raised.

aescolar, 2024-02-27:
CC @Mierunski @awojasinski

nordic-piks, 2024-08-29:
FYI @kl-cruz 
nrf_triage_done

kl-cruz, 2024-08-29:
@bjarki-andreasen We need to check whether this configuration is possible in our hardware

closed 2025-05-24 by github-actions[bot]