zephyrproject-rtos/zephyr #69509: STM32WBA52 bootloader cannot jump to application code.

lnhao, 2024-02-27:
Why can't STM32WBA jump to the APP while STM32F429 can jump and run normally with the same MCUboot and application code?

STM32WBA is aligned according to its own flash sector size - 8K.

Have you ever encountered a similar problem?

Thank you!

nordicjm, 2024-02-27:
Can you fill the information in from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md this is not optional

lnhao, 2024-02-27:
@nordicjm  Will do, thank you！

erwango, 2024-02-27:
> Can you fill the information in from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md this is not optional

That would help to provide efficient support indeed

Anyway, can you have a try with what was done for U5 [here](https://github.com/zephyrproject-rtos/zephyr/blame/main/soc/arm/st_stm32/stm32u5/Kconfig.defconfig.series#L13) ?

lnhao, 2024-02-28:
> That would help to provide efficient support indeed
>
> Anyway, can you have a try with what was done for U5 [...]

Referenced U5, I experimented on the stm32wba, but still can't jump.

Br, thank you!

erwango, 2024-02-28:
> Referenced U5, I experimented on the stm32wba, but still can't jump.

Then, we're coming back to the initial request. If you want us to help investigating this point, please use the bug template form.

lnhao, 2024-02-28:
> > Referenced U5, I experimented on the stm32wba, but still can't jump.
> 
> Then, we're coming back to the initial request. If you want us to help investigating this point, please use the bug template form.

Yes, I want you to help to investigate this point. 
I use a bug template to re-publish an issue ticket, right?

nordicjm, 2024-02-28:
> Yes, I want you to help to investigate this point. I use a bug template to re-publish an issue ticket, right?

No, go to the top post, click the 3 dots in the top right corner and click edit, then add the information to your post and update

lnhao, 2024-02-28:
@erwango @nordicjm Updated

erwango, 2024-02-28:
There maybe an issue in your configuration.
I had a try with `samples/application_development/sysbuild/with_mcuboot`:
```
west build -p always -b nucleo_wba55cg --sysbuild samples/application_development/sysbuild/with_mcuboot
west flash -d build/nucleo_wba55cg/
```
Note that you need to add :
```
	chosen {
		zephyr,code-partition = &slot0_partition;
	};
```

erwango, 2024-02-28:
You used nucleo_wba55cg board, it can jump to application. Yes it is jumping in application.
nucleo_wba52cg and nucleo_wba55cg are pretty close. I doubt it can have an impact.
Also note that BLE isn't available yet on nucleo_wba52cg due to STM32Cube package limitation (provided BLE controller library not compatible with STM32WBA52 cut).

erwango, 2024-02-28:
> If I use the wba55cg board to build an application image. Flashing to HW board of wba52cg It should be able to run, right?

Not sure as it enabled SMPS wich is not supported on WB52... (only difference).

lnhao, 2024-02-29:
> > If I use the wba55cg board to build an application image. Flashing to HW board of wba52cg It should be able to run, right?
> 
> Not sure as it enabled SMPS wich is not supported on WB52... (only difference).

Can you use a WBA52's board to verify bootloader?

lnhao, 2024-02-29:
> @lnhao Did you try to run these exact commands ?
> 
> > west build -p always -b nucleo_wba52cg --sysbuild samples/application_development/sysbuild/with_mcuboot
> > west flash -d build

yes, I have tried.

lnhao, 2024-03-04:
Hi @erwango, do you try to run WBA52's board?

erwango, 2024-03-12:
> Hi @erwango, did you try to run WBA52's board?

I've just managed to get one and I can confirm the issue, which is linked to clock configuration.
As a workaround, you can retrieve clock_configuration from nucleo_wba55cg board:
```diff
&clk_hse {
	status = "okay";
+	hse-div2;
};
&rcc {
-	clocks = <&pll1>;
+	clocks = <&clk_hse>; 
-	clock-frequency = <DT_FREQ_M(96)>;
+	clock-frequency = <DT_FREQ_M(16)>;
	ahb-prescaler = <1>;
-	ahb5-prescaler = <4>;
+	ahb5-prescaler = <2>;
	apb1-prescaler = <1>;
	apb2-prescaler = <1>;
	apb7-prescaler = <1>;
};
```
In parallel, I'm working on a more complete fix.

erwango, 2024-03-13:
> If I want a larger system clock speed, it still depends on PLL.

Sure. I'm looking into it.

erwango, 2024-03-13:
@lnhao See https://github.com/zephyrproject-rtos/zephyr/pull/70175 for more complete fix

lnhao, 2024-03-20:
> @lnhao See #70175 for more complete fix

@erwango Thank you for your update. This is good news. I will try to do this verifying later.

lnhao, 2024-03-21:
Hi @erwango, I have pulled your branch (`fix_wba_chainloaded`) and tested the bootloader jumping. It can jump to the application code. Looking forward to merging into the Zephyr code repository.

erwango, 2024-03-21:
> Hi @erwango, I have pulled your branch (`fix_wba_chainloaded`) and tested the bootloader jumping. It can jump to the application code. Looking forward to merging into the Zephyr code repository.

Ok, so please comment on the PR that you've tested it successfully (I guess you can also approve).

lnhao, 2024-03-21:
Log of successful testing: 
```
*** Booting Zephyr OS build v3.6.0-693-gf9372d62dd9a ***
I: Starting bootloader
I: Image index: 0, Swap type: none
I: Bootloader chainload address offset: 0x20000
*** Booting Zephyr OS build v3.6.0-693-gf9372d62dd9a ***
Address of sample 0x8020000
Hello sysbuild with mcuboot! nucleo_wba52cg
```

erwango, 2024-03-21:
> Log of successful testing:
>
> ```
> *** Booting Zephyr OS build v3.6.0-693-gf9372d62dd9a ***
> I: Starting bootloader
> I: Image index: 0, Swap type: none
> I: Bootloader chainload address offset: 0x20000
> *** [...]

Great, can you put this directly in https://github.com/zephyrproject-rtos/zephyr/pull/70175 ?
This could be useful to speed up the review.

closed 2024-04-03 by fabiobaltieri