zephyrproject-rtos/zephyr #69530: drivers: esp32_at: UDP Sockets Client (STATION) Connection Failure

WilliamGFish, 2024-02-27:
**Description**
when the esp32_at modem is used as a Wi-Fi client (station mode) the UDP connections fail after initial handshake.

With the recent changes to the driver, which removed some 'odd' exception handling and the updating of the bind function (see below). Unfortunately this causes the address for UDP sockets to be updated to 0.0.0.0 terminating connections
[drivers: esp_at: implement bind() and recvfrom() for UDP sockets #68586]

```
static int esp_bind(struct net_context *context, const struct sockaddr *addr,
		    socklen_t addrlen)
{
	struct esp_socket *sock;
	struct esp_data *dev;

	sock = (struct esp_socket *)context->offload_context;
	dev = esp_socket_to_dev(sock);

	if (esp_socket_ip_proto(sock) == IPPROTO_TCP) {
		return 0;
	}

	if (IS_ENABLED(CONFIG_NET_IPV4) && addr->sa_family == AF_INET) {
		LOG_DBG("link %d", sock->link_id);

		if (esp_socket_connected(sock)) {
			return -EISCONN;
		}

		k_mutex_lock(&sock->lock, K_FOREVER);
		sock->dst = *addr;
		sock->connect_cb = NULL;
		sock->conn_user_data = NULL;
		k_mutex_unlock(&sock->lock);

		_sock_connect(dev, sock);

		return 0;
	}

	return -EAFNOSUPPORT;
}
```

**Expected behavior**
UDP connections function as expected

**Impact**
Showstopper for anyone using UDP with the esp32_at driver

**Additional context**
The KConfig file has no default; needs to be added
```
config WIFI_ESP_AT_CIPDINFO_USE
	bool "Use CIPDINFO to get peer ip and port"
        default n
	help
	  Enable AT+CIPDINFO got get peer ip-address and port.
```

The bind function needs to include something akin to below so to only update address if IPD INFO selected:
```
#if defined(CONFIG_WIFI_ESP_AT_CIPDINFO_USE)
		k_mutex_lock(&sock->lock, K_FOREVER);
		sock->dst = *addr;
		sock->connect_cb = NULL;
		sock->conn_user_data = NULL;
		k_mutex_unlock(&sock->lock);

		_sock_connect(dev, sock);
#endif
```

LonesomeUnixCowboy, 2024-02-28:
Intresting... This is not a behavior i have see but maybe i have not tested is thoroughly. I have tested like this:

1. Connect to WiFi as station.

2. create a socket
`int fd = socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP);`

3. bind a adress and port (i packed `addr` with `inet_pton()` before hand)
`rc = bind(fd, (struct sockaddr *) &addr, sizeof(addr));`

4. wait with `poll()` for incoming data
```	
        struct pollfd polls;                                                    
	                                                                                
	polls.fd = fd;                                                              
	polls.events = POLLIN;                                                      
		                                                                 
	 do {                                                                    
		                                                                 
		 rc = poll(&polls, 1, 1000);                                         
		                                                                 
	} while (rc == 0);
```  
	
5. send something to the socket, i used goodol' nc

6. poll returns, recieve with `recvfrom()`
```
	struct sockaddr_in r_addr;                                              
	socklen_t r_addr_len = sizeof(struct sockaddr_in);
	                     
	rc = recvfrom(fd, (void *) rx_msg, sizeof(rx_msg), 0,                       
		      (struct sockaddr *) &r_addr, &r_addr_len);
```

7. Success! (for me atleast)

I have also done `getaddrinfo()` requests (using UDP sockets) and they have worked fine.

What I'am a missing here?

WilliamGFish, 2024-03-01:
Have you done this without the IPD INFO enabled?

I used a nice simple time call:
rc = sntp_simple("time.google.com", SYS_FOREVER_MS, &sntp_time);

LonesomeUnixCowboy, 2024-03-01:
Yes i have. Did you use passive or active mode?

WilliamGFish, 2024-03-01:
CONFIG_WIFI_ESP_AT_SCAN_PASSIVE=n

LonesomeUnixCowboy, 2024-03-01:
I mean AT+CIPRECVMODE. I guess then that you have built with CONFIG_WIFI_ESP_AT_PASSIVE_MODE off. What happens you turn it on?

WilliamGFish, 2024-03-01:
Just be clear I'm using the ESP32 as a modem connected via UART

### Passive mode enabled: Does not boot correctly

> 00:00:00.456,542] \033[0m<inf> wifi_esp_at: Waiting for interface to come up\033[0m
> [00:00:00.459,777] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
>                                             65 74 73 20 4a 75 6e 20  20 38 20 32 30 31 36 20 |ets Jun   8 2016 
>                                             30 30 3a 32 32 3a 35 37                          |00:22:57         \033[0m
> [00:00:00.464,874] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
>                                             72 73 74 3a 30 78 31 20  28 50 4f 57 45 52 4f 4e |rst:0x1  (POWERON
>                                             5f 52 45 53 45 54 29 2c  62 6f 6f 74 3a 30 78 31 |_RESET), boot:0x1
>                                             33 20 28 53 50 49 5f 46  41 53 54 5f 46 4c 41 53 |3 (SPI_F AST_FLAS
>                                             48 5f 42 4f 4f 54 29                             |H_BOOT)          \033[0m
> [00:00:00.467,071] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV  \033[0m
> 0m
> [00:00:01.280,670] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
>                                             4f 4b                                            |OK               \033[0m
> [00:00:01.280,700] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)\033[0m
> [00:00:01.330,871] \033[0m<dbg> modem_cmd_handler: modem_cmd_send_ext: SENT DATA
>                                             41 54 2b 43 57 44 48 43  50 3d 31 2c 31          |AT+CWDHC P=1,1   \033[0m
> [00:00:01.334,960] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
>                                             4f 4b                                            |OK               \033[0m
> [00:00:01.334,991] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)\033[0m
> [00:00:01.385,192] \033[0m<dbg> modem_cmd_handler: modem_cmd_send_ext: SENT DATA
>                                             41 54 2b 43 49 50 4d 55  58 3d 31                |AT+CIPMU X=1     \033[0m
> [00:00:01.387,908] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
>                                             4f 4b                                            |OK               \033[0m
> [00:00:01.387,969] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)\033[0m
> [00:00:01.438,110] \033[0m<dbg> modem_cmd_handler: modem_cmd_send_ext: SENT DATA
>                                             41 54 2b 43 57 4c 41 50  4f 50 54 3d 30 2c 32 33 |AT+CWLAP OPT=0,23\033[0m
> [00:00:01.441,131] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
>                                             4f 4b                                            |OK               \033[0m
> [00:00:01.441,192] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)\033[0m
> [00:00:01.491,333] \033[0m<dbg> modem_cmd_handler: modem_cmd_send_ext: SENT DATA
>                                             41 54 2b 43 57 4d 4f 44  45 3d 31                |AT+CWMOD E=1     \033[0m
> [00:00:01.494,445] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
>                                             4f 4b                                            |OK               \033[0m
> [00:00:01.494,506] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)\033[0m
> [00:00:01.544,677] \033[0m<dbg> modem_cmd_handler: modem_cmd_send_ext: SENT DATA
>                                             41 54 2b 43 57 41 55 54  4f 43 4f 4e 4e 3d 30    |AT+CWAUT OCONN=0 \033[0m
> [00:00:01.547,943] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
>                                             4f 4b                                            |OK               \033[0m
> [00:00:01.547,973] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)\033[0m
> [00:00:01.598,144] \033[0m<dbg> modem_cmd_handler: modem_cmd_send_ext: SENT DATA
>                                             41 54 2b 43 57 4d 4f 44  45 3d 30                |AT+CWMOD E=0     \033[0m
> [00:00:01.606,414] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
>                                             4f 4b                                            |OK               \033[0m
> [00:00:01.606,445] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)\033[0m
> [00:00:01.656,616] \033[0m<dbg> modem_cmd_handler: modem_cmd_send_ext: SENT DATA
>                                             41 54 2b 43 49 50 52 45  43 56 4d 4f 44 45 3d 31 |AT+CIPRE CVMODE=1\033[0m
> [00:00:01.661,712] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
>                                             45 52 52 20 43 4f 44 45  3a 30 78 30 31 30 39 30 |ERR CODE :0x01090
>                                             30 30 30                                         |000              \033[0m
> [00:00:01.662,475] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
>                                             45 52 52 4f 52                                   |ERROR            \033[0m
> 

### Passive mode disabled: Boots as expected.

### SNTP Request
With latest bind() additions request logs show:
[... cut ...]
>                                             53 44 4b 20 76 65 72 73  69 6f 6e 3a 76 33 2e 33 |SDK vers ion:v3.3
>                                             2d 64 65 76 2d 33 34 33  2d 67 37 66 61 39 38 35 |-dev-343 -g7fa985
>                                             39 33 62                                         |93b              \033[0m
> [00:00:01.171,386] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [SDK version:] (len:35)\033[0m
> [00:00:01.171,447] \033[0m<inf> wifi_esp_at: SDK version: v3.3-dev-343-g7fa98593b\033[0m
>

LonesomeUnixCowboy, 2024-03-11:
Yes, that's an explanation. I'm using v2.2.0.0. I have noticed when one enable `CONFIG_WIFI_ESP_AT_DNS_USE` bind will make a connection to 0.0.0.0. But in v2.2.0.0 I have not seen it to close any other UDP sockets.

WilliamGFish, 2024-03-14:
That appears to be the problem, if I use DNS then UDP is opening a dud address socket then closing it.

LonesomeUnixCowboy, 2024-03-15:
It does for me too. I have done some digging and its `dns_resolve_init_locked()` that wants to bind the invalid socket after `esp_dns_work()` started it all.

I dont know why `dns_resolve_init_locked()` whats to bind a socket yet. But i feel its wrong 😅. Im assuming its supposed not to. I appologize I have messed things up 😫

[image]

LonesomeUnixCowboy, 2024-03-18:
I'm a bit lost. Maybe the @mniestroj  can explain whats going on. The dud sockets adress is coming from `dns_resolve_init_locked()`, it's for me `local_addr4` since I'm using only IPV4:

[image]

And this is sent to `net_context_bind()` which then makes the esp driver start a UDP socket with local_addr4 (which is uninitialized).

[image]

So my question is, shouldn't `dns_resolve_init_locked()` bind to the address that the esp driver receives when `on_cmd_cipdns()` gets called, which in turn triggers `esp_dns_work()` and then rest of the DNS reconfiguration?

[image]

LonesomeUnixCowboy, 2024-03-22:
@WilliamGFish i have make a fix i would like you to try [fix_udp_socket](https://github.com/LonesomeUnixCowboy/zephyr/tree/fix_udp_socket). It work for me and i hope it will solves your issue too.

WilliamGFish, 2024-03-22:
> @WilliamGFish i have make a fix i would like you to try [fix_udp_socket](https://github.com/LonesomeUnixCowboy/zephyr/tree/fix_udp_socket). It work for me and i hope it will solves your issue too.

No, only if I remove the esp_bind code does it work at all; It doesn't look like a firmware issue even the IPD parsing fails.

When I set CONFIG_WIFI_ESP_AT_CIPDINFO_USE =y struggles to cope with a DNS lookup, see the Received "SEND OK" : **wifi_esp_at: Invalid cmd end 0x2c, expected 0x3a**
That leads to a failure to parse the +IPD
Data being received is the DNS response
```
*****************************************SNTP Connection Requested*****************************************
Getting the time from time.google.com [00:00:24.045,104] \033[0m<dbg> wifi_esp_at_offload: esp_recv: link_id 0, timeout 0, cb 0x45b69, data 0x20016728\033[0m
[00:00:24.045,196] \033[0m<dbg> wifi_esp_at_offload: esp_sendto: link 0, timeout 0\033[0m
[00:00:24.045,227] \033[0m<dbg> wifi_esp_at_offload: esp_connect: link 0, timeout 5000\033[0m
[00:00:24.045,257] \033[0m<dbg> wifi_esp_at_offload: _sock_connect: *********** _sock_connect ******** PORT: ( 53 )\033[0m
[00:00:24.045,257] \033[0m<dbg> wifi_esp_at_offload: _sock_connect: *********************************************************\033[0m
[00:00:24.045,410] \033[0m<dbg> wifi_esp_at_offload: _sock_connect: link 0, ip_proto UDP, addr 8.8.8.8\033[0m
[00:00:24.045,440] \033[0m<dbg> modem_cmd_handler: modem_cmd_send_ext: SENT DATA
                                            41 54 2b 43 49 50 53 54  41 52 54 3d 30 2c 22 55 |AT+CIPST ART=0,"U
                                            44 50 22 2c 22 38 2e 38  2e 38 2e 38 22 2c 35 33 |DP","8.8 .8.8",53
                                            2c 35 33                                         |,53              \033[0m
[00:00:24.053,314] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
                                            30 2c 43 4f 4e 4e 45 43  54                      |0,CONNEC T       \033[0m
[00:00:24.053,375] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [0,CONNECT] (len:9)\033[0m
[00:00:24.053,833] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
                                            4f 4b                                            |OK               \033[0m
[00:00:24.053,863] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)\033[0m
[00:00:24.053,985] \033[0m<dbg> wifi_esp_at_offload: _sock_send: link 0, len 33\033[0m
[00:00:24.054,168] \033[0m<dbg> modem_cmd_handler: modem_cmd_send_ext: SENT DATA
                                            41 54 2b 43 49 50 53 45  4e 44 3d 30 2c 33 33 2c |AT+CIPSE ND=0,33,
                                            22 38 2e 38 2e 38 2e 38  22 2c 35 33             |"8.8.8.8 ",53    \033[0m
[00:00:24.058,502] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
                                            4f 4b                                            |OK               \033[0m
[00:00:24.058,532] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [OK] (len:2)\033[0m
[00:00:24.058,868] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match direct cmd [>] (ret:1)\033[0m
[00:00:24.064,270] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
                                            52 65 63 76 20 33 33 20  62 79 74 65 73          |Recv 33  bytes   \033[0m
[00:00:24.068,328] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
                                            53 45 4e 44 20 4f 4b                             |SEND OK          \033[0m
[00:00:24.068,389] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match cmd [SEND OK] (len:7)\033[0m
[00:00:24.125,244] \033[1;31m<err> wifi_esp_at: Invalid cmd end 0x2c, expected 0x3a\033[0m
[00:00:24.125,274] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: match direct cmd [+IPD] (ret:4)\033[0m
[00:00:24.129,852] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
                                            2c 30 2c 39 37 2c 38 2e  38 2e 38 2e 38 2c 35 33 |,0,97,8. 8.8.8,53
                                            3a 6e c9 81 80 00 01 00  04 00 00 00 00 04 74 69 |:n...... ......ti
                                            6d 65 06 67 6f 6f 67 6c  65 03 63 6f 6d 00 00 01 |me.googl e.com...
                                            00 01 c0 0c 00 01 00 01  00 00                   |........ ..      \033[0m
[00:00:24.131,225] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
                                            71 00 04 d8 ef 23 00 c0  0c 00 01 00 01 00 00    |q....#.. ....... \033[0m
[00:00:24.132,629] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
                                            71 00 04 d8 ef 23 08 c0  0c 00 01 00 01 00 00    |q....#.. ....... \033[0m
[00:00:24.134,002] \033[0m<dbg> modem_cmd_handler: cmd_handler_process_rx_buf: RECV
                                            71 00 04 d8 ef 23 04 c0  0c 00 01 00 01 00 00    |q....#.. ....... \033[0m
 Failed to acquire SNTP, code -101

```

LonesomeUnixCowboy, 2024-04-09:
@mniestroj fixes the issue with opening a socket to 0.0.0.0:0. So that can be merged in my opinion.

It looks like to me that its a parsing issue that is left to be solved. With `CONFIG_WIFI_ESP_AT_CIPDINFO_USE` enabled togheter with `CONFIG_WIFI_ESP_AT_PASSIVE_MODE` enabled or !enabled, older versions of ESP-AT sends a command that the driver cant handle.

[image]

So i wonder now how the IPD msg looks for you version of ESP-AT @WilliamGFish?

PR #71237 (2024-04-10): drivers: wifi: esp_at: do not connect socket on bind(INADDR_ANY)
All connect() syscalls result in `EISCONN` error, since underlying
`_sock_connect()` is attempted to be called twice. Once from `net_context.c`'s
`bind_default()` (with `INADDR_ANY`) as part of `esp_bind()` and second time as
part of `esp_connect()`.

Do not call `_sock_connect()` from `esp_bind(INADDR_ANY)`, which happens as
part of `connect()`.

Fixes: dbf3d6e9110a ("drivers: esp_at: implement bind() and recvfrom() for
  UDP sockets")
Fixes: #69530
closed 2024-04-10 by dleach02