zephyrproject-rtos/zephyr #69549: UPSQUARED: tests/kernel/timer/timer_behavior/kernel.timer.timer fails.

gbarkadiusz, 2024-02-28:
**Describe the bug**
```
tests/kernel/timer/timer_behavior/kernel.timer.timer  FAILED Timeout (device: up2, 62.739s)
```

**To Reproduce**
Steps to reproduce the behavior:
1. cd <path/to>zephyrproject/zephyr
2. source zephyr-env.sh
3. twister --device-testing --hardware-map <path/to>/map.yaml -T tests/kernel/timer/timer_behavior/kernel.timer.timer -vv

**Expected behavior**
Test pass or is skipped.

**Impact**
annoyance.

**Logs and console output**
```
<path_to>/twister-out/up_squared/tests/kernel/timer/timer_behavior/kernel.timer.timer/handler.log
```
```
*** Booting Zephyr OS build v3.6.0-216-g650a59479c40 (delayed boot 500ms) ***
Running TESTSUITE timer_jitter_drift
===================================================================
START - test_jitter_drift_timer_period
periodic timer behavior test using built-in restart mechanism
collecting time samples for approx 10 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 19200000, kernel tick rate 10000
period duration statistics for 10000 samples (0 rollovers):
expected: 1000 us,       	19200.000000 cycles
min:      995.885417 us, 	19121 cycles
max:      1003.958333 us, 	19276 cycles
mean:     999.999719 us, 	19199.994600 cycles
variance: 1.961379 us, 	723.042771 cycles
stddev:   1.400492 us, 	26.889456 cycles
timer start cycle 490564437, end cycle 682564383,
total time 9999997.187500 us, expected time 10000000.000000 us,
expected time drift 0.000000 us, difference -2.812500 us
PASS - test_jitter_drift_timer_period in 10.071 seconds
===================================================================
START - test_jitter_drift_timer_startdelay
periodic timer behavior test using explicit start with delay
collecting time samples for approx 10 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 19200000, kernel tick rate 10000
period duration statistics for 10000 samples (0 rollovers):
expected: 1000 us,       	19200.000000 cycles
min:      996.406250 us, 	19131 cycles
max:      1003.489583 us, 	19267 cycles
mean:     999.999917 us, 	19199.998400 cycles
variance: 0.870469 us, 	320.889597 cycles
stddev:   0.932989 us, 	17.913391 cycles
timer start cycle 684238438, end cycle 876238422,
total time 9999999.166667 us, expected time 10000000.000000 us,
expected time drift 0.000000 us, difference -0.833333 us
PASS - test_jitter_drift_timer_startdelay in 10.071 seconds
===================================================================
TESTSUITE timer_jitter_drift succeeded
Running TESTSUITE timer_tick_train
===================================================================
START - test_one_tick_timer_train
Initializing 4 Timers, Tick Rate 10000Hz, Expecting 25000 callbacks in 11000 ms
Starting Timers with Skews
Faking busy work, remaining timers is 4, timer callbacks 1
Faking busy work, remaining timers is 4, timer callbacks 641
Faking busy work, remaining timers is 4, timer callbacks 1281
Faking busy work, remaining timers is 4, timer callbacks 1921
Faking busy work, remaining timers is 4, timer callbacks 2563
Faking busy work, remaining timers is 4, timer callbacks 3203
Faking busy work, remaining timers is 4, timer callbacks 3843
Faking busy work, remaining timers is 4, timer callbacks 4485
Faking busy work, remaining timers is 4, timer callbacks 5125
Faking busy work, remaining timers is 4, timer callbacks 5766
```
**Environment :**
 - OS: Linux
 - Toolchain; zephyr-sdk-0.16.5
 - Zephyr Version:  Zephyr OS build v3.6.0-216-g650a59479c40

najumon1980, 2024-03-02:
Could you please verify with following PR:
https://github.com/zephyrproject-rtos/zephyr/pull/69705

KamilxPaszkiet, 2024-03-13:
Currently, the issue is reproduced very sporadically.

@najumon1980 It's hard to know if your PR fixes it.

golowanow, 2024-06-07:
Closing the issue as false negative after investigations on the same setup as the reported.

The test was failing on default Twister timeout (60 seconds) which becomes just few seconds less than it needs for both flash and test execution operations:  just 2-3 seconds of time increase at the particular flash setup affects the overall timings.

At the moment there are no long-term statistics for the test itself running longer than it should on the up2 platform.

closed 2024-06-07 by golowanow