zephyrproject-rtos/zephyr #69552: sysbuild/twister/cmake/kconfig: Issue with setting remote board in twister using sysbuild

nordicjm, 2024-02-28:
**Describe the bug**
There appears to be an issue with sysbuild/twister/cmake/kconfig whereby if there is a sysbuild project which has a remote image which sets a board from a value provided to it and twister provides this value through `extra_args` it seems to have a malformed string, and fails to configure.

**Expected behavior**
String should be passed successfully

**Impact**
Showstopper

**Environment (please complete the following information):**
 - Commit SHA or Version used: `main`

57300, 2024-02-29:
This bug applies to any namespaced config of string type, such as:

```yaml
extra_args:
  SB_CONFIG_BOOT_SIGNATURE_KEY_FILE="foo.pem"
  mcuboot_CONFIG_KERNEL_BIN_NAME="mcuboot"
```

I don't think this needs to be supported by `extra_args` though, with its quirky handling of quotes.
There is special treatment for CMake variables starting with `CONFIG_`: (which didn't need to be merged imho)

https://github.com/zephyrproject-rtos/zephyr/blob/4f34a410fa2c96bd98f4eb2db296bf0c3e8b0d0e/scripts/pylib/twister/twisterlib/runner.py#L1036-L1040

But trying to extend this to `<namespace>_CONFIG_` would risk matching false positives, like `APPLICATION_CONFIG_DIR`.

I'd focus on supporting namespaced configs in `extra_configs`, which already works with `west build -T`. Twister doesn't allow it because it generates a `testsuite_extra.conf` with `extra_configs` contents verbatim*, but it could be updated to generate multiple Kconfig fragments, one per namespace. For example:
* all `SB_CONFIG_*` would go to `SB_EXTRA_CONF_FILE=sysbuild_testsuite_extra.conf`
* all `other_CONFIG_*` would go to `other_EXTRA_CONF_FILE=other_testsuite_extra.conf`
* the rest would go to the old `EXTRA_CONF_FILE=testsuite_extra.conf`

*except for arch/platform filters, which west doesn't handle

gmarull, 2024-03-01:
more details https://github.com/zephyrproject-rtos/zephyr/issues/69669

tejlmand, 2024-03-04:
Proper quoting of Kconfig settings through twister is already supported today, but one must use `extra_configs`, not `extra_args`.

The problem with `extra_configs` is that it generates a conf file and pass this conf file as `-DOVERLAY_CONFIG=<twister-name>.conf`, where it ought to be `-DSB_OVERLAY_CONFIG` when using sysbuild.

This can be seen by updating https://github.com/zephyrproject-rtos/zephyr/blob/5046229bb94cd2683ec1022848991d387152a0d2/scripts/pylib/twister/twisterlib/runner.py#L1061 to `SB_OVERLAY_CONFIG` and run the `sample.yml` in https://github.com/zephyrproject-rtos/zephyr/pull/69652.
That PR will now build correctly if setting `REMOTE_BOARD` as `extra_configs` value.

Will post a fix so that sysbuild is supported in `extra_configs`.

Note: `x_OVERLAY_CONFIG` is deprecated in favor of `x_EXTRA_CONF_FILE`.

tejlmand, 2024-03-04:
> But trying to extend this to <namespace>_CONFIG_ would risk matching false positives, like APPLICATION_CONFIG_DIR.

imo that is really not a risk, because APPLICATION_CONFIG_DIR this is not a Kconfig but a CMake setting, and thus it  should be passed through `extra_args` not `extra_configs`.

nashif, 2024-06-04:
@tejlmand  can you please take a look?

closed 2025-03-23 by github-actions[bot]