zephyrproject-rtos/zephyr #69590: Cannot build any code (including samples) on xiao_esp32c3, worked fine 2 weeks back.

sudharsan-007, 2024-02-29:
[image]

Build not working for xiao_esp32c3 (worked perfectly fine 2 weeks back), but I am able to build the same sample code Hello World on xiao_esp32s3. Initially I thought this could be my project, but none of the code seems to build for xiao_esp32c3 now after updating west. 

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr/samples/hello_world 
2. west build --pristine -b xiao_esp32c3
4. See error

**Environment (please complete the following information):**
 - OS: MacOS

sudharsan-007, 2024-02-29:
I fixed it by updating Zephyr SDK from zephyr-sdk-0.16.4 to zephyr-sdk-0.16.5. 

1. `cd ~` (this is where i have my zephyr sdk installed, you might have to cd into directory containing sdk)
2. `sudo rm -rf zephyr-sdk-0.16.4` (enter password if prompted) 
3. reinstall using zephyr sdk, following instructions in this (zephyr sdk installation page) [link](https://docs.zephyrproject.org/latest/develop/toolchains/zephyr_sdk.html) 
4. deactivate environment `deactivate`
5. cd zephyrproject/zephyr 
6. west update 

I am not sure if west update is needed. But I think this error is because my sdk was in the previous version and the zephyr project was updated using `git pull` and 'west update' inside the zephyr directory

seyednasermoravej, 2024-02-29:
Hi 
previously I have a similar problem, in my case the Wi-Fi didn't work after updating west. In my experience must update everything all together. west, sdk, etc. Your sdk version is 0.16.4 but the last version is 0.16.5.
I tested the hello word project and it worked for me on windows 11.
[image]
[image]
[image]
 Unfortunately, I can't the the code on Mac but if you want I can test on Ubuntu 22.04

closed 2024-02-29 by sudharsan-007

sudharsan-007, 2024-02-29:
Hello Sayed,

Thank you for your quick response. I just tried reinstalling the sdk and it
worked great, I closed the issue with the steps. Originally this issue got
generated without updating anything, but I am sure this could be my laptop
like bash auto updating west or something like that. I will keep my eyes
opened for issues like this in the future. Once again thank you and the
zephyr team.

Best,
Sudu