zephyrproject-rtos/zephyr #69614: doc: Documentation issue in 'releases/index'

alwa-nordic, 2024-02-29:
There is an inconsistency in [Supported Releases].

- [Release Management] says 3.7 (LTS) release is 2024/07/26.

- [Supported Releases] says the following:

  > Support and maintenance for an LTS release stops at least half a year after the following LTS release is published.

- That implies [Supported Releases] **entry for the previous LTS, Zephyr 2.7.5, must have EOL no sooner than 2025/01/26**.
	
[Supported Releases]: https://docs.zephyrproject.org/latest/releases/index.html#supported-releases
[Release Management]: https://github.com/zephyrproject-rtos/zephyr/wiki/Release-Management

kartben, 2024-02-29:
@nashif @aescolar @cfriedt

cfriedt, 2024-02-29:
Given that 2.7.6 is getting cut today, it would probably make some sense to update the wiki too.

PR #69628 (2024-03-01): doc: release: update eol for ltsv2 to 2025-01-26
Update the documentation page to reflect the latest LTSv2 release and tentative end-of-life (EOL) date.

Note: the actual EOL date is still dependant on the official LTSv3 release date and may be subject to change.

Fixes #69614
closed 2024-03-01 by cfriedt