zephyrproject-rtos/zephyr #69617: LE Audio: Add `bt_audio_get_chan_count`

Thalley, 2024-02-29:
**Is your enhancement proposal related to a problem? Please describe.**
In LE audio, the channel or location values (`enum bt_audio_location`) are bitfields. 
When decoding an SDU, which may have multiple frames per frame block, it's a good idea for the receiver to verify the frame block to see if it matches the number of channels for the stream. 

**Describe the solution you'd like**
Implement a function `bt_audio_get_chan_count` that takes an `enum bt_audio_location` and returns the number of channels in that value. 

The implementation would be something like 
```c
uint8_t bt_audio_get_chan_count(enum bt_audio_location chan_allocation)
{
	uint8_t cnt = 0U;

	if (chan_allocation == BT_AUDIO_LOCATION_MONO_AUDIO) {
		return 1;
	}

	while (chan_allocation != 0) {
		cnt += chan_allocation & 1U;
		chan_allocation >>= 1;
	}

	return cnt;
}
```

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

ajf58, 2024-03-04:
Would using `POPCOUNT`[1] instead of the while loop be sufficient here?

[1] https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/toolchain/gcc.h#L267

Thalley, 2024-03-05:
> Would using `POPCOUNT`[1] instead of the while loop be sufficient here?
> 
> [1] https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/toolchain/gcc.h#L267

That is a good suggestion! 
If it works on multiple toolchains and not just GCC, then yes - At least it can replace the loop, but we still need the 
```c
	if (chan_allocation == BT_AUDIO_LOCATION_MONO_AUDIO) {
		return 1;
	}
```

bbaa-demant, 2024-05-15:
assign to @bbaa-demant

commit b0dceffa (2024-05-24): Bluetooth: Audio: add bt_audio_get_chan_count

Implement a function bt_audio_get_chan_count that takes an enum
bt_audio_location and returns the number of channels in that value.

This PR fixes #69617
(https://github.com/zephyrproject-rtos/zephyr/issues/69617)
closed 2024-05-24 by MaureenHelm