zephyrproject-rtos/zephyr #69647: Device Power Manager(system): The contradiction between the documentation and the code

ZhiyuanTang17, 2024-03-01:
The Doc said **When using this type of device power management, the CPU will only enter a low power state only if no device is in the middle of a hardware transaction that cannot be interrupted**. 
But the code said **when device is busy, it will bypass the device instead of exiting the power state flow**.
Is it conflict ？
[image]
[image]

nordicjm, 2024-03-01:
I don't follow what you mean, this function suspends all devices and that check there clearly skips devices that are busy, so will not suspend them

ZhiyuanTang17, 2024-03-01:
> I don't follow what you mean, this function suspends all devices and that check there clearly skips devices that are busy, so will not suspend them

Hi Jamie，
     
     I'm glad that you can reply to you, and I apologize for not expressing my point clearly.
     The doc said the cpu will only enter a low power state when there is **no device in the middle of a hardware transaction** . The code said if the device is busy, zephyr will skip the suspend action of this device and continue to enter the power state, rather than directly terminating the CPU from entering the power state.   Is the implicit meaning that the power supply of the device and the CPU is independent?

Zhiyuan

ceolin, 2024-03-09:
Hi @ZhiyuanTang17 I agree that is confuse or even wrong. There are devices using different power rails than the cpu, or there are power states that may not trigger any power loss for devices, so there is no way to the subsystem to know when this happens. If a device sets it is busy, the subsystem does not try to suspend it, but continue its idle flow ... if a device needs  that the cpu does not use an specific power state it has to set a constraint using `pm_policy_state_lock_get/put`.

There is another way to avoid the cpu sleeping when a device is busy that is enabling the option `CONFIG_PM_NEED_ALL_DEVICES_IDLE`.  When it is enabled, if ANY device sets it is busy the cpu won't enter a low power state.

ZhiyuanTang17, 2024-03-28:
> Hi @ZhiyuanTang17 I agree that is confuse or even wrong. There are devices using different power rails than the cpu, or there are power states that may not trigger any power loss for devices, so [...]

Thanks ceolie! Your explanation clarifies my questions. Since zephyr pm subsystem is an open-source achitecture which needs soc vendor to adapt it, so it should be designed to be adaptable to various scenarios, as you mentioned, whether the device and CPU share power or the device and CPU have separate power sources.

PR #70812 (2024-04-05): doc: pm: Clarify a device pm note
Clarify conditions that prevent the cpu to enter low power state.

Fixes #69647
commit 68c0c554 (2024-04-05): doc: pm: Clarify a device pm note

Clarify conditions that prevent the cpu to enter low power
state.

Fixes #69647
closed 2024-04-05 by carlescufi