zephyrproject-rtos/zephyr #69657: USB crashes on STM32 F411 after some seconds of connection

fcgdam, 2024-03-01:
**Describe the bug**
A USB terminal connection crashes after a few seconds when using the latest release 3.6 or 3.6.99. The issue does not happen when using the 3.5 version.

Basically re-routing either the console, shell or Log subsystem to USB works just fine on version 3.5 and a USB connection (either using minicom or picocom) is stable.

When using version 3.6 or above, it seems to work, but after some time, either garbage starts to appear when using the Shell for example, or just plain disconnects.

First the issue was detected when checking a possible migration from v3.5 to v3.6 on custom boards (based on the STM32F411), but it also happens on a out-of-the-box supported Zephyr board, such as the blackpill_stm32f411ce.

**To Reproduce**
Steps to reproduce the behavior:
1. cd [....]/zephyr/samples/subsys/usb/console
2. west build -b blackpill_f411ce -p -- -DDTC_OVERLAY_FILE=app.overlay
3. west flash --runner openocd
4. picocom --b 115200 /dev/ttyACM0

The output:

```
*** Booting Zephyr OS build v3.6.0 ***
Hello World! arm
Hello World! arm
Hello World! arm
Hello World! arm
Hello World! arm
Hello World! arm
Hello World! arm
Hello World! arm
Hello World! arm
Hell�BzC���g{G�B��zg�G��zg�G��zg�F��F��G��zf�g�f�f��zf�f
```

**Expected behavior**
USB Output/input should be stable and usable.

**Impact**
USB is unusable.

**Logs and console output**
See above.

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Zephyr SDK 0.16.5-1
 - Zephyr RTOS: v3.6

**Additional context**
Bug is reproduced in a standard blackpill board with no devices connected.

FRASTM, 2024-03-01:
This pb also seen with nucleo (144pins) f429zi

sngl3, 2024-03-04:
same issue, works fine on zephyr-v3.5.0-663-gbc43e890fd7a but on v3.6.0 not working. standard console sample in subsys/usb

sngl3, 2024-03-04:
sorry, forgot, also stm32f411ce, not tried something else

Desvauxm-st, 2024-03-06:
@fcgdam
can you try to modify the file drivers/usb/device/usb_dc_stm32.c
in line 134
before 
    #define USB_FIFO_RX_MIN 160
after 
    #define USB_FIFO_RX_MIN 120
I check on nucleo_f429zi it is ok but i have not a F411

fcgdam, 2024-03-06:
Hi, yes I can confirm that with the modification it doesn't crash anymore in a standard blackpill_f411ce board and in one of our developed hardware. USB output is now ok, and the SHELL (for usb input) is also working.

I'm using the released V3.6 and not the git HEAD.

Desvauxm-st, 2024-03-07:
OK good

Desvauxm-st, 2024-03-07:
you can see after the PR:
drivers: usb: device: fix Rx FIFO min size #69897

FRASTM, 2024-04-26:
Should this issue be closed since the PR https://github.com/zephyrproject-rtos/zephyr/pull/70795/ is merged ?

marwaiehm-st, 2024-05-06:
Yes this issue should be closed , it is resolved in the PR https://github.com/zephyrproject-rtos/zephyr/pull/70795

marwaiehm-st, 2024-05-06:
Issue resolved with the PR https://github.com/zephyrproject-rtos/zephyr/pull/70795

closed 2024-05-06 by marwaiehm-st