zephyrproject-rtos/zephyr #69669: Sysbuild string Kconfig options are not quoted properly

gmarull, 2024-03-01:
**Describe the bug**

Let's take a string Kconfig option defined for sysbuild, e.g. in a sample `Kconfig.sysbuild`:

```
source "${ZEPHYR_BASE}/share/sysbuild/Kconfig"

config REMOTE_BOARD
	string "The board used for remote target"
```

When this setting is set via `-DSB_CONFIG_REMOTE_BOARD` (either CMake or via twister `sample.yaml` `extra_args`), it ends up in a generated .conf file without quotes `""`.

**To Reproduce**

Checkout https://github.com/zephyrproject-rtos/zephyr/pull/69652
Run:
```
west build -b nrf5340dk_nrf5340_cpuapp samples/basic/multicore_hello_world --sysbuild -p -- -DSB_CONFIG_REMOTE_BOARD=nrf5340dk_nrf5340_cpunet
```

Observe:
```
/home/gmarull/ws/nordic/zephyrproject/zephyr/build/zephyr/misc/generated/extra_kconfig_options.conf:2: warning: malformed string literal in assignment to REMOTE_BOARD (defined at /home/gmarull/ws/nordic/zephyrproject/zephyr/samples/basic/multicore_hello_world/Kconfig.sysbuild:6). Assignment ignored.
Parsing /home/gmarull/ws/nordic/zephyrproject/zephyr/samples/basic/multicore_hello_world/Kconfig.sysbuild
Loaded configuration '/home/gmarull/ws/nordic/zephyrproject/zephyr/build/empty.conf'
Merged configuration '/home/gmarull/ws/nordic/zephyrproject/zephyr/build/empty.conf'
Merged configuration '/home/gmarull/ws/nordic/zephyrproject/zephyr/build/zephyr/misc/generated/extra_kconfig_options.conf'
Merged configuration '/home/gmarull/ws/nordic/zephyrproject/zephyr/build/empty.conf'

error: Aborting due to Kconfig warnings
```

The contents of `build/zephyr/misc/generated/extra_kconfig_options.conf` misses string quotes `""`:
```
SB_CONFIG_REMOTE_BOARD=nrf5340dk_nrf5340_cpunet
```

The problem can be mitigated if command is invoked as:
```
west build -b nrf5340dk_nrf5340_cpuapp samples/basic/multicore_hello_world --sysbuild -p -- -DSB_CONFIG_REMOTE_BOARD='"nrf5340dk_nrf5340_cpunet"'
```

This issue becomes more of a problem when crafting `sample.yaml` files. The _double quote_ trick above doesn't seem to work with twister. So the solution there is to set `SB_CONFIG_REMOTE_BOARD` in a `.conf` file and apply it in the `sample.yaml` as, e.g.

```
    extra_args:
      SB_CONF_FILE=sysbuild/nrf5340dk_nrf5340_cpunet.conf
```

where `sysbuild/nrf5340dk_nrf5340_cpunet.conf` contains:

```
SB_CONFIG_REMOTE_BOARD="nrf5340dk_nrf5340_cpunet"
```

**Expected behavior**
Sample builds successfully in all cases.

**Impact**
Annoyance, workarounds required.

**Logs and console output**
See above.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.4
 - Commit SHA or Version used https://github.com/zephyrproject-rtos/zephyr/pull/69652

**Additional context**
N/A

nordicjm, 2024-03-01:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/69552 ?

gmarull, 2024-03-01:
> Duplicate of #69552 ?

ops, wasn't aware of it. Let me close and link this one there since this contains useful details.

closed 2024-03-01 by gmarull