zephyrproject-rtos/zephyr #69681: ESP_IDF_PATH is not set when building ESP boards

mobileoverlord, 2024-03-01:
**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

I am attempting to build the example for an espressif supported board https://docs.zephyrproject.org/3.6.0/boards/xtensa/esp32s3_devkitm/doc/index.html

The build fails when attempting to build MCUBoot with the error

```
CMake Error at /src/jschneck/zephyrproject/zephyr/cmake/modules/extensions.cmake:2377 (message):
  Assertion failed: ESP_IDF_PATH is not set
Call Stack (most recent call first):
  /src/jschneck/zephyrproject/zephyr/boards/common/esp32.board.cmake:14 (assert)
  /src/jschneck/zephyrproject/zephyr/boards/xtensa/esp32s3_devkitm/board.cmake:8 (include)
  /src/jschneck/zephyrproject/zephyr/cmake/modules/kernel.cmake:171 (include)
  /src/jschneck/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:138 (include)
  /src/jschneck/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /src/jschneck/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:12 (find_package)
```

Using 
zephyr v3.6.0
zephyr-sdk 0.16.5 

**To Reproduce**
Steps to reproduce the behavior:
1. west blobs fetch hal_espressif
2. west build -b esp32s3_devkitm --sysbuild samples/hello_world
3. See error

**Expected behavior**
I should be able to build and flash the hello world example

**Impact**
Blocker

**Environment (please complete the following information):**
 - OS: Ubunut Linux 22.04
 - Toolchain zephyr-sdk 0.16.5
 - Commit 468eb56cf24

mobileoverlord, 2024-03-01:
This seems to be a host machine installation issue. I removed `ZEPHYR_HOME` and reinstalled zephyr from the getting-started instructions.

closed 2024-03-01 by mobileoverlord

MrMarteng, 2024-08-04:
You apparently missed to execute `west update`, which will download the according dependency packages including esp-idf.