zephyrproject-rtos/zephyr #69730: tests: kernel: mimxrt595_evk/mimxrt595s/cm33 : tests/kernel/common/kernel.common.lto build failure

hakehuang, 2024-03-04:
**Describe the bug**

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/nxa22633/AppData/Local/Temp/1/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/nxa22633/AppData/Local/Temp/1/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/nxa22633/AppData/Local/Temp/1/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=ZH-CN style='tab-interval:21.0pt;word-wrap:break-word;text-justify-trim:
punctuation'>

<h2><span lang=EN style='mso-fareast-font-family:"Times New Roman";mso-ansi-language:
EN'>Failure/Error Case Details<o:p></o:p></span></h2>

CASE_NAME | ZEPHYR_VERSION | RESULT | PLATFORM | TEST_PATH
-- | -- | -- | -- | --
arch.shared_interrupt.lto | v3.6.0-337-g72a17b6d887e | Build   failure | mimxrt595_evk/mimxrt595s/cm33 | tests/kernel/interrupt
kernel.common.lto | v3.6.0-337-g72a17b6d887e | Build   failure | mimxrt595_evk/mimxrt595s/cm33 | tests/kernel/common

</body>

</html>

**To Reproduce**
Steps to reproduce the behavior:
1.  scripts/twister -p mimxrt595_evk/mimxrt595s/cm33  -j 12 -T tests --build-only --inline-logs -M all
2. See error

**Expected behavior**
Build pass

**Logs and console output**

```
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt595_evk_mimxrt595s_cm33/tests/kernel/common/kernel.common.lto/zephyr/zephyr_pre0.map 
: && ccache /opt/toolchains/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 -flto -fno-ipa-sra -ffunction-sections -fdata-sections zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt595_evk_mimxrt595s_cm33/tests/kernel/common/kernel.common.lto/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/cortex_m/cmse/libarch__arm__core__cortex_m__cmse.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/boards/nxp/mimxrt595_evk/libboards__nxp__mimxrt595_evk.a  zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_nxp/hal_nxp/lib..__modules__hal__nxp.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/toolchains/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v8-m.main/nofp"  -L/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt595_evk_mimxrt595s_cm33/tests/kernel/common/kernel.common.lto/zephyr  -lgcc  zephyr/libcode_relocation_source_lib.a  zephyr/arch/common/libisr_tables.a  -Wl,--fatal-warnings  -mcpu=cortex-m33  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && cd /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt595_evk_mimxrt595s_cm33/tests/kernel/common/kernel.common.lto/zephyr && /usr/bin/cmake -E true
/opt/toolchains/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /tmp/cc97UarK.ltrans0.ltrans.o:(.rodata.static_regions+0xc): undefined reference to `__ram_text_reloc_start'
/opt/toolchains/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /tmp/cc97UarK.ltrans0.ltrans.o:(.rodata.static_regions+0x10): undefined reference to `__ram_text_reloc_size'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

INFO    - /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt595_evk_mimxrt595s_cm33/tests/kernel/common/kernel.common.lto/build.log
ERROR   - mimxrt595_evk/mimxrt595s/cm33 tests/kernel/interrupt/arch.shared_interrupt.lto    ERROR : Build failure
INFO    - /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt595_evk_mimxrt595s_cm33/tests/kernel/interrupt/arch.shared_interrupt.lto/build.log
ERROR   - Loading Zephyr module(s) (Zephyr base): extensions
-- Found Python3: /usr/bin/python (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
Loading Zephyr module(s) (Zephyr base (cached)): zephyr_default:kconfig
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.6.0-337-g72a17b6d887e

danieldegrasse, 2024-03-06:
It appears that commit 118f1592ff5d04cdcb introduces this issue- it only occurs when `CONFIG_CODE_RELOCATION=y` the root cause seems to be that LTO generates section names different than the `.text`, `.data`, etc names that the `gen_relocate_app.py` script expects: 

```
arm-none-eabi-objdump build/zephyr/CMakeFiles/zephyr.dir/soc/arm/nxp_imx/rt5xx/flash_clock_setup.c.obj -h

build/zephyr/CMakeFiles/zephyr.dir/soc/arm/nxp_imx/rt5xx/flash_clock_setup.c.obj:     file format elf32-littlearm

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00000000  00000000  00000000  00000034  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .data         00000000  00000000  00000000  00000034  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00000000  00000000  00000000  00000034  2**0
                  ALLOC
  3 .gnu.debuglto_.debug_info 00000ab4  00000000  00000000  00000034  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING, EXCLUDE, OCTETS
  4 .gnu.debuglto_.debug_abbrev 00000129  00000000  00000000  00000ae8  2**0
                  CONTENTS, READONLY, DEBUGGING, EXCLUDE, OCTETS
  5 .gnu.debuglto_.debug_line 00000189  00000000  00000000  00000c11  2**0
                  CONTENTS, READONLY, DEBUGGING, EXCLUDE, OCTETS
  6 .gnu.debuglto_.debug_str 000008e1  00000000  00000000  00000d9a  2**0
                  CONTENTS, READONLY, DEBUGGING, EXCLUDE, OCTETS
```

This breaks the "section name" logic in `gen_relocate_app.py`: https://github.com/zephyrproject-rtos/zephyr/blob/17b1393f2e5f79a5c74fbffcc5e1265b67082b53/scripts/build/gen_relocate_app.py#L62

One option I could see here would be to disable LTO for any file being relocated, but I am wondering if there is a way to keep LTO enabled when code relocation is used. @rakons, do you have any thoughts here? I'm not very familiar with the LTO process.

rakons, 2024-03-06:
I am not familiar with the code relocation process, I will look deeper into it, but basically it seems like that if we wish some sections to stay with its original name we should disable LTO for the specific files. LTO would change section names. It can use a code from one section and create some additional sections that would contain the detected common parts of the code. Then jumps maybe generated between such a section that was not intended. And even if it is compiling and linking now - we have keep in mind that it can happen anytime if the linker detects some code repetition that may be optimized.

rettichschnidi, 2024-03-15:
Because this fails builds in CI, this issue is blocking PRs that touch the kernel code (i.e. #70176).

hakehuang, 2024-03-15:
another issue found with LTO related #70286

decsny, 2024-06-04:
I have switched the priority to low now that the CI is no longer broken and the builds technically do not error anymore, but I believe this issue is still not addressed properly

dleach02, 2024-06-07:
@danieldegrasse @rakons Given the conversation is the solution to disable LTO on files that will relocate? And if yes then can we put together a PR doing this?

dleach02, 2024-11-01:
As @danieldegrasse points out, the LTO changes up the naming symbols in the object file. Turning off LTO for that file allows it to work with the script but then there is a further linking error to deal with.