zephyrproject-rtos/zephyr #69737: Assert subsys/bluetooth/host/att.c:844

martintv, 2024-03-04:
**Describe the bug**
I did a minor adjustment to llpm sample in NCS, so that it ends up sending write commands from both central and peripheral very often. If sample is run in noisy rf-environment, this assert is reproducible pretty quickly. within 5 min normaly.
Assert Im getting:
https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/subsys/bluetooth/host/att.c#L822

 - What target platform are you using?
 NRF52840DK
 
- What have you tried to diagnose or workaround this issue?
I tried a quickfix by disabling shceduler like [this](https://github.com/martintv/sdk-zephyr/commit/209bbcd3995bb711f0911d4c0b3c4b0c2d3808d9) and assert in question does go away. But when I leave it running overnight (so takes longer to reproduce) I still get a different assert [__ASSERT_NO_MSG(req->func);](https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/subsys/bluetooth/host/att.c#L821).

Hypothesis then is that it is the iterator in [att_req_send_process](https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/subsys/bluetooth/host/att.c#L834), but I havent looked into it further.

**To Reproduce**
Steps to reproduce the behavior:
setup ncs enviormnet, checkout this
https://github.com/martintv/sdk-nrf/commits/reproducing_att_assert/
west build --pristine -b nrf52840dk_nrf52840 nrf/samples/bluetooth/llpm
west flash  <--- on two dks.

**Expected behavior**
Shouldn assert

**Impact**
Not sure yet

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

nordicjm, 2024-03-04:
You are reporting an issue in ncs which is not zephyr, ncs bugs go to https://devzone.nordicsemi.com/ so you need to reproduce this bug using upstream zephyr for it to be valid here

jori-nordic, 2024-03-04:
I asked martin to post it here. 
We have had host bug reports that use the host in combination w/ a closed-source controller *and* we have tried to debug them. In that case, martin raises a valid concern, that the code in att_process is not thread-safe.

jori-nordic, 2024-03-04:
Adding medium priority, as this iterator corruption ends up crashing the whole app.

closed 2024-03-07 by jhedberg