zephyrproject-rtos/zephyr #69743: Pointer usage mistake in NVS sample

zchris9, 2024-03-04:
There is a mistake on pointer usage in the NVS sample. The values buf, key and longarray are already pointers, which means that the reference operator must not be used when passing them to the NVS functions.

Array definition:

https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/samples/subsys/nvs/src/main.c#L71-L72

Occurence:

https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/samples/subsys/nvs/src/main.c#L105

https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/samples/subsys/nvs/src/main.c#L112

https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/samples/subsys/nvs/src/main.c#L116

https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/samples/subsys/nvs/src/main.c#L133

https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/samples/subsys/nvs/src/main.c#L152

https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/samples/subsys/nvs/src/main.c#L167

https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/samples/subsys/nvs/src/main.c#L174

https://github.com/zephyrproject-rtos/zephyr/blob/72a17b6d887ef20b3cc08a25d8ea1d622792d57a/samples/subsys/nvs/src/main.c#L190-L191

Laczen, 2024-03-04:
@zchris9, thanks for the remark. It is correct that `buf`, `key` or `longarray` could also be used, but so could `&buf`, `&key`, `&longarray`, or `&buf[0]`, `&key[0]` or `&longarray[0]`. Have you found a case where the use of `&` creates a problem or less optimal code ?

zchris9, 2024-03-04:
@Laczen, you are right, all of the 3 styles are possible. This issue needs no more attention. Thank you!

closed 2024-03-04 by zchris9