zephyrproject-rtos/zephyr #69761: Contributor Nomination tomi-font

bjarki-trackunit, 2024-03-04:
# Nomination

## GitHub User

Provide the following information about the GitHub user:

Full Name: Tomi Fontanilles
GitHub username: tomi-font
Organization: Nordic Semiconductor

## Supporting Documents

#68981
#68651
#68206
#67062
#66522

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

aescolar, 2024-03-04:
CC @nashif @MaureenHelm @carlescufi

bjarki-trackunit, 2024-03-11:
@kartben This one is stuck I think :)

nashif, 2024-03-12:
https://github.com/zephyrproject-rtos/zephyr/invitations

closed 2024-03-12 by nashif