zephyrproject-rtos/zephyr #69789: Endless loop at startup for not optimized application started by bootloader

GeorgeCGV, 2024-03-05:
**Describe the bug**
Application freezes in an endless loop when it is started by the bootloader.

When BL jumps to the APP Zephyr calls `z_arm_init_arch_hw_at_boot`. It detects enabled D-Cache and tries to disable it. To disable it calls `SCB_DisableDCache` that gets stuck in an endless loop. 

The issue is related to https://github.com/ARM-software/CMSIS_5/issues/620. I confirm that applying https://github.com/ARM-software/CMSIS_5/pull/1426 to https://github.com/zephyrproject-rtos/cmsis fixes the problem.
Patch requires an addition:
```
#ifdef CONFIG_NO_OPTIMIZATIONS
#undef __OPTIMIZE__
#endif
```

Currently, Zephyr uses CMSISv5. The upstream fixes the issue in develop branch (_update to v5.9.0 won't fix the problem_) which is not released yet and in CMSISv6 (_not used_).
It could be great to have a mention of the issue in the docs.

**To Reproduce**

1 Build and flash bootloader (_BL_) with enabled D-Cache and I-Cache (`CONFIG_CACHE_MANAGEMENT=y`).
2. Build and flash application  (_APP_) with enabled D-Cache and I-Cache (`CONFIG_CACHE_MANAGEMENT=y`) and
```
CONFIG_DEBUG=y
CONFIG_NO_OPTIMIZATIONS=y
```

**Expected behavior**
Application starts normally.

**Impact**
Might be a showstopper. Developers must be aware of the issue to not brick their devices after an application update.

**Environment (please complete the following information):**
 - Toolchain: zephyr-sdk-0.16.5 (_arm-zephyr-eabi_)
 - Commit SHA: 650a59479c40bd2f2e07b6576a1dcdef77637ad6

GeorgeCGV, 2024-05-07:
Reproduction steps change after [#1922](https://github.com/mcu-tools/mcuboot/pull/1922):

#### Approach 1
The `CONFIG_BOOT_DISABLE_CACHES` must be set to `n`, then follow the "To Reproduce" steps from above.

#### Approach 2
Build and run the bootloader with `CONFIG_NO_OPTIMIZATIONS` and `CONFIG_DEBUG` if `CONFIG_BOOT_DISABLE_CACHES` is enabled to trigger the issue in the bootloader.

snalvc, 2024-05-15:
This also happens in my setup.
Board: `sam_v71_xult` (I detached onboard Xplained Pro board debugger and replaced it w/ picoprobe)
App: `samples/hello_world`
Toolchain: `zephyr-sdk-0.16.5-1`
Zephyr commit SHA: `0a8d03b95f`
prj.conf:
```
CONFIG_DEBUG=y
CONFIG_DEBUG_INFO=y
CONFIG_NO_OPTIMIZATIONS=y
CONFIG_CACHE_MANAGEMENT=y
```
Confirmed applying [ARM-software/CMSIS_5#1426](https://github.com/ARM-software/CMSIS_5/pull/1426) to `cachel1_armv7.h` solves the problem

ithinuel, 2024-11-13:
Unfortunately https://github.com/ARM-software/CMSIS_5/pull/1426 is not in CMSIS 5.9.0.
However, it is in CMSIS_6, so migrating to it could help resolve this issue.

closed 2025-03-30 by github-actions[bot]