zephyrproject-rtos/zephyr #69809: it8xxx2_evb: fail all tests

KamilxPaszkiet, 2024-03-05:
**Describe the bug**
All tests on ITE fail.
I did a bisection and found a commit that breaks the tests: [8dc3f85](https://github.com/zephyrproject-rtos/zephyr/commit/8dc3f856229ce083c956aa301c31a23e65bd8cd8)

**To Reproduce**
Steps to reproduce the behavior:
```
twister -p it8xxx2_evb --device-testing --west-flash="../../itetool/loader.sh" --device-serial=/dev/ttyUSB0 -T tests -T tests/
```

- OS: Linux
- Toolchain: zephyr-sdk-0.16.5
- v3.6.0-402-g3bf61528cce6

aescolar, 2024-03-05:
CC @tejlmand @gmarull @nordicjm

aescolar, 2024-03-05:
CC @Dino-Li @GTLin08 @RuibinChang

KamilxPaszkiet, 2024-03-05:
> Fail how? Please elaborate

I updated the bug description

nordicjm, 2024-03-05:
If you take a board, manually build hello world and flash it, what output do you get?

KamilxPaszkiet, 2024-03-05:
> If you take a board, manually build hello world and flash it, what output do you get?

I build and flash manually...but I still have no output

Log from ite flash tool:
```
Flash via I2C interface...
ITE DLB4 Linux Flash Tool: Version 1.0.6
Current time is Tue Mar  5 16:45:02 2024

Open file: build/zephyr/zephyr.bin
Using libusb v1.0.25.11696

Connecting ITE Device....
Protect Status: mode=ff data=0 2

ITE DLB4 FW Version : 0013
===================================
CHIP ID          : 1302c1 ( 81302 ) 
Flash ID         : c8 65 14
Eraseing...      : 100%
Checking...      : 100%
Programng...     : 100%
Verifying...     : 100%

Protect Status: mode=82 data=0 2
Current time is Tue Mar  5 16:45:10 2024
```

Dino-Li, 2024-03-05:
Please change default SOC variant of it8xxx2_evb to SOC_IT81202_BX and verify test again.
https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/ite/it8xxx2_evb/Kconfig.it8xxx2_evb

KamilxPaszkiet, 2024-03-05:
> Or SOC_IT81302_BX

Changing the SOC to SOC_IT81202_BX or SOC_IT81302_BX enabled reading the output. Hello_world sample pass, kernel/common fails, but there is output for both.
However, fail kernel.common is rather suitable for a separate issue.

PR #69822 (2024-03-05): it8xxx2_evb: use correct SoC
Use SOC_IT81302_BX as the SoC for this board.

Fixes #69809
commit 0ae587ae (2024-03-05): it8xxx2_evb: use correct SoC

Use SOC_IT81302_BX as the SoC for this board.

Fixes #69809
closed 2024-03-05 by fabiobaltieri