zephyrproject-rtos/zephyr #69820: Bad multi-level interrupt encoding for nodes consuming interrupts from multiple aggregators

LaurentiuM1234, 2024-03-05:
**Describe the bug**
When dealing with a node consuming interrupts from multiple aggregators extending different interrupt lines,
the interrupt IDs will not be encoded properly.

**To Reproduce**
Any setup such as the following will do:

```
intc1: interrupt-controller@cafebabe {
    /* some more properties here */
    interrupt-controller;
    #interrupt-cells = <1>;
};

intc2: interrupt-controller@deadbeef {
     /* some more properties here */
    interrupt-controller;
    interrupts-extended = <&intc1 14>;
};

intc3: interruptc-controller@babebeef {
     /* some more properties here */
    interrupt-controller;
    interrupts-extended = <&intc1 15>;
};

consumer_node: ip@cafecafe {
    /* some more properties here */
    interrupts-extended = <&intc2 16>, <&intc3 15>;
};
```

The interrupts from `consumer_node` will be encoded as follows: `14 | ((16 + 1) << 8)` and `14 | ((15 + 1) << 8)`

**Expected behavior**
The L1 interrupts should be 14 and 15 instead of 14 and 14.

**Fix**
The issue is fixed by https://github.com/zephyrproject-rtos/zephyr/pull/68784.

aescolar, 2024-03-05:
CC @ycsin

ycsin, 2024-03-06:
fixed by #68784

PR #69834 (2024-03-08): [Backport v3.6-branch] devicetree.h: switch to DT_IRQ_INTC_BY_IDX for L2 and L3 INTID encodings
Backport 8fec5f361ca73d9877300d3e81f13353c0a7deff~2..8fec5f361ca73d9877300d3e81f13353c0a7deff from #68784.

Fixes #69820
closed 2024-03-06 by ycsin