zephyrproject-rtos/zephyr #69866: STM32 bug with DMA and calibrate

poialex, 2024-03-06:
**Describe the bug**
I'm currently working on a STM32g0b1re.
I was playing around the ADC functionalities and noticed that when using the DMA and calibrate features together, the values returned were wrong (ADC kept returning values of 54-55 regardless of what was the actual voltage inputed on the right ADC pin).
After digging around I have found that during the read function the ADC data register is written during the calibration and the buffer is written immediatly after starting the DMA transfer.
I then checked out the Ref manual for the STM32g0 where I found page 347 that the DMA should be disabled during the calibration (DMAEN = 0).
The fix would then be to disable the DMA during calibration.

**To Reproduce**
Steps to reproduce the behavior:
1. add the following line to the &adc1 desc in the stm32g0b1re devicetree :
```
dmas = < &dmamux1 1 5 (STM32_DMA_PERIPH_TO_MEMORY |
	STM32_DMA_MEM_INC |  STM32_DMA_MEM_16BITS | STM32_DMA_PERIPH_16BITS) >;
	dma-names = "dmamux";
```
and activate both the dma and dmamux through :

```
&dma1 {
	status = "okay";
};

&dmamux1 {
	status = "okay";
};
```

2. build and flash the example code for the adc in samples/drivers/adc
3. start debugging and set a breakpoint on the line adc_stm32_calibrate(dev); (line 984 of adc_stm32.c) and on the line adc_stm32_dma_start(dev, data->buffer, data->channel_count); (line 1045 of adc_stm32.c)
4. Observe the value inside the ADC_DR register before and after stepping over the calibration function, and the value of your buffer before and after stepping over the DMA enable.

**Expected behavior**
Returned conversion values are coherent with the inputed voltage

**Impact**
Returned conversion values are wrong 

**Environment (please complete the following information):**
 - Ubuntu 22.04
 - ZephyrSDK 0.16.5
 - Zephyr version : main

gautierg-st, 2024-03-07:
This concerns more series than just STM32G0. I'm looking at which series are impacted, and I'll push a fix.

gautierg-st, 2024-03-07:
I've pushed #69932 to solve this issue, could you have a look?

poialex, 2024-03-08:
The fix works on my g0, thanks a lot !

PR #69932 (2024-03-12): drivers: adc: stm32: disable dma before calibration
For some STM32 series, DMA has to be disabled before starting ADC calibration.

Fixes #69866
closed 2024-03-12 by henrikbrixandersen