zephyrproject-rtos/zephyr #69902: Compile error while building tests\bluetooth\shell for Silabs board

Dermiste, 2024-03-07:
**Describe the bug**
Error while building the bluetooth shell sample for efr32xg24_dk2601b
[image]

**To Reproduce**
Run command:
`west build -p -b efr32xg24_dk2601b samples/hello_world`

**Expected behavior**
Build completes

**Impact**
Can't run the shell on my board

**Environment (please complete the following information):**
 - Windows
 - Zephyr 3.6.99

** Other informations **
- my workspace was initiated months ago using Zephyr 3.5.99 and updated later on using `west update`
- tests\bluetooth\shell used to work back then

fkokosinski, 2024-03-11:
Hey, thanks for reporting this issue.

> **To Reproduce**
> Run command:
> `west build -p -b efr32xg24_dk2601b samples/hello_world`

Is this the right command to reproduce this issue? Because Hello world build normally for me, and I don't see how it is related to the sample in the title.

Dermiste, 2024-03-11:
My bad I updated the command

pepe2k, 2024-03-12:
@fkokosinski looks like `hal_silabs` related, the `SOC_GECKO_SERIESx` (see https://github.com/zephyrproject-rtos/hal_silabs/blob/master/gecko/CMakeLists.txt#L28) has to be updated to recently introduced `SOC_FAMILY_SILABS_Sx`.

Dermiste, 2024-03-13:
If it's hal related I assume that same kind of error will pop up at some point for other samples and project related to SiLabs boards ? i.e, a fix will happen by itself since the "bug" is wirespread?

pepe2k, 2024-03-13:
> If it's hal related I assume that same kind of error will pop up at some point for other samples and project related to SiLabs boards ? i.e, a fix will happen by itself since the "bug" is wirespread?

I believe the problem is BLE and two SOC series related only and seems to be caused by migration to the new h/w model v2. You can just apply my hotfix (inside `modules/hal/silabs`) for now to be able to build without the error.

Dermiste, 2024-03-13:
I don't work on my Zephyr project anymore, I made this PR because I thought the stack size fix would benefit other people at some point. I guess it will remain a draft until a clean build is once again possible, then I'll check again the stack size requirements

fabiobaltieri, 2024-03-13:
@pepe2k the fix is correct, that symbol changed as part of the hwmv2 migration, could you open a PR for it?

pepe2k, 2024-03-13:
@fabiobaltieri sure, will send PR.

pepe2k, 2024-03-13:
@fabiobaltieri FYI: https://github.com/zephyrproject-rtos/hal_silabs/pull/61

PR #70156 (2024-03-15): manifest: hal_silabs: update module
Brings the following fix:
gecko/cmake: replace `SOC_GECKO_SERIESx` with `SOC_FAMILY_SILABS_Sx`

Fixes: #69902
commit 02b2be0f (2024-03-15): manifest: hal_silabs: update module

Brings the following fix:
gecko/cmake: replace 'SOC_GECKO_SERIESx' with 'SOC_FAMILY_SILABS_Sx'

Fixes: #69902
closed 2024-03-15 by fabiobaltieri