zephyrproject-rtos/zephyr #69905: Hardware model v2: terminology

carlescufi, 2024-03-07:
This issue is here to decide together on the terminology related to boards after the introduction of hardware model v2.
See the [Discord thread](https://discord.com/channels/720317445772017664/1214957177941524501/1214957591969533962) for more context.

Terms to be decided upon
**************************
As a reference example, we will use the nRF5340 Development Kit (`nrf5340dk`, which is a board that has a single SoC (`nrf5340`), with two cores (`cpuapp` and `cpunet`) and a single variant (`ns`).

The terms that need defining are:

a. `nrf5340dk`
b. `nrf5340`, `cpuapp`, `ns`
c. `nrf5340/cpuapp/ns`
d. `nrf5340dk/nrf5340/cpuapp/ns`
e. `nrf5340dk_nrf5340_cpuapp_ns`

Option 1
********
This is the option that maps the closest to what is today in the tree.

a. `nrf5340dk`: **board name**
b. `nrf5340`, `cpuapp`, `ns`: **board identifier** (**soc**, **cpu cluster**, **variant**)
c. `nrf5340/cpuapp/ns`: **board identifiers**
d. `nrf5340dk/nrf5340/cpuapp/ns`: **board target**
e. `nrf5340dk_nrf5340_cpuapp_ns`: **board target**

Option 2
*********
Proposed in the Discord thread.

a. `nrf5340dk`: **board name**
b. `nrf5340`, `cpuapp`, `ns`: **board attributes** (**soc**, **cpu cluster**, **variant**)
c. `nrf5340/cpuapp/ns`: **board configuration**
d. `nrf5340dk/nrf5340/cpuapp/ns`: **board target** or **board target identifier**
e. `nrf5340dk_nrf5340_cpuapp_ns`: **board build identifier**

tejlmand, 2024-03-07:
Let's use:
e. nrf5340dk_nrf5340_cpuapp_ns: **sanitized** board ...

gmarull, 2024-03-07:
## Option 4

a. `nrf5340dk`: **board name**
b. `nrf5340`, `cpuapp`, `ns`:  already defined terms? (see (c))
c. `nrf5340/cpuapp/ns`: **board target identifier**, defined as the composition of `{soc}/{cpu cluster}/{variant}`.
d. `nrf5340dk@x/nrf5340/cpuapp/ns`: **board target**, defined as the composition of `{board name}[@{revision}]/{board target identifier}`, `[]` means optional.
e. `nrf5340dk_x_nrf5340_cpuapp_ns`: **sanitized/normalized board target**

nordicjm, 2024-03-07:
Can we have an option 5 which is a clone of 1 with b and c plurality flipped e.g. b are board identifiers and c is the board identifier (consisting of identifiers)?

nashif, 2024-03-07:
you probably want to pick a different example or generalize it, because for the ignorant user

`nrf5340/cpuapp/ns` and `nrf5340dk/nrf5340/cpuapp/ns` look both like board names

gmarull, 2024-03-07:
please add board revision into the equation

carlescufi, 2024-03-13:
Closing this as decided.

closed 2024-03-13 by carlescufi