zephyrproject-rtos/zephyr #69927: Bluetooth: When receiving "PAST received" event, recv_enabled can be set based on option provided when PAST is enabled

ivaniushkov, 2024-03-07:
When application calls bt_le_per_adv_sync_transfer_subscribe() it can provide an option `BT_LE_PER_ADV_SYNC_TRANSFER_OPT_REPORTING_INITIALLY_DISABLED` to prevent generating of sync reports.

However, when processing "PAST received" event in bt_hci_le_past_received_common() function, this option doesn't take an effect: `sync_info.recv_enabled` field is not set according to option provided earlier in `bt_le_per_adv_sync_transfer_subscribe`, it is only set to `true` by default.

More detailed description from the [comment in related PR](https://github.com/zephyrproject-rtos/zephyr/pull/69908#discussion_r1515858229):
```
Whether this value should be true or false, depends on whether the BT_LE_PER_ADV_SYNC_TRANSFER_OPT_REPORTING_INITIALLY_DISABLED bit was set in the options when bt_le_per_adv_sync_transfer_subscribe was called.

We don't have any way of storing that information currently, so this may be a bit tricky to implement.

Effectively we should do the same as

	sync_info.recv_enabled =
		!atomic_test_bit(pending_per_adv_sync->flags,
				 BT_PER_ADV_SYNC_RECV_DISABLED);
but where the issue here is that we don't have pending_per_adv_sync->flags.

To fix this issue, we need to store the mode from bt_le_per_adv_sync_transfer_subscribe in a way that we can retrieve it again here.
```

jhedberg, 2024-03-07:
@ivaniushkov do I understand correctly that #69908 was just a partial fix for this? Are you planning to work on a follow-up PR to fix the remaining issue(s)?

ivaniushkov, 2024-03-08:
> @ivaniushkov do I understand correctly that #69908 was just a partial fix for this? Are you planning to work on a follow-up PR to fix the remaining issue(s)?

No, my intention was only to fix a bug reported by UBSAN - #69908 fixed an issue when bt_hci_le_past_received_common() returned uninitialized value in sync_info.recv_enabled which is unsafe.
I am not planning to fix #69927 unfortunately but I will be happy if someone else can do that