zephyrproject-rtos/zephyr #69942: Zephyr tensorflow samples fail to build: ex. ./zephyr/samples/modules/tflite-micro/hello_world

foldedtoad, 2024-03-08:
**Describe the bug**
Zephyr tensorflow samples (./zephyr/modules/tflite_micro/hello_world for example) do not build.
This is due to missing tensorflow missing from zephyrproject/modules area.  
The README.md file may also be needing review to guide through the extra build steps needed.

**To Reproduce**
Repro of issue is pretty straight forward; just attempt to build the tensorflow tflite-micro samples.
```
> cd <zephyr_base>
> west build -b qemu_x86 samples/modules/tflite-micro/hello_world
```
The terminal output is shown below.

**Expected behavior**
Sample "Hello World" project for tflite-micro should build without issues.
Question:  is west suppose to download tensorflow module from github or is that a manual procedure?
The README.md file is unclear about this.

**Impact**
Failure to begin evaluation of TensorFlow on Zephyr platforms.

**Logs and console output**

```
user@host:~/zephyr/zephyrproject/zephyr$ west build -b qemu_x86 samples/modules/tflite-micro/hello_world
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/user/zephyr/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world
-- CMake version: 3.26.3
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/user/zephyr/zephyrproject/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.14.0")
-- Board: qemu_x86
-- Found host-tools: zephyr 0.16.5 (/home/user/zephyr/sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/home/user/zephyr/sdk-0.16.5)
-- Found Dtc: /home/user/zephyr/sdk-0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/user/zephyr/zephyrproject/zephyr/boards/x86/qemu_x86/qemu_x86.dts
-- Generated zephyr.dts: /home/user/zephyr/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/user/zephyr/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/user/zephyr/zephyrproject/zephyr/build/zephyr/dts.cmake

warning: TENSORFLOW_LITE_MICRO (defined at modules/tflite-micro/Kconfig:8) was assigned the value
'y' but got the value 'n'. Check these unsatisfied dependencies: 0 (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_TENSORFLOW_LITE_MICRO and/or look up
TENSORFLOW_LITE_MICRO in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing /home/user/zephyr/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/user/zephyr/zephyrproject/zephyr/boards/x86/qemu_x86/qemu_x86_defconfig'
Merged configuration '/home/user/zephyr/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/prj.conf'
Configuration saved to '/home/user/zephyr/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/user/zephyr/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/user/zephyr/sdk-0.16.5/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/12.2.0/../../../../x86_64-zephyr-elf/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/zephyr/sdk-0.16.5/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc
CMake Warning at /home/user/zephyr/zephyrproject/zephyr/subsys/random/CMakeLists.txt:12 (message):
  

      Warning: CONFIG_TIMER_RANDOM_GENERATOR is not a truly random generator.
      This capability is not secure and it is provided for testing purposes only.
      Use it carefully.

-- Using ccache: /usr/bin/ccache
-- Configuring done (5.1s)
-- Generating done (0.1s)
-- Build files have been written to: /home/user/zephyr/zephyrproject/zephyr/build
[... 4 lines cut ...]
-- Zephyr version: 3.6.99 (/home/user/zephyr/zephyrproject/zephyr), build: v3.6.0-159-g4a72cb16cfd5
[10/119] Building CXX object CMakeFiles/app.dir/src/main_functions.cpp.obj
FAILED: CMakeFiles/app.dir/src/main_functions.cpp.obj 
ccache /home/user/zephyr/sdk-0.16.5/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-g++ -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/home/user/zephyr/zephyrproject/zephyr/include -I/home/user/zephyr/zephyrproject/zephyr/build/zephyr/include/generated -I/home/user/zephyr/zephyrproject/zephyr/soc/x86/ia32 -isystem /home/user/zephyr/zephyrproject/zephyr/lib/libc/common/include -isystem /home/user/zephyr/zephyrproject/zephyr/lib/cpp/minimal/include -m32  -fno-strict-aliasing -Os -fcheck-new -std=c++17 -Wno-register -fno-exceptions -fno-rtti -imacros /home/user/zephyr/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -m32 -msoft-float -Wa,--divide --sysroot=/home/user/zephyr/sdk-0.16.5/x86_64-zephyr-elf/x86_64-zephyr-elf -imacros /home/user/zephyr/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/user/zephyr/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/user/zephyr/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/user/zephyr/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -mpreferred-stack-boundary=2 -mno-mmx -mno-sse --specs=picolibc.specs -nostdinc++ -fno-threadsafe-statics -MD -MT CMakeFiles/app.dir/src/main_functions.cpp.obj -MF CMakeFiles/app.dir/src/main_functions.cpp.obj.d -o CMakeFiles/app.dir/src/main_functions.cpp.obj -c /home/user/zephyr/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/src/main_functions.cpp
/home/user/zephyr/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/src/main_functions.cpp:19:10: fatal error: tensorflow/lite/micro/micro_mutable_op_resolver.h: No such file or directory
   19 | #include <tensorflow/lite/micro/micro_mutable_op_resolver.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[12/119] Building CXX object CMakeFiles/app.dir/src/output_handler.cpp.obj
FAILED: CMakeFiles/app.dir/src/output_handler.cpp.obj 
ccache /home/user/zephyr/sdk-0.16.5/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-g++ -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/home/user/zephyr/zephyrproject/zephyr/include -I/home/user/zephyr/zephyrproject/zephyr/build/zephyr/include/generated -I/home/user/zephyr/zephyrproject/zephyr/soc/x86/ia32 -isystem /home/user/zephyr/zephyrproject/zephyr/lib/libc/common/include -isystem /home/user/zephyr/zephyrproject/zephyr/lib/cpp/minimal/include -m32  -fno-strict-aliasing -Os -fcheck-new -std=c++17 -Wno-register -fno-exceptions -fno-rtti -imacros /home/user/zephyr/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -m32 -msoft-float -Wa,--divide --sysroot=/home/user/zephyr/sdk-0.16.5/x86_64-zephyr-elf/x86_64-zephyr-elf -imacros /home/user/zephyr/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/user/zephyr/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/user/zephyr/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/user/zephyr/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -mpreferred-stack-boundary=2 -mno-mmx -mno-sse --specs=picolibc.specs -nostdinc++ -fno-threadsafe-statics -MD -MT CMakeFiles/app.dir/src/output_handler.cpp.obj -MF CMakeFiles/app.dir/src/output_handler.cpp.obj.d -o CMakeFiles/app.dir/src/output_handler.cpp.obj -c /home/user/zephyr/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/ [... cut ...]
/home/user/zephyr/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/src/output_handler.hpp:20:10: fatal error: tensorflow/lite/c/common.h: No such file or directory
   20 | #include <tensorflow/lite/c/common.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[19/119] Building C object zephyr/CMakeFiles/zephyr.dir/lib/cpp/abi/cpp_init_array.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/user/.local/bin/cmake --build /home/user/zephyr/zephyrproject/zephyr/build
```
  
**Environment :**
 - OS: Ubuntu 22.04 LTS
 - Toolchain: zephyr 0.16.5
 - Zephyr v3.6.99

wentong-li, 2024-03-08:
Can you double check if tflite-micro module is registered in and pulled to your local zephyr instance? I met similar issue before and found that tflite-micro became an optional module (read the PR https://github.com/zephyrproject-rtos/zephyr/pull/61505 for details) and would NOT be pulled by default. You need to explicitly ask for it by running
```
        west config manifest.project-filter -- +tflite-micro
        west update
```

henrikbrixandersen, 2024-03-10:
> I believe the README.md file should be updated to show the use of this command, as it is necessary to build the sample code.

That's a good idea. Would you be willing to submit a pull request for adding this information?

PR #70319 (2024-03-19): tflite-micro: update README to pull optional module
tflite-micro is an optional module now, and needs to be pulled for the sample to build.

Fixes #69942
commit d0ae602c (2024-03-19): tflite-micro: update README to pull optional module

tflite-micro is an optional module now, and needs to be
pulled for the sample to build.

Fixes #69942.
closed 2024-03-19 by nashif