zephyrproject-rtos/zephyr #69955: Stm32-WB55RG high power consumption between adv peaks

edantoni96, 2024-03-08:
**Describe the bug**
I'm migrating my project to Zephyr and tried to reach the same level of Power consumption as in the original project using only HAL and FREERTOS. In my original project I could see the power consumption between two advertising peaks reaching the lowest possible level at around 10uA. I saw an example for this feature in samples/boards/stm32/power_mgmt/stm32wb_ble but I haven't been able to achieve the lowest possible power consumption. I tried to build the example as it is but still couldn't manage to get the same result.

BLE CoProcessor Binary : stm32wb5x_BLE_HCILayer_fw.bin (V1.18.0)

**My Observation**
CPU2 wakes up CPU1 from stop2 modes which is unlikely or CPU2 didn't go to STOP mode after advertising event.

**To Reproduce**
Steps to reproduce the behavior:
1. flash the coprocessor binary using version 1.18.0
2. go to zephyrproject
3. west build -b nucleo_wb55rg samples/boards/stm32/power_mgmt/stm32wb_ble
4. flash the sample using west flash / stm32 cube programmer
5. Measure the power using stm32 power monitor
[image]

**Expected behavior**
[37895](https://github.com/zephyrproject-rtos/zephyr/issues/37895)
[image]

**Impact**
The device cant reach expected power budget.

**Environment (please complete the following information):**
 - OS: Linux, & Windows
 - Toolchain Zephyr SDK v0.16.5
 - Zephyr v3.6.0

whati001, 2024-03-19:
Hi All,

I am facing exactly the same problem, as already explained within my [discord message](https://discordapp.com/channels/720317445772017664/883445358821249084/1219371392836042842).
My goal is to replace the existing application build upon ST hal and FreeRTOS with Zephyr.
Unfortunately, as @edantoni96 has mentioned the current consumption during advertisement phase is significantly higher because something does not release the resource during advertisement idle phase.

Just to illustrate my findings, which are exactly what is mentioned by @edantoni96:
1) Board: nucleo_wb55rg
2) Zephyr commit 5d054a7eb8fa7786e6459bf348e3ea1ff7208566
3) Zephyr example: samples/boards/stm32/power_mgmt/stm32wb_ble
3.1) Change BLE advertisement interval to (min) 500 ms and (max) 600 ms

[image]
 As we can see in the screenshot above the power consumption statically increases once the ble host executes `bt_le_adv_start` and remains high until `bt_le_adv_stop`. 

---
The question for me is now, if the `HCI` ble stack of ST simply does not stop CPU2 or if there is something else wrong with the host implementation in zephyr. The current app implementation, with the current consumption, leverages the `full` ble stack of ST.

erwango, 2024-03-20:
> The question for me is now, if the HCI ble stack of ST simply does not stop CPU2 or if there is something else wrong with the host implementation in zephyr.

This is coming from Zephyr integration.

whati001, 2024-03-20:
> > The question for me is now, if the HCI ble stack of ST simply does not stop CPU2 or if there is something else wrong with the host implementation in zephyr.
> 
> This is coming from Zephyr integration.

Nice that you already have a hint from where the issue comes from.
If there is something I can support please let me know. Unfortunately, I am not an expert in zephyr's ble host but happy to help where I can so resolve this issue :)

whati001, 2024-06-25:
@Erwango Any update here?

erwango, 2024-06-25:
> @erwango Any update here?

Unfortunately no.

gellhaar, 2024-11-03:
Providing some additional comments here to keep the issue alive. I can also confirm there is an excessive current consumption involved with the BLE enabled, `nucleo_wb55rg` board in my case.

Based on regular iBeacon sample. Advertising set up with an interval of 4-700 ms. In reality it is around 540 ms. Main thread sleeps forever, e.g. only work expected is when CPU2 (radio core) wakes up and advertises.

`CONFIG_PM=y`, other PM options (device PM) disabled to make debugging easier.

With STM32, LPTIM is used as low-power timer. This timer has a maximum time-out. It can't sleep "forever" but must wake up CPU1 and restart timer and go back to sleep. A clock prescaler is available to adjust the periodicity. In my case a wake-up interval of 1000 ms is configured (default setup).

Some `printk()` tracing on `pm_state_set()` and `pm_state_exit_post_ops()` calls are done, and I can see that they exactly match the LPTIM interval of one second.

Note that in the figure below, my debug tracing causes extra 20 ms (or so) delay between PM exit and enter.

I see that PM always requests `PM_STATE_SUSPEND_TO_IDLE` substate `STOP2` which is good, this is the lowest suspend to idle PM substate.

[image]

The advertisement and LPTIM wake-up intervals can be easily observed in the PPK2 current measurement seen in this figure. What is interesting here is that:

1) `pm_state_set(SUSPEND_TO_IDLE, STOP2)` causes current consumption to drop. Note that I have the ST-LINK attached and do not do any `CONFIG_PM_DEVICE` so there is a consumption bias here.
2) When advertising starts, current consumption goes up as expected.
3) When advertising is completed, current consumption _does not_ drop but stays high until next LPTIM time-out when the `STOP2` mode is entered again.
4) From my debug `printk()` output, I cannot see any additional PM exit/enter calls. Only time PM exit is called is when LPTIM time-outs. Then when kernel realises there is (still) nothing to be done, PM enters `STOP2` again.

Now I am still not sure if it is the CPU2 that doesn't go back to sleep, or if CPU1 is accidentally leaving `STOP2` without the kernel/PM subsystem realising it, possibly because of some CPU2 action.

The radio CPU2 apparently does its own low-power management, so I just assume it should play nicely here and go back to sleep once advertisement is done. I ran ST's own STM32CubeIDE beacon sample as a comparison. This example behaves as it should, going to a low 3.7 uA sleep current between the advertisements.

Still could be a ST radio firmware bug as I'm using the `HCILayer` variant in Zephyr and the full stack for the ST example. FUS version 1.20.0 used by the way.

gellhaar, 2024-11-10:
Some updates here. It was quite clear that current consumption went bananas when CPU2 did BLE advertisements, just as if CPU1 was accidentally woken up or something. Spent some time trying to figure out what happens, but I couldn't see any unsolicited interrupts or premature PM exit.

I did notice one minor discrepancy in how AN5289 says CPU1/CPU2 semaphores should be taken and given, and one difference between this appnote and how the stm32cubeprog checks CPU2 low-power flags related to this. But nothing that changed the PM behaviour when corrected.

Then I continued reading the AN5289 about other semaphores related to clock control, and on page 26 (Rev 18 of the document) I got intrigued about "there can be a race condition when CPU1 releases the USB and CPU2 releases the RNG at the same time, leading to the oscillator to be kept on."

There is also the sem5 for which the appnote writes:

> Sem5 is used to control the RNG/USB CLK48 source clock. The CPU2 updates or switches
> off the clock only when the RNG IP (Sem0) is used.
> To avoid a race condition with the CPU2, when the CPU1 needs to switch off the clock it
> must always first get Sem0, even if not using the RNG IP. This mechanism is shown in the
> BLE P-NUCLEO-WB55.USB dongle examples (see Section 8.1.3 and Figure 9). This does
> not impact the CPU2.
> Sem5 must be taken before changing RNG, ADC, CLK48, SAI1, LPTIM1, LPTIM2, I2C1,
> I2C3, LPUART1, and USART1 source clock on CPU1 (RCC_CCIPR register), to avoid a
> race condition with the CPU2 when it configures CLK48 source clock.

Don't quite understand exactly how the WB clock control and semaphores related to it works (yet), and if they're related to this problem, but as Zephyr's `stm32_clock_control_init()` in `drivers/clock_control/clock_stm32_ll_common.c` gets called on every PM exit, and it does reconfigure a lot (all?) clocks, would it be possible that the mentioned race-condition happens? It could explain the excess current consumption in such case.

Anyway, for better or worse, I added the sem5 locking/releasing around `stm32_clock_control_init()` and something happened. Can't quite explain exactly why, but current consumption is now on-par with the stm32cube heartbeat example. No more excessive current consumption in idle, just the expected short peaks at BLE advertisement and on LPTIM expiry.

[image]

gellhaar, 2024-11-26:
> @gellhaar Could you please share your changes?

The changes are found in my forked repo. I made a few corrections to follow the app notes more in detail. The one that has actual effect on power consumption is this one: https://github.com/gellhaar/zephyr/commit/f34acb5ffaca81d301e96846acd3b555b2d630f3

I haven't had the time yet to double check details. Likely the changes shouldn't be put in `soc/st/stm32/stm32wbx/power.c` but inside `stm32_clock_control_init()` instead as it is related to the use of the 48 MHz clock. As soon as I've figured out the best approach I've intented to put up a pull request.

Or should I perhaps open a formal pull request and get some suggestions from there? Not sure how things are done normally.

gellhaar, 2024-11-26:
The problem is that while my fix works, I'm not sure exactly why it does. Looking at the STM32 Cube BLE heartbeat sample I cannot see any use of this `#define CFG_HW_CLK48_CONFIG_SEMID 5`, and yet that sample is just as low-power as it gets..

gellhaar, 2024-11-26:
I don't fully understand how Zephyr/STM32 clock control is working yet. But the CLK48 handling possibly has room for improvements:

- CLK48 only needs to be enabled when either RNG or USB peripherals are needed. No point in permanently enabling it at system initialisation. This just increases current consumption (if ever so slightly).
- CLK48 HSEM should only be locked when clock is needed.
- AN5289 says CLK48 HSEM shall be locked if RNG, ADC, CLK48, SAI1, LPTIM1, LPTIM2, I2C1, I2C3, LPUART1, and USART1 source clock on CPU1 (RCC_CCIPR register), to avoid a race condition with the CPU2 when it configures CLK48 source clock. I can't really see this being considered anywhere.

Then there are some subtle differences between how low-power enter/exit is done in stm32cube examples, AN5289 and Zephyr implementation if I understand things correctly. Did a few changes, not sure they have any real/noticeable effect in reality: https://github.com/gellhaar/zephyr/commit/9b715f51a4fffa1d90ecea5e953f965160abda56 and https://github.com/gellhaar/zephyr/commit/f60aa03fe9321c3e9495f0bf9bf20ea17bcd067b.

closed 2024-11-29 by kartben