zephyrproject-rtos/zephyr #69961: tests: lpcxpresso55s36 : samples build failure

hakehuang, 2024-03-08:
**Describe the bug**

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<body lang=ZH-CN style='tab-interval:21.0pt;word-wrap:break-word;text-justify-trim:
punctuation'>

CASE_NAME | ZEPHYR_VERSION | RESULT | PLATFORM | TEST_PATH
-- | -- | -- | -- | --
sample.posix.gettimeofday | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/posix/gettimeofday
sample.posix.eventfd | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/posix/eventfd
sample.net.sockets.socketpair | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/net/sockets/socketpair
sample.net.sockets.big_http_download | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/net/sockets/big_http_download
sample.net.sockets.big_http_download.posix | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/net/sockets/big_http_download
sample.net.sockets.big_http_download.ci | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/net/sockets/big_http_download
sample.net.sockets.can.one_socket | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/net/sockets/can
sample.net.sockets.can.two_sockets | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/net/sockets/can
sample.net.sockets.http_get | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/net/sockets/http_get
sample.net.sockets.http_get.posix | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/net/sockets/http_get
sample.net.sockets.dumb_http_server | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/net/sockets/dumb_http_server
sample.net.sockets.dumb_http_server.posix | v3.6.0-610-g39f710136ef7 | Build   failure | lpcxpresso55s36 | samples/net/sockets/dumb_http_server

</body>

</html>

**To Reproduce**
Steps to reproduce the behavior:
scripts/twister -p lpcxpresso55s36  -T samples/net/sockets

**Expected behavior**
Build PASS

**Impact**
board networking features

**Logs and console output**

```
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/src/mcuxClCss_Kdf.c: In function 'mcuxClCss_Ckdf_Sp800108_Async':
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/platforms/LPC55S3x/inc/ip_platform.h:32:32: error: 'CSS_BASE' undeclared (first use in this function); did you mean 'SCS_BASE'?
   32 | #define IP_CSS ((ip_css_Type *)CSS_BASE)
      |                                ^~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/inc/internal/mcuxClCss_Internal.h:37:33: note: in expansion of macro 'IP_CSS'
   37 | #define MCUXCLCSS_ISBUSY (1U == IP_CSS->CSS_STATUS_b.CSS_BUSY)
      |                                 ^~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/src/mcuxClCss_Kdf.c:37:9: note: in expansion of macro 'MCUXCLCSS_ISBUSY'
   37 |     if (MCUXCLCSS_ISBUSY)
      |         ^~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/platforms/LPC55S3x/inc/ip_platform.h:32:32: note: each undeclared identifier is reported only once for each function it appears in
   32 | #define IP_CSS ((ip_css_Type *)CSS_BASE)
      |                                ^~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/inc/internal/mcuxClCss_Internal.h:37:33: note: in expansion of macro 'IP_CSS'
   37 | #define MCUXCLCSS_ISBUSY (1U == IP_CSS->CSS_STATUS_b.CSS_BUSY)
      |                                 ^~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/src/mcuxClCss_Kdf.c:37:9: note: in expansion of macro 'MCUXCLCSS_ISBUSY'
   37 |     if (MCUXCLCSS_ISBUSY)
      |         ^~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/src/mcuxClCss_Kdf.c: In function 'mcuxClCss_Hkdf_Rfc5869_Async':
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/platforms/LPC55S3x/inc/ip_platform.h:32:32: error: 'CSS_BASE' undeclared (first use in this function); did you mean 'SCS_BASE'?
   32 | #define IP_CSS ((ip_css_Type *)CSS_BASE)
      |                                ^~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/inc/internal/mcuxClCss_Internal.h:37:33: note: in expansion of macro 'IP_CSS'
   37 | #define MCUXCLCSS_ISBUSY (1U == IP_CSS->CSS_STATUS_b.CSS_BUSY)
      |                                 ^~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/src/mcuxClCss_Kdf.c:68:9: note: in expansion of macro 'MCUXCLCSS_ISBUSY'
   68 |     if (MCUXCLCSS_ISBUSY)
[... 14 lines cut ...]
      |                                ^~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/inc/internal/mcuxClCss_Internal.h:37:33: note: in expansion of macro 'IP_CSS'
   37 | #define MCUXCLCSS_ISBUSY (1U == IP_CSS->CSS_STATUS_b.CSS_BUSY)
      |                                 ^~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/src/mcuxClCss_Kdf.c:121:9: note: in expansion of macro 'MCUXCLCSS_ISBUSY'
  121 |     if (MCUXCLCSS_ISBUSY)
      |         ^~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/src/mcuxClCss_Kdf.c: In function 'mcuxClCss_TlsGenerateSessionKeysFromMasterKey_Async':
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/platforms/LPC55S3x/inc/ip_platform.h:32:32: error: 'CSS_BASE' undeclared (first use in this function); did you mean 'SCS_BASE'?
   32 | #define IP_CSS ((ip_css_Type *)CSS_BASE)
      |                                ^~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/inc/internal/mcuxClCss_Internal.h:37:33: note: in expansion of macro 'IP_CSS'
   37 | #define MCUXCLCSS_ISBUSY (1U == IP_CSS->CSS_STATUS_b.CSS_BUSY)
      |                                 ^~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/hal/nxp/mcux/mcux-sdk/components/css_pkc/src/comps/mcuxClCss/src/mcuxClCss_Kdf.c:147:9: note: in expansion of macro 'MCUXCLCSS_ISBUSY'
  147 |     if (MCUXCLCSS_ISBUSY)
      |         ^~~~~~~~~~~~~~~~
[172/217] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/ipv6_nbr.c.obj
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  v3.6.0-610-g39f710136ef7

aescolar, 2024-03-20:
Note this is blocking CI, failing also in other samples and tests.
For ex:
https://github.com/zephyrproject-rtos/zephyr/actions/runs/8344001190/job/22836192015?pr=70435#step:12:499
https://github.com/zephyrproject-rtos/zephyr/actions/runs/8347041107/job/22846172139?pr=65116#step:12:842

aescolar, 2024-03-20:
CC @decsny @mmahadevan108

aescolar, 2024-03-20:
The issue can be reproduced with:
1. mkdir build && cd build
1. cmake -GNinja -DBOARD=lpcxpresso55s36  ../samples/hello_world/ **-DCONFIG_ENTROPY_GENERATOR=y**
1. ninja

aescolar, 2024-03-20:
After https://github.com/zephyrproject-rtos/zephyr/commit/280fd59547cd2c069bf4f645b2bffc71235c7fa2 has been merged, this issue is not longer blocking PRs CI. So lowering the priority to its original level.
Once this issue is fixed the change in 280fd59547cd2c069bf4f645b2bffc71235c7fa2 should be reverted to reenable this platform in CI.

PR #71173 (2024-04-10): drivers: lpc55s36: Remove deprecated CSS driver
CSS was deprecated from the mcu-sdk. Removing driver from lpc55s36 to clear build error.

This is a temporary patch to remove the build error.

Fixes #69961
commit 00b46686 (2024-04-10): drivers: lpc55s36: Remove deprecated CSS driver

CSS was deprecated from the mcu-sdk. Removing driver from lpc55s36
to clear build error.

This is a temporary patch to remove the build error.

Fixes #69961
closed 2024-04-10 by nashif