zephyrproject-rtos/zephyr #69965: Board Support Arduino Nano ESP32-S3

SnappierSoap318, 2024-03-08:
Hi, i am new to zephyr and wanted to know if [This](https://docs.arduino.cc/hardware/nano-esp32) board is supported. I didnt see it listed on the [website](https://docs.zephyrproject.org/latest/boards/index.html). Is there any way to get started on it?

PS: i am a noob on board configurations and stuff, so any guidance would be awesome!

SnappierSoap318, 2024-03-08:
The pinouts afaik is same to the that of BLE IoT edition

SuGlider, 2024-03-09:
Pin mapping and schematic at [Altium.WebViewer.Files.WebViewerJob_PDF_SCHPrint (arduino.cc)](https://docs.arduino.cc/resources/schematics/ABX00083-schematics.pdf)

SnappierSoap318, 2024-03-10:
[Arduino ESP32 GitHub](https://github.com/arduino/arduino-esp32)

SnappierSoap318, 2024-03-10:
SOC Datasheet - [NORA-W106-10B](https://content.u-blox.com/sites/default/files/documents/NORA-W10_DataSheet_UBX-21036702.pdf)