zephyrproject-rtos/zephyr #69975: Low throughput over RX L2CAP connection oriented channel

dwyoung43, 2024-03-08:
My team recently upgraded to Zephyr 3.4.0 and during testing noticed a significant degradation to RX throughput over an L2CAP connection oriented channel. Previous to the upgrade, the team was achieving an RX throughput of around 250kbps and after the upgrade were only measuring 30kbps.

After debugging the issue, we were able to determine that this change of behavior was introduced by the following commit:

https://github.com/zephyrproject-rtos/zephyr/commit/2809bace6540b8a25f1ab5c37e778f60333dcb9f

Reading through the commit comments, I understand the rationale for the change, but in our application I believe we have allocated enough buffers to be able issue more than 1 credit. However since this change hardcodes the number of RX credits to 1 and disabled the application hooks to configure the number of RX credits, there is no way to configure more RX credits in our application without completely rolling back this change.

Are there any plans to restore the ability for applications to configure the number of RX credits? It's a significant problem in our application to not be able to set the number of RX credits > 1.

jhedberg, 2024-03-08:
Sounds to me like the initial credits might make sense to be a Kconfig option. @jori-nordic what do you think?

jori-nordic, 2024-03-11:
@jhedberg unfortunately, it's not that easy. I invite you and @dwyoung43 to read the commit message and testcase for a more in-depth answer. I'll give a TLDR:

The stack's "managed" L2CAP API is designed in a way where it cannot know in advance the size of the application's RX buffers. The only thing it knows before receiving an SDU is whether the app has a `.alloc_buf()` RX allocator.
The stack had some slightly broken mitigations, like sending "virtual" RX credits if the SDU was not finished and the credits were depleted.
This stems from the fact that the stack, again, tries to hide the reality of the Bluetooth spec. The application knows best the rate at which it can receive data, since it is the one that manages RX buffers. We should have just exposed a `send_credits` user API in the first place.

Which is what we (@alwa-nordic) did https://github.com/zephyrproject-rtos/zephyr/pull/58155
This is another API, that is more low-level: it doesn't do re-assembly, rather calls-back the application for each PDU received in the SDU. The application can then send credits back using `bt_l2cap_chan_give_credits()` at its leisure.

Writing this (long, sorry) reply, I now realize that there is maybe a way of "fixing" the managed l2cap API: Call the `alloc_buf()` in order to "lock" an application buffer and know its size.

If we do this at the initialization of a channel, from within `bt_l2cap_chan_connect()` for the client and right after `.accept()` for the server, we (the stack) will be able to deduce how many credits the application really has. We could then remove this 1-credit limitation.
What do you think @jhedberg @alwa-nordic ?

alwa-nordic, 2024-03-11:
@dwyoung43, does `seg_recv` solve your problem?

alwa-nordic, 2024-03-11:
> Writing this (long, sorry) reply, I now realize that there is maybe a way of "fixing" the managed l2cap API: Call the alloc_buf() in order to "lock" an application buffer and know its size.

We know the size of the application buffers. It's the MTU set by the application. But this is not useful for sending more credits.

Consider the following example: The app has one buffer of 1000 size. MTU is 1000. MPS is 100. How many credits do we send? MTU/MPS = 10. Let's send 10 credits. 
1. Good: This allows the peer to send a full SDU, split-across 10 PDUs.
2. Bad: Unfortunately this also gives the peer permission to send 10 SDUs of 100 bytes. But the app doesn't have the buffers for this (see premise).

In the case of 2., the application will then run out of buffers and the stack will disconnect the channel.

alwa-nordic, 2024-05-24:
I assume `seg_recv` was a good solution. Fixing throughput over the old API is not possible without altering the API, in which users might as well switch to `seg_recv`. Closing this as resolved.

closed 2024-05-24 by alwa-nordic