zephyrproject-rtos/zephyr #69992: Contributor self-nomination for maass-hamburg

maass-hamburg, 2024-03-08:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Fin Maaß
1. GitHub username: maass-hamburg
1. Organization (optional): Vogl Electronic GmbH

## Supporting Documents

Add links to 3-5 GitHub pull requests, in the Zephyr project, authored or
reviewed by the GitHub user that demonstrate the user's dedication to the
Zephyr project.

#68767
#69036
#69328 
#69476 

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

jukkar, 2024-03-11:
Hi @maass-hamburg, it is not clear from the description, but what networking area you would like to be a contributor, DHCPv4 or something else?

maass-hamburg, 2024-03-11:
@jukkar dhcpv4 sound goog

nashif, 2024-03-12:
https://github.com/zephyrproject-rtos/zephyr/invitations