zephyrproject-rtos/zephyr #70016: Zephyr can't find my custom board directory fter updating to 3.6.99

dolence, 2024-03-11:
After updating from 3.5 to 3.6 my custom boards aren't found anymore. My setup is as follows:

```
── zephyrproject
│   ├── apps
│   │   ├── app1
...
│   │   ├── boards
│   │   │   └── arm
│   │   │       ├── stm32f105_gwr
│   │   │       └── stm32u585_gwr
```

I was used to append the boards root directory using appp CMakeLists.txt file before the update and it worked without any issues:
```
cmake_minimum_required(VERSION 3.20.0)
#list(APPEND BOARD_ROOT ${CMAKE_CURRENT_SOURCE_DIR}/../)
list(APPEND BOARD_ROOT /home/dolence/zephyrproject/apps/boards)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(mux_stm32u5)

FILE(GLOB app_sources src/*.c)
target_sources(app PRIVATE ${app_sources})
```
I've tried to use an absolute path, still get an error:
```
CMake Warning at /home/dolence/zephyrproject/zephyr/cmake/modules/boards.cmake:131 (message):
  BOARD_ROOT element without a 'boards' subdirectory:
```

 - What target platform are you using?
 STM32U585

 - OS: Ubuntu Linux
 - Toolchain: Zephyr SDK

nordicjm, 2024-03-11:
Title says 3.6.99 and body says 3.6, which is it?

dolence, 2024-03-11:
> Title says 3.6.99 and body says 3.6, which is it?

I'm sorry. It is github main, 3.6.99

nordicjm, 2024-03-12:
You need to transition your boards to hwmv2: https://docs.zephyrproject.org/latest/hardware/porting/board_porting.html#hw-model-v2

closed 2024-03-12 by nordicjm