zephyrproject-rtos/zephyr #70031: Config option for saml21 initialization to default clock state before image jump

astonkov, 2024-03-11:
**Is your feature request related to a problem? Please describe.**
Hello everyone,

I'm using mcuboot for saml21 and I've encountered an issue with starting the image. I can successfully load the image through mcuboot, but after the jump to the application, the image does not start. The reason lies in the reinitialization of the clock in functions [gclk_reset](https://github.com/zephyrproject-rtos/zephyr/blob/aa87ed5d8a711c50c5db325b543068ac583d2a6b/soc/atmel/sam0/common/soc_saml2x.c#L32)]. To fix this problem, changes need to be made to the submodule. Is there a way in mcuboot to avoid reinitializing the clock after switching to the image?

**Describe the solution you'd like**
Add a configuration for initializing the controller to its default state before jumping to the image.

**Describe alternatives you've considered**
One alternative would be making changes in the submodule. However, this approach could complicate the process of updating to new versions of the submodule.

nandojve, 2024-03-15:
Is this same issue that this #70064 is trying to solve?

astonkov, 2024-03-15:
Yes, that's it, thank you.

closed 2024-03-15 by astonkov