zephyrproject-rtos/zephyr #70043: Hardware Timer Example

DinkarY, 2024-03-11:
Hello,

I am using "esp32_devkitm".
I want a timer interrupt for 40 microseconds for my application. For that, I want to use a Hardware Timer. However, I found no example or reference document that helped me to create a hardware timer code.

can anyone tell me how to create a Hardware timer code?

Thanks,
Dinkar

RichardSWheatley, 2024-03-12:
ESP has documentation on their website.
https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32/api-reference/peripherals/timer.html

DinkarY, 2024-03-13:
Hello, 
Thanks for the suggestion.
Can I use the same timer-related API of ESP-IDF to create the Zephyr project?
Is there any specific setting I have to add in the Zephyr project to support the hardware timer of esp32?

Thanks,
Dinkar