zephyrproject-rtos/zephyr #70047: Use custom uart handler alongside MCUmgr uart

dethredic, 2024-03-11:
I want to be able to handle serial (uart) communications using a custom protocol. I've set this up using the interrupt driven API. My problem is that I also want to enable mcumgr via serial (uart). I don't see a nice way to make these co-exist. 

I thought of a couple different options: add an API to get the current uart callback so it can be saved/managed, allow the user to enable mcumgr during runtime (not only at SYS_INIT), or pass data from my callback through to the mcumgr handler.

All these options have some drawbacks so I don't know what the best solution is.

I'm using MCUmgr for FWUP, so for my use case I only need to switch from my handler to mcumgr once, but I could see a scenario where switching back and forth could be useful.

nordicjm, 2024-03-12:
It would be a lot of effort to support something like this because it would need functions adding for removing an SMP transport and freeing up the resources of the transport. Instead I would suggest doing similar to what the shell transport does, have your own UART handling code and have it enter a mode whereby it forwards the data on to MCUmgr for processing https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/shell/backends/shell_uart.c#L94 you would need to create a new transport for this (could copy the shell one and rename it)

nordicjm, 2024-03-14:
> I'm actually using USB CDC. It seems straightforward to pass data to smp_shell using `smp_shell_rx_bytes`, but there doesn't seem to be a nice way to supply my own TX function. smp_shell seems directly tied to `shell_uart` and uses its TX function directly.

If you are using USB CDC, why not have 2 instances of it, one for your application and one dedicated for MCUmgr?

dethredic, 2024-03-14:
Ohh, I didn't know about that. I was able to use `CONFIG_USB_COMPOSITE_DEVICE` and get it working! 
It would be nice if there was a way to easily distinguish the two ports host side but I can work with this. Thanks for your help!

closed 2024-09-11 by nordicjm