zephyrproject-rtos/zephyr #70051: hal_nxp update broke i2s_speed test on MIMXRT1170-EVK

DerekSnell, 2024-03-11:
The i2s_speed test no longer passes on the MIMXRT1170-EVK board.  Bisecting found 788ba121373ba74ed197fa99fdc1bd92e9f15a42 which is an update to hal_nxp.  With the previous commit 794ab108278bbac49f16c207794b19a2986ab458, i2s_speed passes on this board.  The last Zephyr release when this test passed was v3.3.0.

The root cause of this issue still needs to be investigated.  Note that for this test to pass on this board, board modifications are required, see the [Readme](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/drivers/i2s/i2s_speed/Readme.txt).  

Debug Console output in failing 788ba121373ba74ed197fa99fdc1bd92e9f15a42:
```
*** Booting Zephyr OS build zephyr-v3.3.0-2945-g788ba121373b ***                
Running TESTSUITE drivers_i2s_speed                                             
===================================================================             
START - test_i2s_transfer_long                                                  
                                                                                
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/i2s/i2s_speed/src/test0
                                                                                
 FAIL - test_i2s_transfer_long in 2.043 seconds                                 
===================================================================             
START - test_i2s_transfer_short                                                 
0->OK                                                                           
1->OK                                                                           
2->OK                                                                           
                                                                                
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/i2s/i2s_speed/src/test)
RX START trigger failed                                                         
 FAIL - test_i2s_transfer_short in 0.019 seconds                                
===================================================================             
TESTSUITE drivers_i2s_speed failed.                                             
Running TESTSUITE drivers_i2s_speed_both_rxtx                                   
===================================================================             
START - test_i2s_dir_both_transfer_long                                         
I2S_DIR_BOTH value is not supported.                                            
 SKIP - test_i2s_dir_both_transfer_long in 0.004 seconds                        
===================================================================             
START - test_i2s_dir_both_transfer_short                                        
I2S_DIR_BOTH value is not supported.                                            
 SKIP - test_i2s_dir_both_transfer_short in 0.004 seconds                       
===================================================================             
TESTSUITE drivers_i2s_speed_both_rxtx succeeded                                 
                                                                                
------ TESTSUITE SUMMARY START ------                                           
                                                                                
SUITE FAIL -   0.00% [drivers_i2s_speed]: pass = 0, fail = 2, skip = 0, total =s
 - FAIL - [drivers_i2s_speed.test_i2s_transfer_long] duration = 2.043 seconds   
 - FAIL - [drivers_i2s_speed.test_i2s_transfer_short] duration = 0.019 seconds  
                                                                                
SUITE SKIP -   0.00% [drivers_i2s_speed_both_rxtx]: pass = 0, fail = 0, skip = s
 - SKIP - [drivers_i2s_speed_both_rxtx.test_i2s_dir_both_transfer_long] duratios
 - SKIP - [drivers_i2s_speed_both_rxtx.test_i2s_dir_both_transfer_short] duratis
                                                                                
------ TESTSUITE SUMMARY END ------                                             
                                                                                
===================================================================             
PROJECT EXECUTION FAILED                                                        
Failed after 1 attempts    
```

EmilioCBen, 2024-07-10:
I have a PR to fix this 
https://github.com/zephyrproject-rtos/zephyr/pull/75726

decsny, 2024-07-10:
reopening because fix was done worng

closed 2024-07-11 by aescolar