zephyrproject-rtos/zephyr #70063: Fail to compile tests/subsys/dsp/basicmath/zdsp.basicmath with xt-clang, for imx8mp_evk/mimx8ml8/adsp

iuliana-prodan, 2024-03-11:
**Describe the bug**
When compiling `tests/subsys/dsp/basicmath/`, for` imx8mp_evk/mimx8ml8/adsp`  with `xt-clang` I get:
```
xt-ld: app/libapp.a(q7.c.obj):(.literal.test_zdsp_add_q7+0x18): undefined reference to `malloc'
xt-ld: app/libapp.a(q7.c.obj):(.literal.test_zdsp_add_q7+0x1c): undefined reference to `log10f'
xt-ld: app/libapp.a(q7.c.obj):(.literal.test_zdsp_add_q7+0x20): undefined reference to `__fpclassifyf'
xt-ld: app/libapp.a(q7.c.obj):(.literal.test_zdsp_add_q7+0x24): undefined reference to `free'
xt-ld: app/libapp.a(q7.c.obj):(.literal.test_zdsp_add_q7_in_place+0x4): undefined reference to `memcpy'
xt-ld: app/libapp.a(q7.c.obj): in function `test_zdsp_add_q7':
/opt/samba/nxa06898/xcc/zephyrproject/zephyr/tests/subsys/dsp/basicmath/src/q7.c:26: undefined reference to `malloc'
xt-ld: app/libapp.a(q7.c.obj): in function `arm_snr_q7':
```

Same works with gcc from zephyr-sdk `(zephyr-sdk-0.16.4/xtensa-nxp_imx8m_adsp_zephyr-elf/bin/xtensa-nxp_imx8m_adsp_zephyr-elf-gcc`)

I can compile `hello_world` sample with `xt-clang`.

**To Reproduce**
Steps to reproduce the behavior:
1. set xt-clang
2. `west build -p -b imx8mp_evk/mimx8ml8/adsp -s tests/subsys/dsp/basicmath/`
3. See error

**Expected behavior**
Test should compile.

**Impact**
Cannot implement Xtensa backend for ZDSP tests.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Xtensa Toolchain version RI-2023.11-linux, Xtensa Core = hifi4_mscale_v2_0_2_prod
 - Commit SHA : zephyr afc7c20d5a67e0e95c7e5bee1e0822c4832335ea

**Additional context**
With gcc compiles:
```
zephyrproject/zephyr$ git diff
diff --git a/tests/subsys/dsp/basicmath/testcase.yaml b/tests/subsys/dsp/basicmath/testcase.yaml
index 7311f8cb906..6aed37ffacf 100644
--- a/tests/subsys/dsp/basicmath/testcase.yaml
+++ b/tests/subsys/dsp/basicmath/testcase.yaml
@@ -6,6 +6,7 @@ tests:
       - sam_e70_xplained/same70q21
       - mps2/an521/cpu0
       - native_sim
+      - imx8mp_evk/mimx8ml8/adsp
     tags: zdsp
     min_flash: 128
     min_ram: 64
```
```
/zephyrproject/zephyr$ ./scripts/twister -p imx8mp_evk/mimx8ml8/adsp -s tests/subsys/dsp/basicmath/zdsp.basicmath
ZEPHYR_BASE unset, using "/opt/samba/nxa06898/zephyrproject/zephyr"
Renaming output directory to /opt/samba/nxa06898/zephyrproject/zephyr/twister-out.1
INFO    - Using Ninja..
INFO    - Zephyr version: v3.6.0-723-gafc7c20d5a67
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /opt/samba/nxa06898/zephyrproject/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0, error:    0
INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 15.92 seconds
INFO    - In total 408 test cases were executed, 0 skipped on 1 out of total 699 platforms (0.14%)
INFO    - 0 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /opt/samba/nxa06898/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /opt/samba/nxa06898/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /opt/samba/nxa06898/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

iuliana-prodan, 2024-03-11:
^^ @stephanosio @nashif @yperess @SiyuanCheng-CN @dcpleung

dcpleung, 2024-05-13:
Hm... since it's with xt-clang, is it possible that your toolchain does not have the necessary functions in its C library??? (which should not be the case...) Maybe it requires some additional compiler flags to be passed?

closed 2024-07-27 by github-actions[bot]