zephyrproject-rtos/zephyr #70065: nanopb sample broken in 3.5.0 & latest

bgreggsmith, 2024-03-11:
**Description**
Following the instructions from https://docs.zephyrproject.org/latest/services/serialization/nanopb.html the sample is not buildable on Debian 12 or Arch/Manjaro.

Using the ```zephyr_nanopb_sources()``` definition in CMake results in the following output (on Arch/Manjaro):

```
    ~/zephws/zephyr    main !1 ?2  west build -p always -b qemu_x86 samples/modules/nanopb --pristine                                                                                            ✔ 
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/bgs/zephws/zephyr/samples/modules/nanopb
-- CMake version: 3.28.3
-- Found Python3: /usr/bin/python (found suitable version "3.11.7", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/bgs/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/bgs/zephws/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: qemu_x86, identifier: atom
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.1 (/home/bgs/zephyr-sdk-0.16.1)
-- Found toolchain: zephyr 0.16.1 (/home/bgs/zephyr-sdk-0.16.1)
-- Found Dtc: /usr/bin/dtc (found suitable version "1.7.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/bgs/zephws/zephyr/boards/qemu/x86/qemu_x86.dts
-- Generated zephyr.dts: /home/bgs/zephws/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/bgs/zephws/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/bgs/zephws/zephyr/build/zephyr/dts.cmake
Parsing /home/bgs/zephws/zephyr/samples/modules/nanopb/Kconfig
Loaded configuration '/home/bgs/zephws/zephyr/boards/qemu/x86/qemu_x86_defconfig'
Merged configuration '/home/bgs/zephws/zephyr/samples/modules/nanopb/prj.conf'
Configuration saved to '/home/bgs/zephws/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/bgs/zephws/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/bgs/zephyr-sdk-0.16.1/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/12.2.0/../../../../x86_64-zephyr-elf/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/bgs/zephyr-sdk-0.16.1/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc
CMake Warning at /home/bgs/zephws/zephyr/subsys/random/CMakeLists.txt:12 (message):
  

      Warning: CONFIG_TIMER_RANDOM_GENERATOR is not a truly random generator.
      This capability is not secure and it is provided for testing purposes only.
      Use it carefully.

-- Found Nanopb: /home/bgs/zephws/modules/lib/nanopb  
-- Found protoc: /usr/bin/protoc
-- Configuring done (2.5s)
-- Generating done (0.0s)
-- Build files have been written to: /home/bgs/zephws/zephyr/build
-- west build: building application
[1/117] Preparing syscall dependency handling

[3/117] Generating nanopb/generator/proto/nanopb_pb2.py
FAILED: nanopb/generator/proto/nanopb_pb2.py /home/bgs/zephws/zephyr/build/nanopb/generator/proto/nanopb_pb2.py 
cd /home/bgs/zephws/zephyr/build && /usr/bin/protoc -I/home/bgs/zephws/zephyr/build/nanopb/generator/proto --python_out=/home/bgs/zephws/zephyr/build/nanopb/generator/proto /home/bgs/zephws/zephyr/build/nanopb/generator/proto/nanopb.proto
Could not make proto path relative: /home/bgs/zephws/zephyr/build/nanopb/generator/proto/nanopb.proto: No such file or directory
[4/117] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (/home/bgs/zephws/zephyr), build: v3.6.0-581-g99707246527a
[5/117] Generating misc/generated/syscalls_subdirs.trigger
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/bgs/zephws/zephyr/build

```
On Debian the failure appears somewhat more severe with cmake apparently attempting to copy /usr/bin into the build directory:

```
bgs@ybuild:~/zephws/zephyr$ west build -b qemu_x86 samples/modules/nanopb --pristine
-- west build: making build dir /home/bgs/zephws/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/bgs/zephws/zephyr/samples/modules/nanopb
-- CMake version: 3.25.1
-- Found Python3: /usr/bin/python3 (found suitable version "3.11.2", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/bgs/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/bgs/zephws/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: qemu_x86, identifier: atom
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/home/bgs/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/home/bgs/zephyr-sdk-0.16.5)
-- Found Dtc: /home/bgs/zephyr-sdk-0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/bgs/zephws/zephyr/boards/qemu/x86/qemu_x86.dts
-- Generated zephyr.dts: /home/bgs/zephws/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/bgs/zephws/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/bgs/zephws/zephyr/build/zephyr/dts.cmake
Parsing /home/bgs/zephws/zephyr/samples/modules/nanopb/Kconfig
Loaded configuration '/home/bgs/zephws/zephyr/boards/qemu/x86/qemu_x86_defconfig'
Merged configuration '/home/bgs/zephws/zephyr/samples/modules/nanopb/prj.conf'
Configuration saved to '/home/bgs/zephws/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/bgs/zephws/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/bgs/zephyr-sdk-0.16.5/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/12.2.0/../../../../x86_64-zephyr-elf/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/bgs/zephyr-sdk-0.16.5/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc
CMake Warning at /home/bgs/zephws/zephyr/subsys/random/CMakeLists.txt:12 (message):
  

      Warning: CONFIG_TIMER_RANDOM_GENERATOR is not a truly random generator.
      This capability is not secure and it is provided for testing purposes only.
      Use it carefully.

-- Using ccache: /usr/bin/ccache
CMake Warning at /home/bgs/zephws/zephyr/CMakeLists.txt:864 (message):
  No SOURCES given to Zephyr library: lib__posix__shell

  Excluding target from build.

-- Found Nanopb: /home/bgs/zephws/modules/lib/nanopb  
-- Found protoc: /usr/bin/protoc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bgs/zephws/zephyr/build
-- west build: building application
[1/117] Preparing syscall dependency handling

[2/117] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (/home/bgs/zephws/zephyr), build: v3.6.0-717-gba2a8bd795db
[4/117] Generating nanopb/generator/nanopb_generator.py, nanopb/generator/proto/nanopb.proto
FAILED: nanopb/generator/nanopb_generator.py nanopb/generator/proto/nanopb.proto /home/bgs/zephws/zephyr/build/nanopb/generator/nanopb_generator.py /home/bgs/zephws/zephyr/build/nanopb/generator/proto/nanopb.proto 
cd /home/bgs/zephws/zephyr/build && /usr/bin/cmake -E copy_directory /usr/bin/ /home/bgs/zephws/zephyr/build/nanopb/generator
Error copying directory from "/usr/bin/" to "/home/bgs/zephws/zephyr/build/nanopb/generator".
[5/117] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/bgs/zephws/zephyr/build

```

**To Reproduce**
Steps to reproduce the behavior:
1. Try to compile nanopb sample for any target

**Expected behavior**
Nanopb sample should compile when pointing CMake to the relevant .proto files and not require manual generation of the headers

**Impact**
Mild annoyance, depending on what the build toolchain expect, possibly a showstopper since I am seeing some really odd crashes using the workaround of using nanopb externally and copying the generated files into the zephyr project manually.

**Logs and console output**
See above

**Environment (please complete the following information):**
 - OS: Debian 12 (nanopb 0.4.7-2, protobuf-compiler 3.21.12-3, libprotobuf-dev 3.21.12-3, grpcio-tools 1.62.1) & Arch/Manjaro (nanopb 0.4.8, protoc 25.2, grpcio-tools 1.62)

**Additional context**
Versions seem equal or greater than those mentioned in https://github.com/zephyrproject-rtos/zephyr/issues/56103
Using either the native or the pypi provided packages appears to make no difference
Bumping the nanopb version west.yml to the commit from https://github.com/zephyrproject-rtos/zephyr/pull/69746 does not help

pdgendt, 2024-03-12:
Hi @bgreggsmith, I tried on a clean Debian 12 docker, starting from the getting started guide, which seems to work fine. I don't really understand what is going wrong.

Build output:
```shell
(.venv) root@c4fd5d58edd0:~/zephyrproject/zephyr# cmake --version
cmake version 3.25.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
(.venv) root@c4fd5d58edd0:~/zephyrproject/zephyr# python3 --version
Python 3.11.2
(.venv) root@c4fd5d58edd0:~/zephyrproject/zephyr# dtc --version
Version: DTC 1.6.1
(.venv) root@c4fd5d58edd0:~/zephyrproject/zephyr# protoc --version
libprotoc 3.21.12
(.venv) root@c4fd5d58edd0:~/zephyrproject/zephyr# cat  /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
(.venv) root@c4fd5d58edd0:~/zephyrproject/zephyr# west build samples/modules/nanopb/ -b qemu_x86
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /root/zephyrproject/zephyr/samples/modules/nanopb
-- CMake version: 3.25.1
-- Found Python3: /root/zephyrproject/.venv/bin/python3 (found suitable version "3.11.2", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /root/.cache/zephyr
-- Zephyr version: 3.6.99 (/root/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: qemu_x86, identifier: atom
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/root/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/root/zephyr-sdk-0.16.5)
-- Found Dtc: /root/zephyr-sdk-0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /root/zephyrproject/zephyr/boards/qemu/x86/qemu_x86.dts
-- Generated zephyr.dts: /root/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /root/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /root/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /root/zephyrproject/zephyr/samples/modules/nanopb/Kconfig
Loaded configuration '/root/zephyrproject/zephyr/boards/qemu/x86/qemu_x86_defconfig'
Merged configuration '/root/zephyrproject/zephyr/samples/modules/nanopb/prj.conf'
Configuration saved to '/root/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/root/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /root/zephyr-sdk-0.16.5/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/12.2.0/../../../../x86_64-zephyr-elf/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/zephyr-sdk-0.16.5/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc
CMake Warning at /root/zephyrproject/zephyr/subsys/random/CMakeLists.txt:12 (message):

      Warning: CONFIG_TIMER_RANDOM_GENERATOR is not a truly random generator.
      This capability is not secure and it is provided for testing purposes only.
      Use it carefully.

-- Using ccache: /usr/bin/ccache
CMake Warning at /root/zephyrproject/zephyr/CMakeLists.txt:864 (message):
  No SOURCES given to Zephyr library: lib__posix__shell

  Excluding target from build.

-- Found Nanopb: /root/zephyrproject/modules/lib/nanopb
-- Found protoc: /usr/bin/protoc
-- Configuring done
-- Generating done
-- Build files have been written to: /root/zephyrproject/zephyr/build
-- west build: building application
[1/117] Preparing syscall dependency handling

[4/117] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (/root/zephyrproject/zephyr), build: v3.6.0-815-g98903d48c3fc
[116/117] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
             RAM:      114720 B        31 MB      0.35%
        IDT_LIST:          0 GB         2 KB      0.00%
Generating files from /root/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: qemu_x86
[117/117] cd /root/zephyrproject/zephyr/build/zephyr && /root/zephyrproject/.venv/bin/python3 /root/zephyrproject/zephyr/scripts/build/check_init_priorities.py --elf-file=/root/zephyrproject/zephyr/build/zephyr/zephyr.elf
```

pdgendt, 2024-03-12:
Are there files missing in the module? What does the following give you?

```shell
(.venv) root@c4fd5d58edd0:~/zephyrproject/zephyr# find ../modules/lib/nanopb/generator -name "*.proto"
../modules/lib/nanopb/generator/proto/nanopb.proto
../modules/lib/nanopb/generator/proto/google/protobuf/descriptor.proto
```

tokangas, 2024-05-27:
@pdgendt I have the same problem with nanopb. I'm working on a project where I encountered this issue and I'm able to replicate the exact behavior with the nanopb sample (as in the original description with Debian). I'm using Ubuntu on WSL and compilation works with Ubuntu 20.04, but not with Ubuntu 22.04.

There are no files missing in the module. The reason for the failure is, that the source directory is incorrect (`/usr/bin`) with Ubuntu 22.04:
```
-- Found Nanopb: /home/tok1/work/ncs/modules/lib/nanopb
...
[120/141] cd /home/tok1/work/ncs/zephyr/samples/modules/nanopb/build && /usr/bin/cmake -E copy_directory /usr/bin /home/tok1/work/ncs/zephyr/samples/modules/nanopb/build/nanopb/generator
FAILED: nanopb/generator/nanopb_generator.py nanopb/generator/proto/nanopb.proto /home/tok1/work/ncs/zephyr/samples/modules/nanopb/build/nanopb/generator/nanopb_generator.py /home/tok1/work/ncs/zephyr/samples/modules/nanopb/build/nanopb/generator/proto/nanopb.proto
cd /home/tok1/work/ncs/zephyr/samples/modules/nanopb/build && /usr/bin/cmake -E copy_directory /usr/bin /home/tok1/work/ncs/zephyr/samples/modules/nanopb/build/nanopb/generator
Error copying directory from "/usr/bin" to "/home/tok1/work/ncs/zephyr/samples/modules/nanopb/build/nanopb/generator".
ninja: build stopped: subcommand failed.
```
And the same step with Ubuntu 20.04:
```
-- Found Nanopb: /home/tok1/work/ncs/modules/lib/nanopb
...
[10/141] cd /home/tok1/work/ncs/zephyr/samples/modules/nanopb/build && /usr/bin/cmake -E copy_directory /home/tok1/work/ncs/modules/lib/nanopb/generator /home/tok1/work/ncs/zephyr/samples/modules/nanopb/build/nanopb/generator`
```

pdgendt, 2024-05-29:
@bgreggsmith can you try with the update proposed in #73444 
It could fix the debian issue, not sure for the manjaro one though.

pdgendt, 2024-06-05:
Closing this issue as #73444 has been merged and no feedback from the submitter.

closed 2024-06-05 by pdgendt