zephyrproject-rtos/zephyr #70068: Trouble with input gpio_keys with device runtime power management

fabiobaltieri, 2024-03-11:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/69900

<div type='discussions-op-text'>

<sup>Originally posted by **hglassdyb** March  7, 2024</sup>
Hello,

I’m trying to migrate a project from Zephyr 3.5 to Zephyr 3.6 and I encountered some issues with the input gpio_keys driver that is used for some buttons:

1. The keys are not responding to presses when using `CONFIG_PM_DEVICE=y`.

    - The problem is that the buttons are now suspended by default by the device runtime power management, and therefore they are not sending the input events. 

2. One of the keys was stuck in the high state (1).

    - The gpio keys interrupt were sending events during the initialization, and one of them stayed high (1), as if the button was always pressed. This happens even if none of the keys are pressed.
    - It seems to come from the fact that the pin is disabled before the interruption is disabled in the `gpio_keys_pm_action()`.
    - The pin is the only one with `GPIO_ACTIVE_LOW` flag, which could also be related to the issue.

So I managed to solve both of these issues with some modifications on the driver, but I'm not sure if this is a bug or a misuse of Zephyr on my side.

Here are the modifications:

```diff
diff --git a/drivers/input/input_gpio_keys.c b/drivers/input/input_gpio_keys.c
index cd277c19fdc..ee50156893b 100644
--- a/drivers/input/input_gpio_keys.c
+++ b/drivers/input/input_gpio_keys.c
@@ -197,6 +197,12 @@ static int gpio_keys_init(const struct device *dev)
 		return ret;
 	}
 
+	ret = pm_device_runtime_get(dev);
+	if (ret < 0) {
+		LOG_ERR("Failed to resume device");
+		return ret;
+	}
+
 	return 0;
 }
 
@@ -229,6 +235,8 @@ static int gpio_keys_pm_action(const struct device *dev,
 	for (int i = 0; i < cfg->num_keys; i++) {
 		const struct gpio_dt_spec *gpio = &cfg->pin_cfg[i].spec;
 
+		if( gpio_flags == GPIO_INPUT )
+		{
 			ret = gpio_pin_configure_dt(gpio, gpio_flags);
 			if (ret != 0) {
 				LOG_ERR("Pin %d configuration failed: %d", i, ret);
@@ -244,6 +252,23 @@ static int gpio_keys_pm_action(const struct device *dev,
 				LOG_ERR("interrupt configuration failed: %d", ret);
 				return ret;
 			}
+
+		}
+		else {
+			if (!cfg->polling_mode) {
+				ret = gpio_pin_interrupt_configure_dt(gpio, int_flags);
+				if (ret < 0) {
+					LOG_ERR("interrupt configuration failed: %d", ret);
+					return ret;
+				}
+			}
+
+			ret = gpio_pin_configure_dt(gpio, gpio_flags);
+			if (ret != 0) {
+				LOG_ERR("Pin %d configuration failed: %d", i, ret);
+				return ret;
+			}
+		}
 	}
 
 	if (action == PM_DEVICE_ACTION_RESUME && cfg->polling_mode) {
```</div>

gmarull, 2024-03-12:
Note: `CONFIG_PM_DEVICE=y` only means "manage devices power state during system-level power transitions", which doesn't make any sense as it's designed today. Unfortunately, it is required to enable `CONFIG_PM_DEVICE_RUNTIME`, but you also have `CONFIG_PM_DEVICE_RUNTIME_EXCLUSIVE` to disable its effects. It's such an embarrassing mess.

fabiobaltieri, 2024-03-12:
yeah but it sounds like there's a bit of  a race condion between the device disconnecting the gpio in the suspend code and the interrupt, I'll look into it

hglassdyb, 2024-03-13:
Hey @fabiobaltieri, so I tried your fix https://github.com/zephyrproject-rtos/zephyr/pull/70128 and it works great.

I'm using a custom board with an stm32h743zi.

Thank you for the help :+1:

fabiobaltieri, 2024-03-13:
Brilliant, thanks for reporting the issue.

hglassdyb, 2024-03-13:
Strange, the issue occurred on a active-low pin that has an external pull-up resistor. Maybe that could be the reason?

fabiobaltieri, 2024-03-13:
Hey you are right! The board "user" button is active high and strapped down externally, switched to a random pin active low and the issue is indeed there. Cool, thanks for pointing it out.

closed 2024-03-14 by dleach02