zephyrproject-rtos/zephyr #70097: Add external hardware triggers to the counter API

JHirsbrunner, 2024-03-12:
## Introduction

It is possible to control the sampling of an ADC in a ST controller with a timer. Also there is an option to pass trigger signals to the timers. My idea would be to extend the existing API with this feature, add flags to the configurations where necessary. Maybe make it configurable with a Kconfig option to enable the feature.
We designed an ASIC with counters and timers of similar functionality and I'm sure other controllers have these as well.

### Problem description

I would like to trigger an ADC from a counter / timer (e.g. TIM1), over a defined time t<sub>meas</sub>. This time is with measured with e. g. TIM2, and will stop TIM1 once TIM2 reaches the top value. For this, I need to be able to setup the output trigger of TIM1 and tell the ADC to listen to this trigger. This might also be a feature that could be added to for the ADC API. Then, once TIM2 reaches top, it will stop TIM1 from counting.

### Proposed change

`COUNTER_EN_HW_TRG=y` enables this feature. There are flags provided in the API to tell the configurations that they should trigger an output signal of the counter. Furthermore, there is a new configuration structure counter_trigger_cfg and function where a hardware trigger can be setup. The number of trigger lines, output triggers and output channels must be specified in counter_config_info, there is

## Detailed RFC

In this section of the document the target audience is the dev team. Upon
reading this section each engineer should have a rather clear picture of what
needs to be done in order to implement the described feature.

### Proposed change (Detailed)

This section is freeform - you should describe your change in as much detail
as possible. Please also ensure to include any context or background info here.
For example, do we have existing components which can be reused or altered.

By reading this section, each team member should be able to know what exactly
you're planning to change and how.

### Dependencies

Highlight how the change may affect the rest of the project (new components,
modifications in other areas), or other teams/projects.

### Concerns and Unresolved Questions

List any concerns, unknowns, and generally unresolved questions etc.

## Alternatives

List any alternatives considered, and the reasons for choosing this option
over them.

JHirsbrunner, 2024-03-12:
sorry, I accidently pressed ctrl+enter instead of ctrl+backspace, currently editing the text

henrikbrixandersen, 2024-03-13:
Could this be achieved solely through hardware configuration via devicetree?

I have previously written a counter driver, with support for an input channel/pin: https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/counter/counter_mcux_lptmr.c

decsny, 2024-03-18:
Triggers can come from a lot of sources beyond just counters such as GPIO, PWMs, comparators, even communication peripherals like USB can sometimes have a signal within an SOC that toggles every frame, or an I2S clock signal can be a trigger. And even your example of ADC being the thing triggered, well, ADC can also cause a trigger on some SOCs. And many other things than ADC can take a trigger from within an SOC, such as DMA, comparators, DACs, even some secondary cores can configure their interrupt tables differently based on internal chip trigger signals.

So my point here is, I think some API regarding triggers and signals being passed around within an SoC should be more broad than the counter api or adc api. I have been thinking this for some time that there should be some kind of system in zephyr to configure how to multiplex signals like these and create some mapping. And I agree with brix's sentiment that in most of these cases devicetree should be sufficient to describe the connections statically.

JHirsbrunner, 2024-03-19:
> This is a possibility, however, the same counter, say TIM1, _cannot be used for different functions in that case_

I take that back, really underestimated the possibilities of the devicetree, especially the "#*-cells" stuff with phandle-arrays. However, there is still a function / macro missing to setup such a hardware trigger mapping. Snyder's idea is probably the right approach; ever since opening this issue, I was thinking of doing the same for the ADC API.

carlescufi, 2024-03-19:
Architecture WG:

- @henrikbrixandersen mentions that this is a bit like pinctrl: there could be a common API, but this is typically very SoC-specific like pinctrl
- @decsny also agrees that this deserves its own, new driver class, perhaps named "interconnect". See also https://www.kernel.org/doc/html/latest/driver-api/interconnect.html
- @henrikbrixandersen  suggests that @JHirsbrunner uses Devicetree to solve his particular problem, unless he wants to actually go and solve the bigger problem in the form of a new interconnect subsystem

JHirsbrunner, 2024-03-19:
> @henrikbrixandersen suggests that @JHirsbrunner uses Devicetree to solve his particular problem, unless he wants to actually go and solve the bigger problem in the form of a new interconnect subsystem

I will try to solve my problem via dt for now, as it seems to be possible for my use-case. If I find some free time, I will take a look at the bigger problem, however, no promise made here.
Thanks for the fast handling of this issue.