zephyrproject-rtos/zephyr #70136: Failed to build hello world with CONFIG_POSIX_SYSCONF_IMPL_FULL=y

yashi, 2024-03-13:
**Describe the bug**

Since cdc36d090c0c0abc1006f8c3f10100b6c54d385f (posix: sysconf: add a "small" implementation option), the hello world sample failed to build.

**To Reproduce**
```
west build -b nucleo_f429zi zephyr/samples/hello_world -- -DCONFIG_POSIX_SYSCONF=y -DCONFIG_POSIX_SYSCONF_IMPL_FULL=y
```

**Expected behavior**
Succeed to build the sample `hello_world`.

**Logs and console output**
```
❯ rm -rf build && west build -b nucleo_f429zi zephyr/samples/hello_world -- -DCONFIG_POSIX_SYSCONF=y -DCONFIG_POSIX_SYSCONF_IMPL_FULL=y         ─╯
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/yashi/src/zephyr-project/zephyr/samples/hello_world
-- CMake version: 3.28.3
-- Found Python3: /home/yashi/src/zephyr-project/.venv/bin/python (found suitable version "3.11.8", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/yashi/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/yashi/src/zephyr-project/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: nucleo_f429zi, identifier: stm32f429xx
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/home/yashi/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/home/yashi/zephyr-sdk-0.16.5)
-- Found Dtc: /usr/bin/dtc (found suitable version "1.7.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/yashi/src/zephyr-project/zephyr/boards/st/nucleo_f429zi/nucleo_f429zi.dts
-- Generated zephyr.dts: /home/yashi/src/zephyr-project/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/yashi/src/zephyr-project/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/yashi/src/zephyr-project/build/zephyr/dts.cmake
Parsing /home/yashi/src/zephyr-project/zephyr/Kconfig
Loaded configuration '/home/yashi/src/zephyr-project/zephyr/boards/st/nucleo_f429zi/nucleo_f429zi_defconfig'
Merged configuration '/home/yashi/src/zephyr-project/zephyr/samples/hello_world/prj.conf'
Merged configuration '/home/yashi/src/zephyr-project/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/home/yashi/src/zephyr-project/build/zephyr/.config'
Kconfig header saved to '/home/yashi/src/zephyr-project/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/yashi/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/yashi/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
CMake Warning at /home/yashi/src/zephyr-project/zephyr/CMakeLists.txt:864 (message):
  No SOURCES given to Zephyr library: lib__posix__shell

  Excluding target from build.

-- Configuring done (3.1s)
-- Generating done (0.0s)
-- Build files have been written to: /home/yashi/src/zephyr-project/build
-- west build: building application
[1/134] Preparing syscall dependency handling

[3/134] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (/home/yashi/src/zephyr-project/zephyr), build: v3.6.0-851-gbe9a85bf6c5e
[100/134] Building C object zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj
FAILED: zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj 
ccache /home/yashi/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCORE_CM4 -DHSE_VALUE=8000000 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DSTM32F429xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/yashi/src/zephyr-project/zephyr/kernel/include -I/home/yashi/src/zephyr-project/zephyr/arch/arm/include -I/home/yashi/src/zephyr-project/zephyr/include -I/home/yashi/src/zephyr-project/build/zephyr/include/generated -I/home/yashi/src/zephyr-project/zephyr/soc/st/stm32 -I/home/yashi/src/zephyr-project/zephyr/soc/st/stm32/common/. -I/home/yashi/src/zephyr-project/zephyr/drivers -I/home/yashi/src/zephyr-project/zephyr/soc/st/stm32/stm32f4x/. -I/home/yashi/src/zephyr-project/modules/hal/cmsis/CMSIS/Core/Include -I/home/yashi/src/zephyr-project/zephyr/modules/cmsis/. -I/home/yashi/src/zephyr-project/modules/hal/stm32/stm32cube/stm32f4xx/soc -I/home/yashi/src/zephyr-project/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include -I/home/yashi/src/zephyr-project/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/Legacy -I/home/yashi/src/zephyr-project/modules/hal/stm32/stm32cube/common_ll/include -isystem /home/yashi/src/zephyr-project/zephyr/lib/libc/common/include -Wshadow -fno-strict-aliasing -Os -imacros /home/yashi/src/zephyr-project/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/yashi/zephyr-sdk-0.16.5/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/yashi/src/zephyr-project/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/yashi/src/zephyr-project/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/yashi/src/zephyr-project/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/yashi/src/zephyr-project=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -std=c99 -MD -MT zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj -MF zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj.d -o zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj -c /home/yashi/src/zephyr-project/zephyr/lib/posix/options/sysconf.c
In file included from /home/yashi/src/zephyr-project/zephyr/lib/posix/options/sysconf.c:7:
/home/yashi/src/zephyr-project/zephyr/lib/posix/options/sysconf.c: In function 'sysconf':
/home/yashi/src/zephyr-project/zephyr/include/zephyr/posix/unistd.h:137:45: error: 'CONFIG_MSG_COUNT_MAX' undeclared (first use in this function)
  137 | #define _POSIX_MQ_OPEN_MAX                  CONFIG_MSG_COUNT_MAX
      |                                             ^~~~~~~~~~~~~~~~~~~~
/home/yashi/src/zephyr-project/zephyr/include/zephyr/posix/unistd.h:192:39: note: in expansion of macro '_POSIX_MQ_OPEN_MAX'
  192 | #define MQ_OPEN_MAX                   _POSIX_MQ_OPEN_MAX
      |                                       ^~~~~~~~~~~~~~~~~~
/home/yashi/src/zephyr-project/zephyr/lib/posix/options/sysconf.c:227:24: note: in expansion of macro 'MQ_OPEN_MAX'
  227 |                 return MQ_OPEN_MAX;
[... 22 lines cut ...]
                 from /home/yashi/src/zephyr-project/zephyr/include/zephyr/kernel_includes.h:36,
                 from /home/yashi/src/zephyr-project/zephyr/include/zephyr/kernel.h:17,
                 from /home/yashi/src/zephyr-project/zephyr/include/zephyr/posix/posix_types.h:18,
                 from /home/yashi/src/zephyr-project/zephyr/include/zephyr/posix/unistd.h:9:
/home/yashi/src/zephyr-project/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/Legacy/stm32_hal_legacy.h:461:39: error: 'FLASH_PAGE_SIZE' undeclared (first use in this function); did you mean 'FLASH_CR_PSIZE'?
  461 | #define PAGESIZE                      FLASH_PAGE_SIZE
      |                                       ^~~~~~~~~~~~~~~
/home/yashi/src/zephyr-project/zephyr/lib/posix/options/sysconf.c:235:24: note: in expansion of macro 'PAGESIZE'
  235 |                 return PAGESIZE;
      |                        ^~~~~~~~
/home/yashi/src/zephyr-project/zephyr/include/zephyr/posix/unistd.h:219:39: error: 'CONFIG_MAX_TIMER_COUNT' undeclared (first use in this function)
  219 | #define TIMER_MAX                     CONFIG_MAX_TIMER_COUNT
      |                                       ^~~~~~~~~~~~~~~~~~~~~~
/home/yashi/src/zephyr-project/zephyr/lib/posix/options/sysconf.c:257:24: note: in expansion of macro 'TIMER_MAX'
  257 |                 return TIMER_MAX;
      |                        ^~~~~~~~~
[124/134] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/yashi/src/zephyr-project/build
```

**Environment (please complete the following information):**
 - OS: Debian Sid
 - Toolchain: Zephyr SDK 0.16.5
 - Commit SHA: cdc36d090c0c0abc1006f8c3f10100b6c54d385f

yashi, 2024-03-13:
@cfriedt Would you mind taking a look at this? Let me know if you need more info.  Thanks.

cfriedt, 2024-03-13:
Sure, I can check that out. I should probably rename he "FULL" config option to "SMALL" to be consistent with the original issue

yashi, 2024-03-13:
it doesn't seems to be limited to stm32. `qemu_cortex_m3` failes as well.

```
❯ rm -rf build && west build -b qemu_cortex_m3 zephyr/samples/hello_world -- -DCONFIG_POSIX_SYSCONF=y -DCONFIG_POSIX_SYSCONF_IMPL_FULL=y        ─╯
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/yashi/src/zephyr-project/zephyr/samples/hello_world
-- CMake version: 3.28.3
-- Found Python3: /home/yashi/src/zephyr-project/.venv/bin/python (found suitable version "3.11.8", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/yashi/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/yashi/src/zephyr-project/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: qemu_cortex_m3, identifier: ti_lm3s6965
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/home/yashi/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/home/yashi/zephyr-sdk-0.16.5)
-- Found Dtc: /usr/bin/dtc (found suitable version "1.7.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/yashi/src/zephyr-project/zephyr/boards/qemu/cortex_m3/qemu_cortex_m3.dts
-- Generated zephyr.dts: /home/yashi/src/zephyr-project/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/yashi/src/zephyr-project/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/yashi/src/zephyr-project/build/zephyr/dts.cmake
Parsing /home/yashi/src/zephyr-project/zephyr/Kconfig
Loaded configuration '/home/yashi/src/zephyr-project/zephyr/boards/qemu/cortex_m3/qemu_cortex_m3_defconfig'
Merged configuration '/home/yashi/src/zephyr-project/zephyr/samples/hello_world/prj.conf'
Merged configuration '/home/yashi/src/zephyr-project/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/home/yashi/src/zephyr-project/build/zephyr/.config'
Kconfig header saved to '/home/yashi/src/zephyr-project/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/yashi/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/yashi/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
CMake Warning at /home/yashi/src/zephyr-project/zephyr/CMakeLists.txt:864 (message):
  No SOURCES given to Zephyr library: lib__posix__shell

  Excluding target from build.

-- Configuring done (2.8s)
-- Generating done (0.0s)
-- Build files have been written to: /home/yashi/src/zephyr-project/build
-- west build: building application
[1/110] Preparing syscall dependency handling

[3/110] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (/home/yashi/src/zephyr-project/zephyr), build: v3.6.0-851-gbe9a85bf6c5e
[74/110] Building C object zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj
FAILED: zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj 
ccache /home/yashi/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/yashi/src/zephyr-project/zephyr/kernel/include -I/home/yashi/src/zephyr-project/zephyr/arch/arm/include -I/home/yashi/src/zephyr-project/zephyr/include -I/home/yashi/src/zephyr-project/build/zephyr/include/generated -I/home/yashi/src/zephyr-project/zephyr/soc/ti/lm3s6965 -I/home/yashi/src/zephyr-project/zephyr/soc/ti/lm3s6965/. -I/home/yashi/src/zephyr-project/modules/hal/cmsis/CMSIS/Core/Include -I/home/yashi/src/zephyr-project/zephyr/modules/cmsis/. -isystem /home/yashi/src/zephyr-project/zephyr/lib/libc/common/include -Wshadow -fno-strict-aliasing -Os -imacros /home/yashi/src/zephyr-project/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m3 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/yashi/zephyr-sdk-0.16.5/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/yashi/src/zephyr-project/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/yashi/src/zephyr-project/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/yashi/src/zephyr-project/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/yashi/src/zephyr-project=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -std=c99 -MD -MT zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj -MF zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj.d -o zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj -c /home/yashi/src/zephyr-project/zephyr/lib/posix/options/sysconf.c
In file included from /home/yashi/src/zephyr-project/zephyr/lib/posix/options/sysconf.c:7:
/home/yashi/src/zephyr-project/zephyr/lib/posix/options/sysconf.c: In function 'sysconf':
[... cut ...]
  219 | #define TIMER_MAX                     CONFIG_MAX_TIMER_COUNT
      |                                       ^~~~~~~~~~~~~~~~~~~~~~
/home/yashi/src/zephyr-project/zephyr/lib/posix/options/sysconf.c:257:24: note: in expansion of macro 'TIMER_MAX'
  257 |                 return TIMER_MAX;
      |                        ^~~~~~~~~
[99/110] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/yashi/src/zephyr-project/build
```

cfriedt, 2024-03-13:
@yashi - the main part that is difficult to solve is stm32 specific

https://github.com/zephyrproject-rtos/hal_stm32/blob/main/stm32cube/stm32f4xx/drivers/include/Legacy/stm32_hal_legacy.h#L461

It seems that nothing defines `FLASH_PAGE_SIZE`.

The fox for this issue will be 2 parts
1. Platform independent change (just about done)
2. Stm32 change

cfriedt, 2024-03-13:
@erwango 

The remaining issue that is not fixed by #70138 seem to be due to `FLASH_PAGE_SIZE` not being defined in stm32 legacy headers.

https://github.com/zephyrproject-rtos/hal_stm32/blob/main/stm32cube/stm32f4xx/drivers/include/Legacy/stm32_hal_legacy.h#L461

Is there a way that either `PAGE_SIZE` can be reserved for POSIX (which would be preferable, since it does not only relate to flash, but to thread stack sizes, etc) or whether `FLASH_PAGE_SIZE` can be defined for these headers?

cfriedt, 2024-03-13:
Set to medium since there may be a few stm32 platforms affected.

yashi, 2024-03-13:
Reopening this due to `FLASH_PAGE_SIZE` not fixed yet.

cfriedt, 2024-03-13:
It was automatically closed - thanks for reopening

erwango, 2024-03-20:
Ok, I see the issue there are no real flash pages on STM32F4 (as on STM32F7).
There are 2 alternatives:
- Define a dummy FLASH_PAGE_SIZE on these series
- Remove the culprit `#define PAGESIZE FLASH_PAGE_SIZE`

Question:
What if `PAGESIZE` is not defined ?

yashi, 2024-03-20:
> What if `PAGESIZE` is not defined ?

`unistd.h` defines `PAGE_SIZE` from `CONFIG_POSIX_PAGE_SIZE_BITS`, and `PAGESIZE` from `PAGE_SIZE`.

In the stm32 hal's perspective `PAGESIZE` is for flash memory, but for POSIX it's memory paging.  Both might be 4 KB but I feel odd to define memory page size from flash memory page size.  And no other HAL defines it.

erwango, 2024-03-21:
> And no other HAL defines it.

This is interesting. How does it build on other platforms then ?

cfriedt, 2024-03-21:
> - Remove the culprit `#define PAGESIZE FLASH_PAGE_SIZE`
> 
> Question:
> What if `PAGESIZE` is not defined ?

I would guess everything would be fine if the HAL does not define PAGESIZE.

vChavezB, 2024-04-06:
I got this error as well for board `stm32h747i_disco/stm32h747xx/m7"`

erwango, 2024-05-27:
Will actually be fixed by #73363

PR #70138 (2024-03-13): posix: sysconf: use Z_SC_VAL_IFDEF for sysconf constants
Previously, CONFIG_MAX_TIMER_COUNT and CONFIG_MSG_COUNT_MAX were used without checking if CONFIG_TIMER and CONFIG_POSIX_MQUEUE were enabled.

Use Z_SC_VAL_IFDEF() to conditionally define those constants based on their gating Kconfig values, otherwise, they are defined to -1.

Partially
Fixes #70136
PR #73363 (2024-06-03): west.yml: hal_stm32: Don't include legacy headers anymore
We're not supposed to rely on symbols provided by these files. In case it is required stm32 zephyr code base should be updated rather than relying on legacy definitions.

Additionally:
Fixes #70136
commit a8b04606 (2024-06-03): west.yml: hal_stm32: Don't include legacy headers anymore

We're not supposed to rely on symbols provided by these files.
In case it is required stm32 zephyr code base should be updated rather
than relying on legacy definitions.

Additionally:
Fixes #70136
closed 2024-05-28 by dnltz