zephyrproject-rtos/zephyr #70153: driver: bluetooth: userchan: Stuck in read operation

LingaoM, 2024-03-13:
Hi All, 

When I was testing a Nordic Controller, I discovered that Zephyr's Host might have problems HCI in userchan mode.

Reproduction steps

1. You need to integrate the test patch I provided(https://github.com/LingaoM/zephyr/commit/3cabf9bfe6bb099b6ea5f4fdadc87e5b7d36f761).

2. west build -b native_sim zephyr/samples/bluetooth/mesh -p auto (Must be native_sim, so HCI userchan mode is used.)

3. west build -b bbc_microbit_v2 zephyr/samples/bluetooth/mesh_provisioner -p auto (This could be the nRF52840 DK either one is fine.) & west flash
4. If you don't have a NEW PTS Dongle, please find the nRF52840 Dongle and flash [pts.zip](https://github.com/zephyrproject-rtos/zephyr/files/14586013/pts.zip) (this is the Controller firmware provided by Bluetooth SIG) Use `nrfutil dfu usb-serial -pkg pts.zip  -p /dev/ttyACM0`.
Mount The Controller.
>>>  sudo btattach -B /dev/ttyACM0 -S 1000000 -R
Attaching Primary controller to /dev/ttyACM0
Switched line discipline from 0 to 15
Device index 0 attached
Manufacturer: 2024

6. Run zephyr.exe generated in the second step. Use `sudo build/zephyr/zephyr.exe --bt-dev=hci0`.

7. Open new terminal, `sudo btmon -w btsnoop.pcapng`.
8. Wait patiently for a while, and you can see that the PB-ADV PB-GATT network provision test is being executed repeatedly.
``` C
Waiting for unprovisioned beacon...
Provisioning over PB-GATT dddd0000000000000000000000000000
Waiting for node to be added...
Added node 0x0002
Waiting for unprovisioned beacon...
Provisioning over PB-ADV dddd0000000000000000000000000000
Waiting for node to be added...
Added node 0x0002
Waiting for unprovisioned beacon...
Provisioning over PB-GATT dddd0000000000000000000000000000
Waiting for node to be added...
Added node 0x0002
Waiting for unprovisioned beacon...
Provisioning over PB-ADV dddd0000000000000000000000000000
Waiting for node to be added...
Added node 0x0002
Waiting for unprovisioned beacon...
Provisioning over PB-GATT dddd0000000000000000000000000000
Waiting for node to be added...
Added node 0x0002
```
 9. The problem will arise when the program in step 6 suddenly crashes.
 ``` C
[00:01:32.220,000] <dbg> bt_driver: rx_thread: calling read()
[00:01:32.220,000] <dbg> bt_driver: rx_thread: Calling bt_recv(0x565c5db4)
[00:01:32.220,000] <dbg> bt_hci_core: bt_recv: buf 0x565c5db4 len 59
[00:01:32.220,000] <dbg> bt_driver: rx_thread: calling read()
[00:01:32.220,000] <dbg> bt_driver: rx_thread: Calling bt_recv(0x565c5dd0)
[00:01:32.220,000] <dbg> bt_hci_core: bt_recv: buf 0x565c5dd0 len 47
[00:01:32.220,000] <dbg> bt_driver: rx_thread: calling read()
[00:01:32.220,000] <dbg> bt_gatt: bt_gatt_attr_read: handle 0x0000 offset 0 length 2
[00:01:32.220,000] <dbg> bt_gatt: bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:01:32.220,000] <dbg> bt_gatt: bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:01:32.220,000] <dbg> bt_gatt: bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:01:32.220,000] <dbg> bt_gatt: bt_gatt_attr_read: handle 0x0000 offset 0 length 2
[00:01:32.220,000] <dbg> bt_gatt: bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:01:32.220,000] <dbg> bt_gatt: bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:01:32.220,000] <dbg> bt_gatt: bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:01:32.220,000] <dbg> bt_gatt: db_hash_gen: Hash: 
                                  fb 77 e2 2d 82 8a 4b 4d  45 c6 c0 32 1e cf 57 af |.w.-..KM E..2..W.
ASSERTION FAIL [err == 0] @ ZEPHYR_BASE/subsys/bluetooth/host/hci_core.c:340
	command opcode 0x2039 timeout with err -11
@ ZEPHYR_BASE/subsys/bluetooth/host/hci_core.c:340
[00:02:32.210,000] <err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:02:32.210,000] <err> os: z_fatal_error: Current thread: 0x565caf80 (unknown)
[00:02:32.210,000] <err> os: k_sys_fatal_error_handler: Halting system
Exiting due to fatal error
```
10. You can see from the timestamp that the program suddenly got stuck, and there is available more data in btsnoop.pcapng.

cvinayak, 2024-03-13:
> ```c
> ASSERTION FAIL [err == 0] @ ZEPHYR_BASE/subsys/bluetooth/host/hci_core.c:340
>    command opcode 0x2039 timeout with err -11
> @ ZEPHYR_BASE/subsys/bluetooth/host/hci_core.c:340
> ```

It appears the HCI Controller is not responding. Is this the crash in the  `mesh_provisioner` sample on the `bbc_microbit_v2` or  the crash on the `native_posix`?

> 6. Run zephyr.exe generated in the second step. Use sudo build/zephyr/zephyr.exe --bt-dev=hci0.

Crash in the `native_posix`  connected to the PTS dongle, right?

LingaoM, 2024-03-13:
> It appears the HCI Controller is not responding. Is this the crash in the  `mesh_provisioner` sample on the `bbc_microbit_v2` or  the crash on the `native_posix`?
>
>
>
>
>
>
> Crash in the `native_posix` [...]

At first I thought it was like the controller didn't reply, but I looked at the btsnoop log and found that the controller did reply. Then I opened the log and tested it in detail, and found that the call read() in userchan was stuck, which caused the problem.

LingaoM, 2024-03-13:
> It appears the HCI Controller is not responding. Is this the crash in the mesh_provisioner sample on the bbc_microbit_v2 or the crash on the native_posix?

native

cvinayak, 2024-03-13:
> > It appears the HCI Controller is not responding. Is this the crash in the mesh_provisioner sample on the bbc_microbit_v2 or the crash on the native_posix?
> 
> native

@jori-nordic Does this seem like deadlock in HCI drivers Tx and Rx threads? (is this deadlock limited to driver us in native_posix?)

@LingaoM May be, if this is deadlock, can it be reproduced purely between two target boards?

jori-nordic, 2024-03-13:
no idea

LingaoM, 2024-03-13:
> May be, if this is deadlock, can it be reproduced purely between two target boards?

This problem cannot be detected using a real board

jhedberg, 2024-03-13:
This might also be a Linux bug. If you look at the `userchan.c` code it's quite simple - the `read()` is always followed by `poll()` indicating that there is data to be read (`POLLIN`).

LingaoM, 2024-03-14:
[image]

The reason was found, it was K_FOREVER caused by this BT_HCI_EVT_LE_EXT_ADVERTISING_REPORT.

LingaoM, 2024-03-14:
This is a problem introduced by https://github.com/zephyrproject-rtos/zephyr/commit/6ede31428de3e03c4e482d3b83610245b777d192.

hermabe, 2024-03-14:
> This is a problem introduced by [6ede314](https://github.com/zephyrproject-rtos/zephyr/commit/6ede31428de3e03c4e482d3b83610245b777d192). Previous i push this fix [8b54f08](https://github.com/zephyrproject-rtos/zephyr/commit/8b54f08a690e2817be111ed43cd33ec05dfe25b3).

We have been through this before: https://github.com/zephyrproject-rtos/zephyr/issues/43246. Extended advertising reports cannot be discarded.

LingaoM, 2024-03-14:
> We have been through this before: #43246. Extended advertising reports cannot be discarded.

But the problem remains.

carlescufi, 2024-04-11:
> But the problem remains.

@cvinayak please describe the sequence here so we can discuss it in the meeting.
@LingaoM why does this only happen with the userchan driver? the changes you linked are not driver-specific, they are on the general Host side.

carlescufi, 2024-06-06:
@LingaoM how should we continue here? Do you have a proposal?

LingaoM, 2024-06-10:
extended advertising events also need to add a discard mechanism to avoid thread deadlock.

LingaoM, 2024-06-25:
> @LingaoM why does this only happen with the userchan driver? the changes you linked are not driver-specific, they are on the general Host side.

@carlescufi  I guess this change cause difference: https://github.com/zephyrproject-rtos/zephyr/commit/6316ca44af6094919673e25bd5cc3baa112e461e , the userchan use `K_FOREVER` for Extended Advertising Event.

LingaoM, 2024-06-25:
But I don’t think either one is an ideal implementation.
1. For userchan implementation, use K_FOREVER, although extended advertising event discarding is avoided, but deadlock is caused.
2. For Controller implementation, use K_SECONDS(10), deadlock is avoided, but the host status is inconsistent, resulting in ASSERT(https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/host/scan.c#L673)

I fully understand that extended advertising cannot simply be set to discarding, as this will destroy the HCI state, but we have to find a way to solve this matter instead of leaving it alone.

closed 2024-07-05 by LingaoM