zephyrproject-rtos/zephyr #70161: Bluetooth extended adverti

rokobert, 2024-03-13:
**Notes (delete this)**
Github Discussions (https://github.com/zephyrproject-rtos/zephyr/discussions)
are available to first verify that the issue is a genuine Zephyr bug and not a
consequence of Zephyr services misuse.

This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.

**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

hermabe, 2024-03-13:
> [00:00:00.051,000] bt_hci_core: opcode 0x2036 status 0x01

This is the error "Unknown command", check that you have enabled support in the controller or contact your controller vendor.

erwango, 2024-03-13:
This feature is not implemented. See https://github.com/zephyrproject-rtos/zephyr/issues/51430

rokobert, 2024-03-14:
@erwango, thank you for answer. Sorry for not search open issues.
Since there is no date when this features will be implemented I have a question.
Is it possible to implement under Zephyr RTOS STM32 Bluetooth SDK?
In short. Using "stm32wb5x_BLE_Stack_full_extended_fw.bin" on CPU2, use source code from STM32CUBE_WB SDK and integrate into Zephyr app. Is there any guides yet?

erwango, 2024-03-14:
> In short. Using "stm32wb5x_BLE_Stack_full_extended_fw.bin" on CPU2, use source code from STM32CUBE_WB SDK and integrate into Zephyr app. Is there any guides yet?

No, but [here](https://github.com/zephyrproject-rtos/zephyr/discussions/62001) you'll find a discussion about preliminary work. You could ping guys there to know what is their status and if they're willing to collaborate and upstream a solution with your help.

closed 2024-03-15 by rokobert