zephyrproject-rtos/zephyr #70164: fs_write failed

AydinyanNarek, 2024-03-13:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/70163

<div type='discussions-op-text'>

<sup>Originally posted by **AydinyanNarek** March 13, 2024</sup>
Hello,

# Introdcution

I am using `STM32` microcontroller.
My application logger logs data to the Ram-FS, instead of flash memory. (Flash memory has a fixed count of write operations).
I have created `Ram-FS(/etc and /tmp)` using FALSH_SIMULATOR.
It looks like this.
```
/ {
	soc {
		flash_controller0: flash-controller@0 {
			compatible = "zephyr,sim-flash";
			reg = <0x00000000 DT_SIZE_K(64)>;

			#address-cells = <1>;
			#size-cells = <1>;
			erase-value = <0xff>;

			flash_sim0: flash_sim@0 {
				status = "okay";
				compatible = "soc-nv-flash";
				erase-block-size = <512>;
				write-block-size = <1>;
				reg = <0x00000000 DT_SIZE_K(64)>;

				partitions {
					compatible = "fixed-partitions";
					#address-cells = <1>;
					#size-cells = <1>;

					lfsram_partition0: partition@0 {
						label = "/tmp";
						reg = <0x00000000 0x00008000>;
					};

					lfsram_partition1: partition@1 {
						label = "/etc";
						reg = <0x00008000 0x00010000>;
					};
				};
			};
		};
	};
};

    fstab {
        compatible = "zephyr,fstab";
        lfs1: lfs1 {
			compatible = "zephyr,fstab,littlefs";
			read-size = <512>;
			prog-size = <512>;
			cache-size = <4096>;
			lookahead-size = <512>;
			block-cycles = <512>;
            partition = <&storage_partition>;
			mount-point = "/opt";
			automount;
        };

          lfsram0: lfsram0 {
	          compatible = "zephyr,fstab,littlefs";
	          mount-point = "/tmp";
	          partition = <&lfsram_partition0>;
	          automount;
	          read-size = <16>;
	          prog-size = <16>;
	          cache-size = <64>;
	          lookahead-size = <32>;
	          block-cycles = <512>;
          };
          
          lfsram1: lfsram1 {
	          compatible = "zephyr,fstab,littlefs";
	          mount-point = "/etc";
	          partition = <&lfsram_partition1>;
	          automount;
	          read-size = <16>;
	          prog-size = <16>;
	          cache-size = <64>;
	          lookahead-size = <32>;
	          block-cycles = <512>;
          };
    };
```
# Notes. 
1. `/opt` is using flash0 memory. (just skip this)
2. The combined log size is less than 10kb, which is less then the /tmp partition size
3. I am not writing 10kb log at once. I am doing small logs (max 512 bytes at a time).

# Problems.
I am writing logs to `/tmp` directory using `std::ofstream`
1. The file remains empty Until I call a close function on the stream. (std::endl, std::flash, rdbuff(0,0,) ..etc doesn't help)
2. Using the open\close combination after each write helps the data to appear in the file, however when I am doing it multiple times(let's say more than 50)  the `fs_write` returns error code `-22` which is `INVALID ARGUMENT ERROR` and printing errno returns `The Socket is not available`.

Does someone have an idea about this ?

 

</div>

Laczen, 2024-03-14:
@AydinyanNarek your second partition is to large to fit in the specified 64kB of the simulator, reduce the size of the second partition or increase the simulator size.

AydinyanNarek, 2024-03-14:
@Laczen what do you mean saying  `second partition` ?

Laczen, 2024-03-14:
> @Laczen what do you mean saying `second partition` ?

lfsram_partition1 starts at 32kB and is 64kB, so this needs 96kB.

AydinyanNarek, 2024-03-15:
> > @Laczen what do you mean saying `second partition` ?
> 
> lfsram_partition1 starts at 32kB and is 64kB, so this needs 96kB.

Thanks, it works now. 
Should I open an issue related to `std::flush` related functional?
The problem is that  `std::ostream` internal buffer's(`rdbuf`) data will not be written to the file without closing the file.